dsmrreader/dsmr-reader #386: Website geeft 504 Gateway Time-Out

hunter-nl, 2017-11-20:
Nadat ik alles volgens de handleiding een frisse installatie heb uitgevoerd zonder problemen, probeer ik de dashboard te openen. Echter ik krijg 504 Gateway Time-Out melding.

in nginx error log:
2017/11/20 15:02:00 [error] 536#536: *1 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.123.10, server: _, request: "GET / HTTP/1.1", upstream: "http://unix:///var/tmp/gunicorn--dsmr_webinterface.socket/", host: "dsmr.local"

in /var/tmp/ bestaat de socket.
-rw-r--r-- 1 dsmr dsmr    3 Nov 20 14:56 dsmrreader--dsmr_backend.pid
-rw-r--r-- 1 dsmr dsmr    4 Nov 20 14:56 gunicorn--dsmr_webinterface.pid
srwxrwxrwx 1 dsmr dsmr    0 Nov 20 14:56 gunicorn--dsmr_webinterface.socket=

supervisor> status
dsmr_backend                     RUNNING   pid 571, uptime 0:17:42
dsmr_datalogger                  RUNNING   pid 1293, uptime 0:00:10
dsmr_webinterface                RUNNING   pid 569, uptime 0:17:43
supervisor> tail -f dsmr_webinterface
==> Press Ctrl-C to exit <==
O] Worker exiting (pid: 1015)
[2017-11-20 15:07:09 +0100] [1102] [INFO] Booting worker with pid: 1102
[2017-11-20 15:08:10 +0100] [569] [CRITICAL] WORKER TIMEOUT (pid:1102)
[2017-11-20 15:08:10 +0100] [1102] [INFO] Worker exiting (pid: 1102)
[2017-11-20 15:08:11 +0100] [1118] [INFO] Booting worker with pid: 1118
[2017-11-20 15:09:12 +0100] [569] [CRITICAL] WORKER TIMEOUT (pid:1118)
[2017-11-20 15:09:13 +0100] [1137] [INFO] Booting worker with pid: 1137
[2017-11-20 15:10:13 +0100] [569] [CRITICAL] WORKER TIMEOUT (pid:1137)
[2017-11-20 15:10:14 +0100] [1137] [INFO] Worker exiting (pid: 1137)
[2017-11-20 15:10:15 +0100] [1228] [INFO] Booting worker with pid: 1228
[2017-11-20 15:11:15 +0100] [569] [CRITICAL] WORKER TIMEOUT (pid:1228)
[2017-11-20 15:11:16 +0100] [1228] [INFO] Worker exiting (pid: 1228)
[2017-11-20 15:11:17 +0100] [1252] [INFO] Booting worker with pid: 1252
[2017-11-20 15:12:17 +0100] [569] [CRITICAL] WORKER TIMEOUT (pid:1252)
[2017-11-20 15:12:18 +0100] [1283] [INFO] Booting worker with pid: 1283
[2017-11-20 15:13:19 +0100] [569] [CRITICAL] WORKER TIMEOUT (pid:1283)
[2017-11-20 15:13:19 +0100] [1283] [INFO] Worker exiting (pid: 1283)
[2017-11-20 15:13:20 +0100] [1300] [INFO] Booting worker with pid: 1300

django.log is leeg.

Wat gaat er fout en hoe kan ik dit oplossen?

dennissiemensma, 2017-11-20:
Op wat voor hardware draai je dit? En draaien er nog andere applicaties?

Mijn eerste gedachte is dat de applicatie niet genoeg tijd/kracht heeft om te reageren, hoewel dat vrij vreemd is.

Je zou ook nog kunnen kijken of de applicatie wel reageert wanneer je bijvoorbeeld de backend uitzet: `sudo supervisorctl stop dsmr_backend`

hunter-nl, 2017-11-21:
Ik draai het op een Pi Zero W. Maar blijkbaar is dat niet krachtig genoeg. Is er een mogelijkheid om wel data te lezen/op te slaan met Pi Zero W en dit elders op andere Pi3 de data te verwerken/te tonen in web?

helmo, 2017-11-21:
Ja, je kan dan enkel de datalogger op de pi zero zetten, en de rest elders installeren. Zie http://dsmr-reader.readthedocs.io/en/latest/api.html#api-v1-remote-datalogger

dennissiemensma, 2017-11-21:
Een raspberryPi Zero W is inderdaad veel te traag hiervoor. 

Zoals Helmo al aangeeft kun je hem waarschijnlijk wel gebruiken als datalogger en dan de applicatie op een krachtigere RaspberryPi of server of iets draaien.

Mocht je er niet uitkomen dan hoor ik het graag.

closed 2017-11-22 by dennissiemensma