dsmrreader/dsmr-reader #387: There are too many unprocessed telegrams.

casperleenheer, 2017-11-20:
Hi,

Ik heb sinds vorige week een nieuwe DSMR 5 meter. Deze meter genereert veel meer telegrams dan mijn oude meter. Zeker als ik vol op het stroom ga, krijg ik wel 2 telegrams per seconde. Bij gewoon gebruik zie ik eigenlijk altijd (na ongeveer 10 seconden)

There are too many unprocessed telegrams. 
Unprocessed telegrams: 51

Ik heb de datalogger op 10 seconden staan. Is dit iets om mij zorgen over te maken?

Groeten,

Casper

HugoDaBosss, 2017-11-20:
DSMR 5 geeft in principe elke seconde een telegram. Als je dus de consumption configuration op 10 seconden (per reading) hebt staan is dit eigenlijk 1 seconde. Deze tekst is (dacht ik) aangepast in de nieuwe release. 
Uit beschrijving van de unprocessed telegrams op de status pagina:

"This indicates how many stored telegrams are waiting to be processed. This should reset around each minute mark, depending on whether you've enabled grouping readings in the configuration. The stored telegrams are processed by the DSMR_BACKEND process and thus can be processed retroactively as well (by design)."

Bij mij zelf (ook DSMR 5) zie ik dat zodra ik de consumption configuration op per reading zet het aantal telegrammen blijft oplopen, ook boven de 60.

dennissiemensma, 2017-11-20:
Het hangt er vanaf of het aantal onverwerkte telegrammen blijft oplopen, na meerdere minuten. Je zou kunnen kijken of dat gebeurt en mogelijk staat er in de logfile van `dsmr_backend` meer informatie als dat gebeurt.

Als dit werkelijk issues veroorzaakt, zal ik kijken voor een mogelijke oplossing. Ik heb zelf een v4-meter, dus ik kan het sowieso niet thuis uitproberen/controleren, helaas.

pyrocumulus, 2017-11-21:
Dit doet hij bij mij ook al sinds het begin. Bij mij loopt de teller op tot een stuk of 60-70 voordat de backend de lijst weer helemaal opruimt. Daarna loopt de teller weer langzaam op. Verder niet problematisch maar de statuspagina wekt wel de indruk dat er een probleem is, dat er eigenlijk niet is :)

De grens waarboven het als 'error' gepresenteerd wordt zou een beetje hoger moeten liggen denk ik, verder is het puur een visueel probleempje.

HugoDaBosss, 2017-11-21:
Ik heb pull request #389 gemaakt die de grens verhoogt afhankelijk van de instelling voor het groeperen van metingen.

dennissiemensma, 2017-11-21:
Bedankt voor jullie input. Ik zal tevens naar de pull request kijken. Het zou zo kunnen dat de waarschuwing op de status-pagina inderdaad te strak afgesteld staat.

dennissiemensma, 2017-11-21:
Het is nog wat nazorg van de DSMR v5 ondersteuning. Er is een fix opgenomen voor de eerstvolgende release.

closed 2017-11-21 by dennissiemensma

dennissiemensma, 2017-11-24:
`v1.11` is [gereleased](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.11.0), inclusief dit ticket.