dsmrreader/dsmr-reader #390: Gas- en elektriciteitsverbruik vanaf start energie contract.

tpjanssen, 2017-11-21:
Is het een idee om het gas- en elektriciteitsverbruik (en terug levering) te kunnen monitoren vanaf de start van je huidige energiecontract?

dennissiemensma, 2017-11-21:
Wat bedoel je exact met monitoren? Doel je op het kunnen selecteren wat je verbruik was tijdens een specifieke periode?

tpjanssen, 2017-11-21:
Precies! Dus het verschil in meterstanden tussen nu en een zelf te bepalen datum in het verleden.

dennissiemensma, 2017-11-21:
Oke duidelijk, lijkt me een goede toevoeging. Ik zet hem op de backlog.

dennissiemensma, 2018-06-08:
Ik heb een opzet gemaakt voor een aparte pagina waar **alle contracten** getoond worden, inclusief de prijzen en het verbruik binnen die periode. Volgens mij is dat een mooie toevoeging, nog iets uitgebreider dan de wens in dit ticket. Dat hoeven we ook niet zelf datums gaan selecteren.

Hieronder twee screenshots met wat testdata voor een voorbeeld:
[image]

[image]

dennissiemensma, 2018-06-08:
Gemerged naar `development` voor volgende release.

closed 2018-06-08 by dennissiemensma

tpjanssen, 2018-06-11:
Mooi! Enig idee wanneer de volgende release komt?

dennissiemensma, 2018-06-12:
Bij deze in [v1.19](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.19.0).