dsmrreader/dsmr-reader #391: DSMR_datalogger issue

hunter-nl, 2017-11-25:
Ik heb een DSMR 5.0 en die zou elke seconde een telegram leveren (en dit werkt ook goed als ik rechtstreeks de serial poort uitlees met CU commando.
Echter met DSMR_datalogger krijg ik maar slechts eens in de 1 a 3 minuten een telegram.
Dit is de logging van de DSMR_datalogger:
```Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
/XMX5LG....

1-3:0.2.8(50)
0-0:1.0.0(171125180617W)
0-0:96.1.1(serialnummer)
1-0:1.8.1(000215.656*kWh)
1-0:1.8.2(000236.131*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(01.035*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00002)
0-0:96.7.9(00000)
1-0:99.97.0(0)(0-0:96.7.19)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(233.0*V)
1-0:31.7.0(005*A)
1-0:21.7.0(01.035*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(serialnummer)
0-1:24.2.1(171125180509W)(00062.813*m3)
!559B
Command completed. Sleeping for 0.25 second(s)...

Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
/XMX5LG.....

1-3:0.2.8(50)
0-0:1.0.0(171125180747W)
0-0:96.1.1(serialnummer)
1-0:1.8.1(000215.686*kWh)
1-0:1.8.2(000236.131*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(01.166*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00002)
0-0:96.7.9(00000)
1-0:99.97.0(0)(0-0:96.7.19)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(232.0*V)
1-0:31.7.0(005*A)
1-0:21.7.0(01.166*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(serialnummer)
0-1:24.2.1(171125180509W)(00062.813*m3)
!2643
Command completed. Sleeping for 0.25 second(s)...

Starting infinite command loop...
```
Hoe kan dit verbeterd worden zodat het wel welke telegram verwerkt?

hunter-nl, 2017-11-26:
Ik heb de oorzaak gevonden. Op de Raspberry pi 3, moet de bluetooth uitgeschakeld worden omdat deze ook de serial poort gebruikt.
De volgende twee regels dienen in de /boot/config.txt opgenomen te worden
```
enable_uart=1
dtoverlay=pi3-disable-bt
```
Na reboot werkt de serial poort /dev/ttyS0 (of /dev/serial0) perfect.

closed 2017-11-26 by hunter-nl

dennissiemensma, 2017-11-26:
Bedankt voor je melding en tevens het terugkoppelen van je gevonden oplossing. Mocht iemand anders hier tegenaan lopen, dan is bovenstaande goed om te weten!