dsmrreader/dsmr-reader #393: restart dsmr_backend

HFeenstra, 2017-11-30:
Hoi, eind oktober 2017 en nu eind november 2017 opnieuw zie ik dat de dsmr_backend vast loopt.
Ik heb 1.11 geïnstalleerd maar het valt me op dat ik nu opnieuw deze moet herstarten.
Zie verder geen enkele foutmelding op de frontpage van de pagina, dus daar kan ik ook niets terugvinden.
Heb vervolgens als: `sudo supervisorctl` en `restart dsmr_backend` het weer aan de praat gekregen. 

Hebben meer hier last van? 1.9 en daarvoor, gebeurde dit nooit bij mij....

dennissiemensma, 2017-12-01:
Wat geeft de logfile aan?

HFeenstra, 2017-12-06:
Zou dit graag delen, maar weet niet hoe ik deze te voorschijn kan krijgen. Hoe deze uit te lezen? Middels welk programma?

dennissiemensma, 2017-12-06:
Zie http://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

Je kunt doorgaans `view LOGFILE`, `tail -n 100 LOGFILE` of `cat LOGFILE` gebruiken op de commandline. Of `tail -f LOGFILE` om de logfile te monitoren terwijl die gevuld wordt. Waarbij LOGFILE de betreffende log is.

dennissiemensma, 2017-12-12:
Ben je nog iets wijzer geworden uit de logs?

closed 2018-01-07 by dennissiemensma