dsmrreader/dsmr-reader #396: Enkel tarief elektriciteit: gecombineerd tarief tonen op 'Statistieken'-pagina

jorisvervuurt, 2017-12-07:
Wij maken gebruik van een enkel tarief voor de verbruikte elektriciteit. In de interface-configuratie heb ik 'Voeg tarieven samen' aangevinkt en bij de tarieven van de energieleverancier hetzelfde bedrag ingevuld voor zowel dal- als piektarief.

Op de 'Dashboard'-pagina wordt netjes één tarief 'Elektriciteit (enkel tarief)' getoond.
Op de 'Statistieken'-pagina worden echter twee losse tarieven getoond.

Als ik het goed begrijp, registreert de meter - ongeacht of je voor enkel tarief kiest - twee losse tarieven. Het is dus wel interessant om op de 'Statistieken'-pagina de twee gescheiden tarieven te kunnen zien. Bij het handmatig doorgeven van de meterstanden, moet ik de twee losse tarieven bij elkaar optellen.

Is het mogelijk om op de 'Statistieken'-pagina naast de twee losse tarieven, ook een gecombineerd tarief te tonen? Simpelweg een optelling van de geregistreerde dal- en piektarieven dus. ;-)

dennissiemensma, 2017-12-07:
Bedankt voor je verzoek. Als ik het goed begrijp doel je dus op het eerste blokje op die pagina, betreffende de meterstanden? En die wil je bij elkaar opgeteld hebben?

jorisvervuurt, 2017-12-07:
Dat is inderdaad wat ik bedoel. 😉 

Nu ziet het er zo uit:
[image]

Zoiets lijkt mij ideaal:
[image]

dennissiemensma, 2017-12-07:
Oke bedankt voor de opheldering, duidelijk!

dennissiemensma, 2018-01-16:
Gaat mee in volgende release.

closed 2018-01-16 by dennissiemensma

jorisvervuurt, 2018-01-17:
Top!

dennissiemensma, 2018-01-23:
De betreffende release is zojuist uitgebracht.