dsmrreader/dsmr-reader #397: Updaten mogelijk maken vanuit webinterface

pyrocumulus, 2017-12-08:
Het zou fijn zijn om DSMR te kunnen updaten vanuit de webinterface zelf. Het is uiteraard puur een wens uit gemak, geen vereiste, maar het verlaagt de drempel van het gebruik wel weer wat. Ik merk dat ik nu regelmatig pas later een update doe omdat ik even een moment moet vinden om via SSH in te loggen en de boel te starten.

dennissiemensma, 2017-12-09:
Bedankt voor je suggestie! Ik heb dit al een keer eerder geprobeerd te maken, maar kreeg het niet helemaal lekker werkend.

Het issue is namelijk dat ik niet kan garanderen dat alle (toekomstige) databasemigraties, die met een deploy meekomen, voltooien binnen de timeout van de webinterface. Daarom is het updaten alleen mogelijk via SSH.

closed 2017-12-09 by dennissiemensma

pyrocumulus, 2017-12-09:
Dat maakt het inderdaad lastiger, ik snap je punt.

De meeste andere webapps met een soortgelijk probleem (potentieel langdurige upgrades) lossen dat op door zo'n tijdelijke auto-refresh pagina te tonen tijdens de update. Dus niet te wachten op die enkele klik op 'Update' en dan pas te refreshen als alles klaar is. Ik weet niet exact hoe je het noemt noch hoe je het doet, maar bijvoorbeeld mijn Synology laat keurig een pagina tot wel 10 minuten lang zien, totdat de update op de achtergrond klaar is. Deze ververst dan uiteindelijk ook nog eens automatisch. Ik vermoed dat het een pagina is met een hoop client-side code.

Ik weet dat je hem gesloten hebt maar wilde bovenstaande toch nog even meegeven :-)

dennissiemensma, 2017-12-09:
Ik begrijp waarop je doelt, maar ik weet helaas niet hoe zij dat exact doen. Ik zou het uiteindelijk nog wel via het `dsmr_backend`-proces kunnen doen, maar dat vind ik zelf relatief veel complexiteit voor zo'n feature. 

Daarnaast zijn er niet heel frequent nieuwe releases, dus dat is voornamelijk mijn afweging op dit moment. :]