dsmrreader/dsmr-reader #399: Veel calls naar api.buienradar

dbmnl, 2017-12-10:
Hoi Dennis,

Sinds deze week heeft Pi-Hole een functie om verder terug in het verleden te kijken in plaats van slechts 24 uur geleden. Toen ik echter ging kijken voor de afgelopen dagen vielen mij 2 spikes op.
[image]

Bij nader onderzoek bleken dit calls naar api.buienradar te zijn. Het enige wat een call doet hier naartoe is dsmrreader. Maar het aantal is aan de hoge kant en wel heel vaak achter elkaar.
[image]

Vermoedelijk zal dit komen doordat api.buienradar niet beschikbaar is, maar het is natuurlijk onnodig om dan 1000 calls te doen :)

Kan jij hier eens naar kijken of mij laten weten of je nog meer informatie nodig hebt. Ik zat al te kijken in de logs directory, maar zie dat de dsmrreader.log voor het laatst op 19 oktober is bijgewerkt. In de logfile van django staat niets relevants. En bij supervisor logs kan ik ook geen relevante informatie vinden of foutmeldingen.

dennissiemensma, 2017-12-10:
Bedankt voor het melden. Ik zal er foutafhandeling in bouwen zodat die het pas na 5 minuten opnieuw probeert.

dennissiemensma, 2017-12-11:
Ik heb er een mogelijke fix voor gemaakt voor de volgende release.

closed 2017-12-11 by dennissiemensma

dennissiemensma, 2018-01-14:
Zojuist uitgebracht in [release v1.12](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.12.0).