dsmrreader/dsmr-reader #400: Postgres installatie

jicho, 2017-12-10:
Hi,

Onlangs heb ik DSMR-reader geïnstalleerd volgens de handleiding. Software werkt als een trein, maar Postgres... 'k Krijg er hoofdpijn van :(

Heeft iemand, heel toevallig, een uitleg hoe je Postgres het beste kan installeren zodat je geen data kwijt bent bij een RasPi herstart? Out-of-the-box is Postgres alles behalve robust, MySQL is dan toch een stuk aardiger bij een spontane herstart en/of een reboot commando.

Heb een poging gedaan om op Google te vinden, maar niets nuttigs kunnen vinden voor iemand zonder Postgres ervaring die een DSMR-reader database wil hebben die tegen een stootje kan.

Besef me dat dit misschien de beste plek is om deze vraag te stellen. Heb hem hier echter neergezet voor mogelijk andere DSMR-reader gebruikers zonder Postgres kennis :)

dennissiemensma, 2017-12-10:
Bedankt voor je vraag. PostgreSQL zou juist robuust moeten zijn t.o.v. MySQL, helemaal op gebied van dataintegriteit. Ik ga er vanuit dat een standaardinstallatie van PostgreSQL afdoende ACID is, net zoals MySQL dat is met InnoDB.

* Kun je aangeven wat je exact ervaart? Hoeveel data raak je kwijt bij een reboot? (een aantal minuten of uren)
*  Heb je al gekeken in de logfiles van PostgreSQL of daar een foutmelding in staat?

jicho, 2017-12-10:
Wanneer ik mijn pi herstart via "reboot" ben ik mijn data kwijt. Vanavond was ik na de herstart 7 dagen gegevens kwijt. Zojuist heb ik nog eens een reboot gedaan, op dat moment was ik alle data kwijt dit sinds de laatste herstart van een paar uur geleden.

Heb gekeken, n.a.v. de vraag, of ik de log's kan vinden, maar helaas... Volgens "show log_directory ;" zou ik de directory moeten kunnen vinden. 'k Krijg echter alleen "pg_log" terug als opslag. Na wat zoeken heb ik de hele directory nergens terug kunnen vinden.

dennissiemensma, 2017-12-11:
Oke, dat klinkt niet heel goed. Voor de test heb ik m'n raspberry hier een (clean) reboot gegeven. Alle data is bewaard gebleven. 
Ik heb hier nog nooit iemand hier over gehoord qua dataverlies, afgezien van een fysiek corrupt SD-kaartje. Ik vermoed dat er specifiek is anders is in jouw situatie. 

Logfiles zouden doorgaans in `/var/log/` moeten staan. Bij mij staan ze (voor 9.4) in: `/var/log/postgresql/postgresql-9.4-main.log` en daar staat de reboot gewoon in:

```
2017-12-11 14:44:52 UTC [28322-2] LOG:  received fast shutdown request
2017-12-11 14:44:52 UTC [28322-3] LOG:  aborting any active transactions
2017-12-11 14:44:52 UTC [20243-1] dsmrreader@dsmrreader FATAL:  terminating connection due to administrator command
2017-12-11 14:44:52 UTC [1424-1] dsmrreader@dsmrreader FATAL:  terminating connection due to administrator command
2017-12-11 14:44:52 UTC [28327-2] LOG:  autovacuum launcher shutting down
2017-12-11 14:44:52 UTC [28324-1] LOG:  shutting down
2017-12-11 14:44:52 UTC [28324-2] LOG:  database system is shut down
2017-12-11 14:45:06 UTC [642-1] LOG:  database system was shut down at 2017-12-11 14:44:52 UTC
2017-12-11 14:45:06 UTC [643-1] [unknown]@[unknown] LOG:  incomplete startup packet
2017-12-11 14:45:06 UTC [642-2] LOG:  MultiXact member wraparound protections are now enabled
2017-12-11 14:45:06 UTC [641-1] LOG:  database system is ready to accept connections
2017-12-11 14:45:06 UTC [647-1] LOG:  autovacuum launcher started
```

Heb jij die logfile wel? En staat er iets in?

Verder zou je kunnen kijken of de datadir wel bestaat. Voor mijn situatie is dat: `du -msc /var/lib/postgresql/9.4/main/`
```
1512	/var/lib/postgresql/9.4/main/
```

Tot slot zou je kunnen kijken of je SD-kaart niet read-only gemount is (ik had dan overigens wel verwacht dat je database uberhaupt niets opslaat): 
`mount`

Ik heb alles standaard, dus ik zoek de root en `mount` geeft: 
`/dev/mmcblk0p2 on / type ext4 (rw,noatime,data=ordered)`
Het vlaggetje `rw` aan het eind geeft aan dat die read/write is.

jicho, 2017-12-11:
Hi, de log files heb ik over het hoofd gezien :( De folder is wel aanwezig, er wordt echter niets in weggeschreven na 2 december... Daarnaast staat postgresql 9.6 op mijn Pi.

Dit is het laatste wat er in de log staat:

```
2017-12-02 22:17:05.507 CET [413] LOG:  database system was interrupted; last known up at 2017-12-02 22:22:59 CET
2017-12-02 22:17:05.653 CET [415] [unknown]@[unknown] LOG:  incomplete startup packet
2017-12-02 22:17:05.704 CET [413] LOG:  database system was not properly shut down; automatic recovery in progress
2017-12-02 22:17:05.715 CET [413] LOG:  redo starts at 0/8DAD3E50
2017-12-02 22:17:05.783 CET [413] LOG:  invalid record length at 0/8DAFFC88: wanted 24, got 0
2017-12-02 22:17:05.783 CET [413] LOG:  redo done at 0/8DAFFC60
2017-12-02 22:17:05.783 CET [413] LOG:  last completed transaction was at log time 2017-12-02 22:25:07.066307+01
2017-12-02 22:17:05.840 CET [413] LOG:  MultiXact member wraparound protections are now enabled
2017-12-02 22:17:05.861 CET [420] LOG:  autovacuum launcher started
2017-12-02 22:17:05.861 CET [410] LOG:  database system is ready to accept connections
```

Bij een reboot (zojuist gedaan) komt er niets meer bij (en wederom alles foetsie van vandaag ;) De rechten op /var/log/postgreql:

```
drwxrwxr-t  2 root        postgres     4096 Nov 13 06:25 postgresql
```

En van de log-files:

```
pi@piweb:/var/log/postgresql $ ls -la
total 20
drwxrwxr-t  2 root     postgres 4096 Nov 13 06:25 .
drwxr-xr-x 12 root     root     4096 Dec  2 22:17 ..
-rw-r-----  1 postgres adm       903 Dec  2 22:17 postgresql-9.6-main.log
-rw-r-----  1 postgres adm       720 Nov 13 06:25 postgresql-9.6-main.log.1
-rw-r-----  1 postgres adm       874 Nov  5 06:25 postgresql-9.6-main.log.2.gz
```

Mijn Postgresl installatie was ook standaard, heb de stappen gevolgd die in de handleiding stonden. Heb onlangs alleen wel een update voorbij zien komen die geïnstalleerd is (9.4 -> 9.6). Het lijkt er een beetje op dat daar iets niet goed is gegaan.

Het systeem is niet readonly:

```
pi@piweb:/var/log/postgresql $ mount
/dev/mmcblk0p2 on / type ext4 (rw,noatime,data=ordered)
```

Het begint er een beetje op te lijken dat er iets niet goed is gegaan bij de upgrade van de postgresql. Vermoed dat deze upgrade rond 2 december plaats heeft gevonden aangezien toen de wereld gestopt is met normaal werken (aldus de datum van de bestanden)

Grappig en irritant genoeg... Als ik de log files verwijder, dan zijn dezelfde bestanden een herstart later weer terug...

Volgens mij moet ik maar eens gaan nadenken over een herinstallatie van postgresql en een schone start. De auto-dropbox-backup (die prima werkt) kan misschien nog een beetje hoop bieden, mits die export geen corrupte data bevat. Anders bij dropbox maar in de versie geschiedenis duiken.

Tot nu toe zijn postgresql en ik geen vrienden. Ken helaas nog iemand die een soortgelijke ervaring heeft. Niet i.c.m. DSMR-reader, maar wel op een hosting server. Daar zorgde een stroomstoring simulatie ervoor dat de hele postgresql database corrupt raakte. Dit was een standaard installatie. Mijn voordeel... Deze data is leuk om te hebben, maar kan gemist worden aangezien ik pas begonnen ben met verzamelen.

'k Ga nog wel verder spelen, mocht er wat bovenwater komen, dan meld ik het!

jicho, 2017-12-11:
Kleine update.

Ondanks dat mount een rw laat zien valt de SD kaart in zijn geheel niet beschrijven. Heb hem zojuist uit mijn RasPi gehaald en een poging gedaan om in /boot te schrijven. Zodra de SD kaart opnieuw gelezen wordt zijn alle wijzigingen foetsie.

De normale weg lijkt om rw aan de cmdline.txt toe te voegen als het alleen "mis" gaat op de RasPi, nu heb ik echter ook een read only modus op een andere computer i.c.m. de SD kaart.

Sommigen hebben last van de read only actie op het moment dat en een distributie upgrade plaats vindt. Die heb ik ook uitgevoerd, heb er alleen op een later moment last van gekregen.

Alternatief op het thema is dat de SD kaart corrupt is geraakt en daardoor in read only mode terecht is gekomen.

Na goed, snappen doe ik het niet en meer zin om het uit te zoeken heb ik ook niet. 'k Ga maar eens een format proberen en zet alles opnieuw erop.

dennissiemensma, 2017-12-11:
Bedankt voor je uitgebreide updates. Mooi dat dit een en ander verklaart. Persoonlijk raad ik je aan om in dit geval simpelweg een andere SD-kaart te gebruiken, van dit soort kuren wordt je inderdaad echt niet blij.

jicho, 2017-12-13:
Graag gedaan. De nieuwe SD-kaart is inmiddels binnen, nu nog even een momentje vinden op hem weer in te richten ;)

Bedankt voor het meedenken en de tips!

closed 2017-12-13 by jicho

dennissiemensma, 2017-12-13:
Top, succes :]