dsmrreader/dsmr-reader #401: Restore PostgresDB

allureGOT, 2017-12-12:
Hallo Dennis,

Ik heb vanavond een raspberry pi vervangen met een intel NUC.
Op de Pi heb ik een DB dump gemaakt van de DB (300MB groot).

Ik heb de installatie handleiding gevolgd en de DB proberen te restoren op de NUC, echter loop ik direct na het restoren tegen de onderstaande melding aan.

Kun je me vertellen hoe ik deze dubbele waarde verwijder?

[image]

dennissiemensma, 2017-12-12:
Kan het zijn dat je bij de installatie stap 8 ook hebt uitgevoerd? De `./manage.py migrate`, voordat je de database hebt teruggezet?

En anders zul je de dubbele waarde handmatig moeten verwijderen, maar ik weet niet of het dan al op meer plekken is misgegaan eigenlijk. Bekijken kan met:

* `sudo su - postgres`
* `psql dsmrreader`
* `select * from dsmr_datalogger_dataloggersettings;`

Je krijgt dan zoiets te zien, maar dan twee records vermoed ik:
```
dsmrreader=# select * from dsmr_datalogger_dataloggersettings;
 id | track |   com_port   | dsmr_version | track_phases | verify_telegram_crc 
----+-------+--------------+--------------+--------------+---------------------
  1 | t     | /dev/ttyUSB0 |            4 | t            | t
(1 row)
```

dennissiemensma, 2017-12-12:
Het beste is overigens om de database te wipen en dan een backup terug te zetten. Dat kan ook na de installatie nog. Dan weet je zeker dat je een schone 1-op-1 kopie hebt gebruikt.

dennissiemensma, 2017-12-12:
Dat laatste kan overigens met:
* `sudo supervisorctl stop all`
* `sudo su - postgres`
* `dropdb dsmrreader ; createdb -O dsmrreader dsmrreader ; psql -f PAD-NAAR-JE-BACKUP.sql dsmrreader`
** Of zie de instructies in de docs
* `sudo supervisorctl start all`

allureGOT, 2017-12-12:
Dat is wat ik gedaan had, echter heb ik toch ergens een steekje laten vallen.
Na het handmatig verwijderen van de entry loopt het mechanisme weer met de oude data.

Wederom, heel erg bedankt, ik ga morgen even kijken of hij de huidige data weer netjes bijschrijft in de DB.

allureGOT, 2017-12-13:
Probleem opgelost! :)

dennissiemensma, 2017-12-13:
Top, fijn om te horen. Bedankt voor de terugkoppeling

closed 2017-12-13 by dennissiemensma