dsmrreader/dsmr-reader #402: Lost django password

RemcoLeonard, 2017-12-13:
Is there a way to change the username/password for the configuration page?
I lost them.....

jeroenpeters1986, 2017-12-13:
Hi @RemcoLeonard, if you want; you can create a new admin user by following this instruction https://dsmr-reader.readthedocs.io/en/latest/installation.html?highlight=createsuperuser#bootstrapping, starting from 'Create an application superuser.'

dennissiemensma, 2017-12-13:
See the lnik Jeroen wrote above, it should help you. Let us know if you still have trouble creating a superuser account.

closed 2017-12-13 by dennissiemensma

VincentSC, 2020-04-22:
For new version it's possible to change the password of the default user 'dsmr'. See https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html and search for "changepassword".

Log in as user `dsmr`, go to the directory `dsmr-reader` and run:
> dsmr@raspberrypi:~/dsmr-reader $ `./manage.py changepassword admin`