dsmrreader/dsmr-reader #403: mislukte upgrade naar 1.10

mkruiver, 2017-12-16:
Na het upgraden naar 1.10 is mijn DSMR installatie helaas kaduuk.
Onder versie 1.9 is de adapter van mijn pi een keer kapot gegaan, maar na het vervangen van de adapter draaide alles weer.
Ik heb toen met deploy.sh geprobeerd te upgraden, maar dat lukte helaas niet.

Als ik naar mijn webportal surf, verteld Nginx mij het volgende:

> 502 Bad Gateway
> 
> nginx/1.6.2

Als ik nu deploy krijg ik het volgende te zien

> ./deploy.sh
> 
> 
>  --- You are currently running version: 
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
> AttributeError: 'module' object has no attribute '__version__'
> 
>      [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)
> 
> [!] Halted in pre-deployment: Either user ABORTED or script FAILED

./pre-deploy.sh 

> 
>  --- You are currently running version: 
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
> AttributeError: 'module' object has no attribute '__version__'
> 
>      [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)
> 
> 

In Supervisor zie ik dat de processen steeds gekilled worden.

> upervisor> status
> dsmr_backend                     STARTING   
> dsmr_datalogger                  RUNNING    pid 12232, uptime 0:00:02
> dsmr_webinterface                RUNNING    pid 12225, uptime 0:00:04
> supervisor> status
> dsmr_backend                     STARTING   
> dsmr_datalogger                  RUNNING    pid 12239, uptime 0:00:02
> dsmr_webinterface                STARTING   
> supervisor> status
> dsmr_backend                     RUNNING    pid 12243, uptime 0:00:01
> dsmr_datalogger                  RUNNING    pid 12239, uptime 0:00:03
> dsmr_webinterface                RUNNING    pid 12236, uptime 0:00:04
> supervisor> status
> dsmr_backend                     RUNNING    pid 12243, uptime 0:00:02
> dsmr_datalogger                  STARTING   
> dsmr_webinterface                STARTING   
> supervisor> status
> dsmr_backend                     RUNNING    pid 12243, uptime 0:00:03
> dsmr_datalogger                  STARTING   
> dsmr_webinterface                STARTING   
> supervisor> status
> dsmr_backend                     STARTING   
> dsmr_datalogger                  RUNNING    pid 12246, uptime 0:00:02
> dsmr_webinterface                STARTING   
> supervisor> status
> dsmr_backend                     STARTING   
> dsmr_datalogger                  RUNNING    pid 12246, uptime 0:00:02
> dsmr_webinterface                STARTING   
> supervisor> status
> dsmr_backend                     RUNNING    pid 12252, uptime 0:00:01
> dsmr_datalogger                  RUNNING    pid 12246, uptime 0:00:03
> dsmr_webinterface                STARTING   
> supervisor> status
> dsmr_backend                     RUNNING    pid 12252, uptime 0:00:01
> dsmr_datalogger                  RUNNING    pid 12246, uptime 0:00:03
> dsmr_webinterface                RUNNING    pid 12245, uptime 0:00:03
> supervisor> status
> dsmr_backend                     RUNNING    pid 12252, uptime 0:00:02
> dsmr_datalogger                  RUNNING    pid 12246, uptime 0:00:04
> dsmr_webinterface                RUNNING    pid 12245, uptime 0:00:04
> supervisor> status
> dsmr_backend                     RUNNING    pid 12252, uptime 0:00:02
> dsmr_datalogger                  STARTING   
> dsmr_webinterface                RUNNING    pid 12245, uptime 0:00:04
> supervisor> status

Zou je me willen helpen met troubleshooten / restoren?
Mijn files zijn gewoon nog toegangkelijk dus ik zou een database-backup terug kunnen zetten. Ik vroeg mij trouwens af of het ook mogelijk is om settings terug te zetten van een 'oude' DSMR installatie?

dennissiemensma, 2017-12-17:
Zou je eens willen kijken of de `virtualenv` nog goed is? De foute lijken erop dat Django niet gevonden kan worden en die is geinstalleerd in een aparte omgeving. Als het goed is is er een mapje: `/home/dsmr/.virtualenvs/`.

Wat krijg je te zien als je dit doet?
* `sudo su - dsmr`
* `python -c 'import dsmrreader ; print(dsmrreader.__version__)'`

dennissiemensma, 2017-12-17:
Je zou ook de applicatie handmatig kunnen draaien ,om te zien of die een specifieke fout geeft.

* Stop eerst dsmr_webinterface in Supervisor (ander krijg je sowieso een foutmelding): `sudo supervisorctl stop dsmr_webinterface`.
* `sudo su - dsmr`
* `/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi`

dennissiemensma, 2017-12-17:
En als antwoord op je vraag of je settings terug kan zetten van oude installatie. Dat kan zeker, maar dan zul je moeten graven in de database-dump zelf. Vermoedelijk op het woordje `setting`. Dan kom je alle opties tegen. Ze staan helaas niet op een aparte/overzichtelijke plek opgeslagen, ze zijn onderdeel van de database.

dennissiemensma, 2017-12-20:
@mkruiver ben je verder gekomen met de oplossing?

mkruiver, 2017-12-22:
Dan krijg ik

pi@raspberrypimeterkast:~ $ sudo su dsmr
(dsmrreader)dsmr@raspberrypimeterkast:~/dsmr-reader $ python -c 'imrt
dsmrreader  ; print(dsmrreader.__version__)'
Traceback (most recent call last):
 File "<string>", line 1, in <module>
AttributeError: 'module' object has no attribute '__version__'
(dsmrreader)dsmr@raspberrypimeterkast:~/dsmr-reader $

mkruiver, 2017-12-22:
Dan krijg ik

(dsmrreader)dsmr@raspberrypimeterkast:~/dsmr-reader $
(dsmrreader)dsmr@raspberrypimeterkast:~/dsmr-reader $
/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests
500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid
/var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
[2017-12-22 18:31:15 +0100] [3296] [INFO] Starting gunicorn 19.7.1
[2017-12-22 18:31:15 +0100] [3296] [INFO] Listening at:
unix:/var/tmp/gunicorn--dsmr_webinterface.socket (3296)
[2017-12-22 18:31:15 +0100] [3296] [INFO] Using worker: sync
[2017-12-22 18:31:15 +0100] [3300] [INFO] Booting worker with pid: 3300
[2017-12-22 18:31:18 +0100] [3300] [ERROR] Exception in worker process
Traceback (most recent call last):
 File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/
arbiter.py", line 578, in spawn_worker
   worker.init_process()
 File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/workers/
base.py", line 126, in init_process
   self.load_wsgi()
 File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/workers/
base.py", line 135, in load_wsgi
   self.wsgi = self.app.wsgi()
 File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/app/
base.py", line 67, in wsgi
   self.callable = self.load()
 File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/app/
wsgiapp.py", line 65, in load
   return self.load_wsgiapp()
 File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/app/
wsgiapp.py", line 52, in load_wsgiapp
   return util.import_app(self.app_uri)
 File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/gunicorn/
util.py", line 352, in import_app
   __import__(module)
 File "/home/dsmr/dsmr-reader/dsmrreader/wsgi.py", line 16, in <module>
   application = get_wsgi_application()
 File
"/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/
wsgi.py", line 13, in get_wsgi_application
   django.setup(set_prefix=False)

mkruiver, 2017-12-22:
File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packa
                         ges/django/__init__.py", line 22, in setup
   configure_logging(settings.LOGGING_CONFIG, settings.LOGGING)
 File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packa
                         ges/django/conf/__init__.py", line 56, in
__getattr__
   self._setup(name)
 File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packa
                         ges/django/conf/__init__.py", line 41, in _setup
   self._wrapped = Settings(settings_module)
 File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packa
                         ges/django/conf/__init__.py", line 110, in __init__
   mod = importlib.import_module(self.SETTINGS_MODULE)
 File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/importlib/
                         __init__.py", line 109, in import_module
   return _bootstrap._gcd_import(name[level:], package, level)
 File "/home/dsmr/dsmr-reader/dsmrreader/settings.py", line 1, in
                        <module>
   from dsmrreader.config.production import *
 File "/home/dsmr/dsmr-reader/dsmrreader/config/production.py", li
                         ne 6, in <module>
   from dsmrreader.config.base import *
 File "/home/dsmr/dsmr-reader/dsmrreader/config/base.py", line 195
                         , in <module>
   DSMRREADER_VERSION = dsmrreader.__version__
AttributeError: 'module' object has no attribute '__version__'
[2017-12-22 18:31:18 +0100] [3300] [INFO] Worker exiting (pid: 3300
                         )
[2017-12-22 18:31:18 +0100] [3296] [INFO] Shutting down: Master
[2017-12-22 18:31:18 +0100] [3296] [INFO] Reason: Worker failed to
                          boot.

mkruiver, 2017-12-22:
Hoe kan ik in de database grasduinen? En bedoel je dat ik de settings in
plaintext kan vinden en vervolgens moet overnemen in de nieuwe installatie?

Ik vind deze top output trouwens gek, python3 gebruikt alle resources.

top - 18:34:41 up  4:29,  3 users,  load average: 1.45, 1.82, 2.18
Tasks: 156 total,   3 running, 153 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.5 us,  1.3 sy, 31.6 ni, 66.6 id,  0.0 wa,  0.0 hi,  0.0 si,
 0.0 st
KiB Mem:    947736 total,   356280 used,   591456 free,    45380 buffers
KiB Swap:   102396 total,        0 used,   102396 free.   186232 cached Mem

 PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
3560 dsmr      25   5   25928  21004   6836 <210046836> R  62.6  2.2   0
<626220>:01.89 python3
3562 dsmr      30  10   17272  12704   6656 <127046656> S  14.6  1.3   0
<146130>:00.44 python3
   7 root      20   0       0      0      0 S   0.3  0.0   0 <03000>:52.89
rcu_sched
 254 root      20   0       0      0      0 S   0.3  0.0   0 <03000>:00.98
kworker/2:3
 793 root      20   0   16020   9716   3876 <97163876> S   0.3  1.0   3
<03103>:29.24 supervisord
 916 pi        20   0    3700    220 <2003700220>     16 S   0.3  0.0   0
<03000>:00.12 ssh-agent
3559 pi        20   0    5216   2516   2096 R   0.3  0.3   0 <03030>:00.04
top
   1 root      20   0   22876   4008   2744 <40082744> S   0.0  0.4   0
<00040>:03.99 systemd
   2 root      20   0       0      0      0 S   0.0  0.0   0 <00000>:00.00
kthreadd
   3 root      20   0       0      0      0 S   0.0  0.0   0 <00000>:00.49
ksoftirqd/0
   5 root       0 -20       0      0      0 S   0.0  0.0   0 <00000>:00.00
kworker/0:0H
   8 root      20   0       0      0      0 S   0.0  0.0   0 <00000>:00.00
rcu_bh
   9 root      rt   0       0      0      0 S   0.0  0.0   0 <00000>:00.71
migration/0
  10 root      rt   0       0      0      0 S   0.0  0.0   0 <00000>:00.73
migration/1
  11 root      20   0       0      0      0 S   0.0  0.0   0 <00000>:00.37
ksoftirqd/1
  13 root       0 -20       0      0      0 S   0.0  0.0   0 <00000>:00.00
kworker/1:0H
  14 root      rt   0       0      0      0 S   0.0  0.0   0 <00000>:00.72
migration/2
  15 root      20   0       0      0      0 S   0.0  0.0   0 <00000>:00.39
ksoftirqd/2
  17 root       0 -20       0      0      0 S   0.0  0.0   0 <00000>:00.00
kworker/2:0H
  18 root      rt   0       0      0      0 S   0.0  0.0   0 <00000>:00.71
migration/3

dennissiemensma, 2017-12-22:
Ik denk dat python3 veel resources gebruikt omdat hij constant start, crasht en opnieuw probeert.

Het lijkt erop dat je virtualenv niet meer helemaal lekker is. Je kunt die het beste verwijderen, aanmaken en opnieuw daarin installeren.

* sudo su - dsmr
* rm -rf ~/.virtualenvs
* mkdir ~/.virtualenvs
* virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3
* source ~/.virtualenvs/dsmrreader/bin/activate
* pip3 install -r dsmrreader/provisioning/requirements/base.txt -r dsmrreader/provisioning/requirements/postgresql.txt

Dan zou Django en alles weer goed geinstalleerd moeten zijn.

mkruiver, 2017-12-22:
Dennis bedankt voor de snelle support, het lijkt niet helemaal goed te gaan
helaas.

  Running setup.py install for coreschema

  Running setup.py install for itypes

*** Error compiling
'/tmp/pip-build-ifnpctmw/jinja2/jinja2/asyncfilters.py'...
  File "/tmp/pip-build-ifnpctmw/jinja2/jinja2/asyncfilters.py", line 7
    async def auto_to_seq(value):
            ^
SyntaxError: invalid syntax

*** Error compiling
'/tmp/pip-build-ifnpctmw/jinja2/jinja2/asyncsupport.py'...
  File "/tmp/pip-build-ifnpctmw/jinja2/jinja2/asyncsupport.py", line 22
    async def concat_async(async_gen):
            ^
SyntaxError: invalid syntax

  Running setup.py install for MarkupSafe

    building 'markupsafe._speedups' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall
-Wstrict-prototypes -g -fstack-protector-strong -Wformat
-Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.4m
-I/home/dsmr/.virtualenvs/dsmrreader/include/python3.4m -c
markupsafe/_speedups.c -o build/temp.linux-armv7l-3.4/markupsafe/_speedups.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1
-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g
-fstack-protector-strong -Wformat -Werror=format-security
-D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.4/markupsafe/_speedups.o -o
build/lib.linux-armv7l-3.4/markupsafe/_speedups.cpython-34m.so
Successfully installed crcmod coreapi django django-colorfield
django-debug-toolbar django-filter djangorestframework django-solo dropbox
gunicorn paho-mqtt pyserial python-dateutil pytz raven requests urllib3
psycopg2 uritemplate coreschema itypes sqlparse six idna chardet certifi
jinja2 MarkupSafe
Cleaning up...
(dsmrreader)dsmr@raspberrypimeterkast:~/dsmr-reader $ sudo supervisorctl
status

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for dsmr:
(dsmrreader)dsmr@raspberrypimeterkast:~/dsmr-reader $ exit
logout
pi@raspberrypimeterkast:~ $ sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING    pid 2028, uptime 0:00:01
dsmr_webinterface                FATAL      Exited too quickly (process log
may have details)
pi@raspberrypimeterkast:~ $ sudo su - dsmr
(dsmrreader)dsmr@raspberrypimeterkast:~/dsmr-reader $ ./deploy.sh

 --- You are currently running version:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: 'module' object has no attribute '__version__'

     [!] FAILED to call Django (did you activate the 'dsmrreader'
VirtualEnv before running?)

[!] Halted in pre-deployment: Either user ABORTED or script FAILED
(dsmrreader)dsmr@raspberrypimeterkast:~/dsmr-reader $

mkruiver, 2017-12-22:
Logfile attached

dennissiemensma, 2017-12-22:
Werkt het nu ook niet als je het proces in Supervisor herstart? Of had je dat al gedaan?

mkruiver, 2017-12-22:
Nee helaas niet, die stopt nu met een FATAL error op de webinterface.

[putty.log](https://github.com/dennissiemensma/dsmr-reader/files/1583315/putty.log)

dennissiemensma, 2017-12-22:
Pfoe, erg lastig zo op afstand debuggen eerlijk gezegd. Je zou nog kunnen proberen om de complete installatie te verwijderen (behalve de database uiteraard) en opnieuw te installeren. Het zou immers moeten werken met een schone installatie, of we missen iets heel voor de hand liggends, maar ik zie het even niet.

mkruiver, 2017-12-23:
Dennis, kun je me nog adviseren welke database/backup ik moet pakken? In de backups folder zie ik wat gekke dingen wat betreft datum/filesize. Ik denk dat de backup-thursday.sql.gz van 245mb de meest recente en integere file is (nov24)?

Of is er ook nog een "live" database op een andere locatie?
Overings vind ik het gek dat de niet gezipte versie van Thursday bijna 3 gb is, of is gz echt zo efficient?

dennissiemensma, 2017-12-24:
Ik zou inderdaad beginnen met de meest recente backup. Je hoeft overigens niet de database te verwijderen als je alleen de applicatie helemaal opnieuw wil installeren.

Gzip is behoorlijk efficient wanneer het aankomt op cijferdata (zoals dsmr-reader), dus het klopt dat de compressie zo efficient is. Ter vergelijking, mijn gzip backup is 98 MB en uitgepakt 847 MB.

mkruiver, 2017-12-27:
Ik heb deze situatie maar aangegrepen om gelijk Raspbian te updaten en een clean install te doen. Ik heb de database kunnen terugzetten (een iets oudere, de meest recente was corrupt - zie log - ERROR:  missing data for column "electricity_delivered_1" ) en DSMR-reader draait weer. Datalogger loopt nog niet want de P1 interface hangt nog niet aan de pi, dat ga ik nu proberen.

[putty27.log](https://github.com/dennissiemensma/dsmr-reader/files/1589774/putty27.log)

Bedankt voor de support Dennis!

mkruiver, 2017-12-27:
En hij draait weer!

closed 2017-12-27 by mkruiver

dennissiemensma, 2017-12-27:
Top, dat is erg fijn om te horen, dan ben je ook niet al te veel data kwijt!

mkruiver, 2017-12-29:
Oja, nog even ter referentie: na het succesvol restoren van een database backup stonden ook al mijn settings weer terug! Na een restore ben je dus echt meteen weer up and running incl settings voor backups, mindergas upload, etc!