dsmrreader/dsmr-reader #407: More than one pvoutput account

mkruiver, 2017-12-29:
Dennis, would it be possible to add a secondary pvoutput account to upload the same data twice to two different accounts?

dennissiemensma, 2017-12-30:
Sure

dennissiemensma, 2018-06-06:
@mkruiver ik krijg dit niet makkelijk of mooi in het systeem. 

Echter, de code die ervoor nodig is om dit te realiseren is vrij beknopt, dus ik wil proberen of ik dit aan de praat krijg via een soort plugin systeem. Daarmee zou ik bepaalde hooks kunnen maken waar anderen weer wat mee doen.
Voor dit specifieke verzoek in het ticket heb ik zelf wat in elkaar gedraaid wat je kunt uitproberen. Ik neem aan dat je deze functionaliteit nog steeds zoekt?

dennissiemensma, 2018-06-06:
Even alvast een stappenplan zodat ik het niet vergeet.

* Zorgen dat je de laatste versie draait, zodat de verschillen zo klein mogelijk zijn (v1.18).

* Wisselen naar branch `origin/407-pvoutput-accounts-and-plugins`

* In `dsmr_plugins/modules/pvoutput.py` (niet-bestaand bestand) dit als inhoud gebruiken:
```
import requests

from dsmr_pvoutput.models.settings import PVOutputAddStatusSettings
from dsmr_pvoutput.signals import pvoutput_upload

def handle_pvoutput_upload(sender, **kwargs):
    response = requests.post(
        PVOutputAddStatusSettings.API_URL,
        headers={
            'X-Pvoutput-Apikey': 'XXXXXXXXXX',
            'X-Pvoutput-SystemId': 'YYYYYYYYYY',
        },
        data=kwargs['data']
    )

    if response.status_code != 200:
        print(' [!] PVOutput upload failed (HTTP {}): {}'.format(response.status_code, response.text))

pvoutput_upload.connect(receiver=handle_pvoutput_upload)
```

* In bovenstaande file de waarden voor `XXXXXXXXXX` en `YYYYYYYYYY` invullen voor je tweede PVOutput account.

* In `dsmrreader/settings.py`:
```
DSMRREADER_PLUGINS = [
    'dsmr_plugins.modules.pvoutput'
]
```

* `./deploy.sh` uitvoeren

mkruiver, 2018-06-06:
Hi Dennis, ik zou dit nog steeds graag gebruiken!

dennissiemensma, 2018-06-11:
Ik zal deze dan ook toevoegen aan de volgende release, gezien het een plugin is, kan het toch weinig kwaad. En het is makkelijk uit te breiden bij issues.

dennissiemensma, 2018-06-11:
Gemerged naar `development`. Ik laat deze nog even openstaan om te weten of dit de gewenste feature is.

dennissiemensma, 2018-06-12:
Bij deze in [v1.19](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.19.0).

dennissiemensma, 2018-06-19:
@mkruiver ben je al in de gelegenheid geweest om dit uit te proberen?

mkruiver, 2018-06-19:
Ik heb v1.19 draaien, maar de plugin nog niet 'gemaakt'.

dennissiemensma, 2018-06-28:
Voor nu sluit ik deze. Mocht het toch niet werken dan hoor ik het tzt graag van je :]

closed 2018-06-28 by dennissiemensma