dsmrreader/dsmr-reader #408: PVOutput only used

Martinvdm, 2017-12-30:
Hi,

1. Because i am uploading my PV system directory to PV Output (netto), i would like to upload Used consumption counters (low and normal) only and not the gerenated counters.

Is this possible?

Martin

dennissiemensma, 2017-12-30:
Hi, ik weet niet helemaal wat je bedoelt (ik heb zelf geen zonnepanelen), dus je zult me goed moeten uitleggen wat je wil. 
Het is momenteel alleen mogelijk om juist alleen je verbruik, via DSMR-reader, te uploaden naar PVOutput. Is dat wat je bedoelt?

Zie ticket #9 voor de complete geschiedenis van de integratie.

Martinvdm, 2017-12-30:
Ok begrijp ik het goed dat dus wel electricity_returned (opgewekt) in DSMR wordt verwerkt en weergegeven maar de implementatie van PVOutput alleen electricity_delivered zal uploaden naar PVOutput?

dennissiemensma, 2017-12-30:
Alle metingen van de slimme meters zijn netto, dus eventueel verbruik in je huis is daarin al verrekend. Daarom is de aanbevolen route voor PVOutput om je zonnepanelen-omvormer alle opbrengst direct naar PVOutput te laten sturen. En DSMR-reader het (netto) verbruik. Dan zou het mogelijk zijn om bruto opbrengst en netto verbruik te tonen.

Wellicht dat iemand uit #9 je hierbij kan helpen, zoals `pyrocumulus`, want mijn kennis van alles m.b.t. zonnepanelen, PVOutput en er omheen is zeeeeer beperkt. De kans is groot dat we het beide over iets anders hebben. 
De huidige implementatie van DSMR-reader stuurt een **oplopende** *dagwaarde* van het verbruik, volgens je slimme meter, naar PVOutput. Dus gemeten vanaf middernacht (en begint dus elke dag weer bij nul). (https://github.com/dennissiemensma/dsmr-reader/blob/v1.11.0/dsmr_pvoutput/services.py#L55)

Martinvdm, 2017-12-30:
bedankt voor de heldere uitleg. Ik heb inmiddels de waardes naar PVOutput laten sturen en dit werkt uitermate goed en betreft alleen het gebruik (laag en normaal). Des al niettemin is het met zonnepanelen zo dat de energie die je opwekt eerst in je eigen huis verbruikt zal worden en daarna pas wat over is de meter door zal gaan (weg van de minste weerstand wordt eerst bewandeld).

Dit betekend dat als je dus de telramen van Used en Generated zou aftrekken om tot een netto resultaat zou willen komen dat dit niet een juiste uitkomst zou zijn.

Maar op zich gaat dit nu helemaal goed. Aangezien je de netto meetwaarden pakt van de Used telramen en de bruto meetwaarden direct vanuit je PV inverter. Op die manier kom je wel tot een netto verbruik met aftrek van zonnepanelen.

bedankt

dennissiemensma, 2017-12-31:
Fijn om te horen, bedankt voor je terugkoppeling :]

closed 2017-12-31 by dennissiemensma