dsmrreader/dsmr-reader #409: Enable power generation

Martinvdm, 2017-12-30:
Hi,

I do not get any values for Power Generation. Have enabled DSMR2.2 and working fine for Power Consumption. For Generation it gives: No data.

Should i enable this somewhere?

dennissiemensma, 2017-12-30:
Hi, ik neem aan dat je dit allemaal hebt ingesteld: http://dsmr-reader.readthedocs.io/nl/latest/pvoutput.html ?

Je zou anders nog in de logs kunnen kijken van het `dsmr_backend` proces. Daar zou sowieso elke X minuten informatie voorbij moeten komen over PVOuput, wanneer correct ingesteld. Elke data upload wordt gemeld met: ` - PVOutput | Uploading data @ ...`

Je kunt dit zien in de logfile `/var/log/supervisor/dsmr_backend.log` (http://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html)

Martinvdm, 2017-12-30:
Morgen nog eens kijken, maar het lijkt erop dat het even een uurtje nodig had om het telraam in DSMR reader te activeren. Als ik morgen weer opwek ga ik nog eens kijken.

Martinvdm, 2017-12-31:
Helaas is de Power Generation nog steeds 0
[image]

dennissiemensma, 2017-12-31:
Ooh sorry ik dacht dat dit over PVOutput ging. In dat geval zul je de output van je slimme meter moeten controleren.

Zou je eens willen kijken wat je telegrammen zijn in de output van `/var/log/supervisor/dsmr_datalogger.log` (als je de standaard datalogger gebruikt)?

Ter vergelijking, bij mij heb ik deze waardes:

```
...
1-0:1.8.1(002440.944*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:1.8.2(002386.725*kWh)
1-0:2.8.2(000000.000*kWh)
...
```

De `1-0:2.8.1` en `1-0:2.8.2` is de teruglevering. Zijn die bij jou wel aanwezig, en gevuld?

Martinvdm, 2018-01-01:
Working just fine now. Wans't sunny enough to activate the meter and consuming all generated power in the house it self! Great project, thanks for all the effort. 👍

dennissiemensma, 2018-01-01:
Top, bedankt voor je feedback.

closed 2018-01-01 by dennissiemensma