dsmrreader/dsmr-reader #410: [API] Toevoegen huidige verbruik aan consumption/today

FutureCow, 2017-12-31:
Ik maak gebruik van de API om live mijn verbruik te kunnen volgen in een ander programma. Dit werkt prima met api/v2/datalogger/dsmrreading. Echter merk ik dat deze wel 4 seconden nodig heeft voordat er een response komt. 
Als ik gebruik maken van api/v2/cunsumtion/today is de response er wel direct, echt zit hier het huidige verbruik/opwek niet bij. Zou het mogelijk zijn deze hieraan toe te voegen?

dennissiemensma, 2018-01-01:
Ik weet niet of ik de verschillende soorten gegevens met elkaar wil mixen. Bedoel je met 'huidige verbruik/opwek' het Wattage dat verbruikt/opgewekt wordt?

Het verschil in response komt door de grootte van de dataset. De metingen nemen 99% van de gegevens in beslag. Gezien ik nu meerdere signalen krijg dat de data wat (te) groot wordt, zal ik op korte termijn verder gaan met #72. Daarmee zouden alle verzoeken en API calls m.b.t. metingen ook stukken sneller moeten gaan.

FutureCow, 2018-01-01:
Het lastige is dat ik alle gegevens van het dashboard ook op een andere plek wil zien:
- electricity_currently_delivered
- electricity_currently_returned
- electricity1
- electricity1_returned
- gas

Dit zijn denk ik de meest gebruikte gevens om via de API door te sturen?

Ik heb hier alleen 2 keer een API call voor nodig waarvan 1 zeer snel en 1 erg traag. Mijn database na bijna 1 jaar:
 dsmrreader | dsmrreader | UTF8     | nl_NL.UTF-8 | nl_NL.UTF-8 |                       | 5912 MB | pg_default |

dennissiemensma, 2018-01-01:
De oorzaak van de traagheid is de grootte van de data. Dus als een andere API-call dezelfde informatie opzoekt, zal die uiteindelijk ook 4 seconden duren, vandaar dat het alleen (goed) op te lossen is met #72. 

Dat ruimt dus ook je database voor een groot deel op, zie de dev comments in dat ticket:
https://github.com/dennissiemensma/dsmr-reader/issues/72#issuecomment-336256458
(van bijna 1 GB naar slechts 5 MB)

dennissiemensma, 2018-01-17:
@FutureCow heb je inmiddels de nieuwe retentie-feature (sinds zondag) kunnen uitproberen? Uiteindelijk zullen je API-calls daar veel sneller van worden.

FutureCow, 2018-01-17:
Bedankt voor je vraag, gister update gedaan naar v1.12. Verschil is enorm. Had 4GB aan data in 30.000.000 telegrams. Nu op 1 maand gezet en het is geslonken naar 2.600.000. Hierdoor is de api call nu in 0.48 s klaar. Een stuk beter dan de 4 seconde dat het was!
Wat is verder het nadeel van het verwijderen van alle data? Opnieuw opbouwen maakt me niet zoveel uit, maar worden andere statistieken hier ook uit gehaald?

dennissiemensma, 2018-01-18:
De gegevens worden voornamelijk gebruikt in de grafieken op het dashboard, maar doorgaans is alles ouder dan een week al niet heel relevant (in de zin van, "3 dagen geleden op 14:23 is het Wattage van L2 1200W geweest").

Waar het verder nog voor gebruikt wordt, zijn de stats van de hoogste/laagste Wattage op de Statistieken-pagina.

dennissiemensma, 2018-06-06:
Ik ben nu even kwijt of voor dit ticket nog follow-up voor nodig was, maar ik had het idee dat je nu met de snellere API de gegevens die je zocht snel kon ophalen. :]

closed 2018-06-06 by dennissiemensma