dsmrreader/dsmr-reader #411: Processing of telegrams stopped

mkruiver, 2017-12-31:
Sinds vannacht 05:12 is het processen van telegrammen gestopt...

Telegramverwerking
Er zijn te veel onverwerkte telegrammen.

Onverwerkte telegrammen: 26.399 

dsmr_backend geeft een unrelated error over diskspace (iets eerder) en geeft als laatste aan: creating new backup.

> pi@raspberrypimeterkast:/var/log/supervisor $ sudo supervisorctl tail -f dsmr_backend
==> Press Ctrl-C to exit <==
03+01:00 (0.126 kW)
 - Processed reading: 22472664 @ 2017-12-31 05:12:04+01:00 (0.126 kW)
 - Processed reading: 22472665 @ 2017-12-31 05:12:05+01:00 (0.126 kW)
 - Processed reading: 22472666 @ 2017-12-31 05:12:06+01:00 (0.124 kW)
 - Processed reading: 22472667 @ 2017-12-31 05:12:07+01:00 (0.125 kW)
 - Processed reading: 22472668 @ 2017-12-31 05:12:08+01:00 (0.131 kW)
 - Processed reading: 22472669 @ 2017-12-31 05:12:09+01:00 (0.132 kW)
 - Processed reading: 22472670 @ 2017-12-31 05:12:10+01:00 (0.126 kW)
 - Processed reading: 22472671 @ 2017-12-31 05:12:11+01:00 (0.127 kW)
 - Processed reading: 22472672 @ 2017-12-31 05:12:18+01:00 (0.129 kW)
 - Processed reading: 22472673 @ 2017-12-31 05:12:19+01:00 (0.129 kW)
 - Processed reading: 22472674 @ 2017-12-31 05:12:20+01:00 (0.127 kW)
 - Processed reading: 22472675 @ 2017-12-31 05:12:21+01:00 (0.127 kW)
 - Processed reading: 22472676 @ 2017-12-31 05:12:22+01:00 (0.126 kW)
 - Processed reading: 22472677 @ 2017-12-31 05:12:23+01:00 (0.129 kW)
 - Creating new backup.
^C
pi@raspberrypimeterkast:/var/log/supervisor $

Postgres laad de hele cpu

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
 1084 postgres  20   0  198964  26592  23400 D  65.1  2.8   7:35.89 postgres
 1083 dsmr      30  10   12700   7492   5792 S  22.0  0.8   2:33.04 pg_dump
 1081 dsmr      25   5   44528  35872  10240 S  12.2  3.8   0:55.70 python3
 1087 postgres  20   0  197056  15636  13744 D   1.3  1.6   0:04.58 postgres
 1298 pi        20   0    8208   3316   2768 R   1.0  0.3   0:00.19 top

Ik heb al een deploy.sh en reboot van de pi geprobeerd.
Er lijken geen dubbele readings in de DB te zitten, hij is gewoon gestopt.... backup maken loopt vast?
Versie 1.11 trouwens.

mkruiver, 2017-12-31:
` - Processed reading: 22472644 @ 2017-12-31 05:11:36+01:00 (0.128 kW)
 - Processed reading: 22472645 @ 2017-12-31 05:11:37+01:00 (0.127 kW)
 - Processed reading: 22472646 @ 2017-12-31 05:11:38+01:00 (0.128 kW)
 - Processed reading: 22472647 @ 2017-12-31 05:11:39+01:00 (0.127 kW)
 - Processed reading: 22472648 @ 2017-12-31 05:11:40+01:00 (0.126 kW)
 - Processed reading: 22472649 @ 2017-12-31 05:11:41+01:00 (0.126 kW)
 - Processed reading: 22472650 @ 2017-12-31 05:11:42+01:00 (0.126 kW)
 - Processed reading: 22472651 @ 2017-12-31 05:11:43+01:00 (0.124 kW)
 - Processed reading: 22472652 @ 2017-12-31 05:11:44+01:00 (0.127 kW)
 - Processed reading: 22472653 @ 2017-12-31 05:11:45+01:00 (0.126 kW)
 - Processed reading: 22472654 @ 2017-12-31 05:11:54+01:00 (0.127 kW)
 - Processed reading: 22472655 @ 2017-12-31 05:11:55+01:00 (0.125 kW)
 - Processed reading: 22472656 @ 2017-12-31 05:11:56+01:00 (0.126 kW)
 - Processed reading: 22472657 @ 2017-12-31 05:11:57+01:00 (0.125 kW)
 - Creating new backup.
pg_dump: [archiver] could not write to output file: No space left on device
 - PVOutput | Uploading data @ 05:12
 - PVOutput | Delaying the next export until: 2017-12-31 16:03:03.531736+00:00
 >>> Uncaught exception :: could not extend file "base/17049/17397": No space left on device
HINT:  Check free disk space.
 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_consumption>> ::   File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backen$
    raise current_response

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/dispatch/dispatcher.py", line 228, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_consumption/apps.py", line 20, in _on_backend_called_signal
    dsmr_consumption.services.compact_all()

  File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 21, in compact_all
    compact(dsmr_reading=current_reading)
File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 46, in compact
    dsmr_reading.save(update_fields=['processed'])

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/base.py", line 808, in save
    force_update=force_update, update_fields=update_fields)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/base.py", line 838, in save_base
    updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/base.py", line 905, in _save_table
    forced_update)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/base.py", line 955, in _do_update
    return filtered._update(values) > 0

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 664, in _update
    return query.get_compiler(self.db).execute_sql(CURSOR)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1199, in execute_sql
    cursor = super(SQLUpdateCompiler, self).execute_sql(result_type)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 894, in execute_sql
    raise original_exception

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 884, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/raven/contrib/django/client.py", line 114, in execute
    return real_execute(self, sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)
 return self.cursor.execute(sql, params)
[... 92 lines cut ...]

2017-12-31 16:58:07.542832+01:00: Calling backend services
 - Processed reading: 22472658 @ 2017-12-31 05:11:58+01:00 (0.125 kW)
 - Processed reading: 22472659 @ 2017-12-31 05:11:59+01:00 (0.126 kW)
 - Processed reading: 22472660 @ 2017-12-31 05:12:00+01:00 (0.125 kW)
 - Processed reading: 22472661 @ 2017-12-31 05:12:01+01:00 (0.126 kW)
 - Processed reading: 22472662 @ 2017-12-31 05:12:02+01:00 (0.126 kW)
 - Processed reading: 22472663 @ 2017-12-31 05:12:03+01:00 (0.126 kW)
 - Processed reading: 22472664 @ 2017-12-31 05:12:04+01:00 (0.126 kW)
 - Processed reading: 22472665 @ 2017-12-31 05:12:05+01:00 (0.126 kW)
 - Processed reading: 22472666 @ 2017-12-31 05:12:06+01:00 (0.124 kW)
 - Processed reading: 22472667 @ 2017-12-31 05:12:07+01:00 (0.125 kW)
 - Processed reading: 22472668 @ 2017-12-31 05:12:08+01:00 (0.131 kW)
 - Processed reading: 22472669 @ 2017-12-31 05:12:09+01:00 (0.132 kW)
 - Processed reading: 22472670 @ 2017-12-31 05:12:10+01:00 (0.126 kW)
 - Processed reading: 22472671 @ 2017-12-31 05:12:11+01:00 (0.127 kW)
 - Processed reading: 22472672 @ 2017-12-31 05:12:18+01:00 (0.129 kW)
 - Processed reading: 22472673 @ 2017-12-31 05:12:19+01:00 (0.129 kW)
 - Processed reading: 22472674 @ 2017-12-31 05:12:20+01:00 (0.127 kW)
 - Processed reading: 22472675 @ 2017-12-31 05:12:21+01:00 (0.127 kW)
 - Processed reading: 22472676 @ 2017-12-31 05:12:22+01:00 (0.126 kW)
 - Processed reading: 22472677 @ 2017-12-31 05:12:23+01:00 (0.129 kW)
 - Creating new backup.

`

mkruiver, 2017-12-31:
Ik heb wat oude database-backups verwijdert om ruimte te maken. Kijken wat 'ie doet.

pi@raspberrypimeterkast:~ $ sudo supervisorctl tail -f dsmr_backend             ==> Press Ctrl-C to exit <==
12+01:00 (0.207 kW)
 - Processed reading: 22473712 @ 2017-12-31 05:38:13+01:00 (0.204 kW)
 - Processed reading: 22473713 @ 2017-12-31 05:38:14+01:00 (0.206 kW)
 - Processed reading: 22473714 @ 2017-12-31 05:38:16+01:00 (0.206 kW)
 - Processed reading: 22473715 @ 2017-12-31 05:38:23+01:00 (0.208 kW)
 - Processed reading: 22473716 @ 2017-12-31 05:38:24+01:00 (0.205 kW)
 - Processed reading: 22473717 @ 2017-12-31 05:38:25+01:00 (0.208 kW)
 - Processed reading: 22473718 @ 2017-12-31 05:38:26+01:00 (0.208 kW)
 - Processed reading: 22473719 @ 2017-12-31 05:38:27+01:00 (0.209 kW)
 - Processed reading: 22473720 @ 2017-12-31 05:38:28+01:00 (0.210 kW)
 - Processed reading: 22473721 @ 2017-12-31 05:38:29+01:00 (0.209 kW)
 - Processed reading: 22473722 @ 2017-12-31 05:38:30+01:00 (0.206 kW)
 - Processed reading: 22473723 @ 2017-12-31 05:38:31+01:00 (0.208 kW)
 - Processed reading: 22473724 @ 2017-12-31 05:38:32+01:00 (0.208 kW)
 - Processed reading: 22473725 @ 2017-12-31 05:38:33+01:00 (0.207 kW)
 - Creating new backup.

mkruiver, 2017-12-31:
Hij draait weer na het vrijmaken van schijfruimte, rest de vraag: hoe loopt mijn nieuwe installatie van Raspbian op een 16gb SD-kaartje zo snel vol?

mkruiver, 2018-01-01:
Ik zie dat de Postgresql folder ruim 4gb in beslag neemt, onder andere door een aantal files van 1gb op de dag van de db restore. Kan ik die verwijderen?

-rwxr-xr-x 1 postgres postgres 1024M Dec 27 13:20 17151
-rwxr-xr-x 1 postgres postgres 1024M Dec 27 13:25 17151.1

mkruiver, 2018-01-01:
Is er trouwens een manier om oude telegrams (oid) uit de db te verwijderen zodat deze <3GB wordt? Op deze manier is het uiteindelijk onhoudbaar om alles te archiveren over x aantal jaar.

dennissiemensma, 2018-01-01:
Ik weet niet zeker welke interne files van Postgresql weg kunnen. Je zou nog even kunnen googlen daarop, maar wellicht is het veiliger om de retentie van de backups (handmatig) te verkleinen. Het huidige mechanisme van backups is nooit berekend geweest op DSMR v5, dus mogelijk wil je alleen de laatste drie of vier backups bewaren.

Je kunt het opschonen met find + delete vlag. Bijvoorbeeld dat in een cronjob te zetten. Dan gooit hij alleen de files weg ouder dan X dagen.

Dry-runnen kan met: 
`find /home/dsmr/dsmr-reader/backups/ -type f -iname '*.sql*' -mtime +3` (dan zie je wat die raakt)
Daadwerkelijk verwijderen met `-delete` er achter: 
`find /home/dsmr/dsmr-reader/backups/ -type f -iname '*.sql*' -mtime +3 -delete`

De oplossing voor de lange termijn is het oplossen van #72 , dus daar zal ik binnenkort weer mee verder.

dennissiemensma, 2018-01-01:
Dus voor de duidelijkheid, de enige goede oplossing is naar mijn idee #72, alleen is dat nu wat versneld nodig door DSMR v5 (grote hoeveelheid telegrammen)

dennissiemensma, 2018-01-02:
Mocht het trouwens toch nog een issue zijn tot die tijd, je kunt handmatig metingen verwijderen in de database met:

`delete from dsmr_datalogger_dsmrreading where processed = True and timestamp < now() - interval '3 months';`

Je kunt dus zelf bepalen tot hoeveel maanden terug je gegevens wilt verwijderen. Voor de zekerheid maak je het beste een backup van de gegevens (de nachtelijke backup) en sla die op een aparte plek op. Dan heb je die metingen altijd nog.

Let op dat PostgreSQL gegevens niet direct verwijdert van de schijf, maar ze eerst markeert als verwijderbaar. Er is een proces `autovacuum` in de database die langzaam de gegevens opruimt. Je kunt dit forceren door handmatig te vacuumen met:
* `sudo su - postgres`
* `vacuumdb -v -f dsmrreader`

Dit kan enige tijd in beslag nemen, afhankelijk van de disk en de hoeveelheid gegevens.

dennissiemensma, 2018-01-02:
Ik sluit deze voor nu, omdat #72 dit moet oplossen. Als je nog vragen hebt hierover, hoor ik ze uiteraard graag.

closed 2018-01-02 by dennissiemensma

dennissiemensma, 2018-01-14:
Zojuist is #72 uitgebracht in [release v1.12](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.12.0).

dennissiemensma, 2018-01-14:
Extra aandacht voor de [opmerkingen in de docs](http://dsmr-reader.readthedocs.io/nl/latest/retention.html#notes-warnings), om daadwerkelijk ruimte vrij te maken, zul je na enkele dagen van retentieopschoning een keer handmatig `vacuumdb` moeten draaien.

mkruiver, 2018-01-16:
Het gaat voorlopig heel langzaam, op een pi3, maar de database backups worden kleiner.

pi@raspberrypimeterkast:/home/dsmr/dsmr-reader/backups $ sudo du -h -s * | sort -nr | head
334M    dsmrreader-postgresql-backup-Sunday.sql.gz
333M    dsmrreader-postgresql-backup-Saturday.sql.gz
323M    dsmrreader-postgresql-backup-Monday.sql.gz
287M    dsmrreader-postgresql-backup-Tuesday.sql.gz

dennissiemensma, 2018-01-17:
Oke dat is mooi (en naar verwachting)! Achteraf gezien had de feature in #72 al veel eerder beschikbaar mogen zijn, maar ik had persoonlijk niet helemaal door dat DSMRv5-meters voor dusdanig veel telegrammen (en schijfgebruik) zorgt.

In dit tempo vermoed ik dat je over een week mogelijk al rond bent qua retentie. Mocht je exacte details willen zien over wat die doet in een nacht, en waar die is geeindigd in de ochtend, [zie dan dit comment](https://github.com/dennissiemensma/dsmr-reader/issues/72#issuecomment-357708670).

mkruiver, 2018-01-17:
Het gaat nu harder (en mn cronjob werkt niet, om backups ouder dan 3dagen weg te gooien).

(dsmrreader) dsmr@raspberrypimeterkast:~/dsmr-reader/backups $ ls -lah
total 1.5G
drwxr-xr-x  2 dsmr dsmr 4.0K Jan 17 04:00 .
drwxr-xr-x 22 dsmr dsmr 4.0K Jan 14 13:57 ..
-rw-r--r--  1 dsmr dsmr 323M Jan 15 04:08 dsmrreader-postgresql-backup-Monday.sql.gz
-rw-r--r--  1 dsmr dsmr 333M Jan 13 11:21 dsmrreader-postgresql-backup-Saturday.sql.gz
-rw-r--r--  1 dsmr dsmr 334M Jan 14 04:09 dsmrreader-postgresql-backup-Sunday.sql.gz
-rw-r--r--  1 dsmr dsmr 287M Jan 16 04:16 dsmrreader-postgresql-backup-Tuesday.sql.gz
-rw-r--r--  1 dsmr dsmr 198M Jan 17 04:00 dsmrreader-postgresql-backup-Wednesday.sql.gz
-rw-r--r--  1 dsmr dsmr    4 Dec 25 11:40 .dummy
(dsmrreader) dsmr@raspberrypimeterkast:~/dsmr-reader/backups $

mkruiver, 2018-01-17:
334M    dsmrreader-postgresql-backup-Sunday.sql.gz
333M    dsmrreader-postgresql-backup-Saturday.sql.gz
323M    dsmrreader-postgresql-backup-Monday.sql.gz
287M    dsmrreader-postgresql-backup-Tuesday.sql.gz
198M    dsmrreader-postgresql-backup-Wednesday.sql.gz
(dsmrreader) dsmr@raspberrypimeterkast:~/dsmr-reader/backups $

dennissiemensma, 2018-01-17:
Fijn, dat gaat de goede kant op. Ik neem aan dat je inmiddels door de kleine backup's ook al iets meer speling hebt op je (eerder) volle schijf?

mkruiver, 2018-01-19:
Dit gaat nu goed! Ook PVO output heeft (bijna) geen gaps meer snachts.

(dsmrreader) dsmr@raspberrypimeterkast:~/dsmr-reader/backups $ ls -laht
total 921M
drwxr-xr-x  2 dsmr dsmr 4.0K Jan 19 03:46 .
-rw-r--r--  1 dsmr dsmr  27M Jan 19 03:46 dsmrreader-postgresql-backup-Friday.sql.gz
-rw-r--r--  1 dsmr dsmr  89M Jan 18 03:53 dsmrreader-postgresql-backup-Thursday.sql.gz
-rw-r--r--  1 dsmr dsmr 198M Jan 17 04:00 dsmrreader-postgresql-backup-Wednesday.sql.gz
-rw-r--r--  1 dsmr dsmr 287M Jan 16 04:16 dsmrreader-postgresql-backup-Tuesday.sql.gz
-rw-r--r--  1 dsmr dsmr 323M Jan 15 04:08 dsmrreader-postgresql-backup-Monday.sql.gz
drwxr-xr-x 22 dsmr dsmr 4.0K Jan 14 13:57 ..
-rw-r--r--  1 dsmr dsmr    4 Dec 25 11:40 .dummy
(dsmrreader) dsmr@raspberrypimeterkast:~/dsmr-reader/backups $

mkruiver, 2018-01-22:
Disk usage is nu onder de 40%, opgelost dus!

dennissiemensma, 2018-01-22:
Mooi, erg fijn om te horen.