dsmrreader/dsmr-reader #412: Discarded telegram with future timestamp: /KFM5KAIFA-METER

MWolthuis, 2018-01-01:
Hallo Dennis,

Allereerst de beste wensen! Nog steeds zeer tevreden met dsmr-reader.

Helaas heeft dit nieuwe jaar mij wel meteen een probleem opgeleverd met de applicatie.

1 januari iets na 02:00 is de applicatie gestopt met het verwerken van de telegrammen. Deze zie ik met "supervisor> tail -f dsmr_datalogger" wel gewoon binnen komen.

Django.log laat het volgende zien:
[2018-01-01 21:44:29,652] ERROR @ services | Discarded telegram with future timestamp: /KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(180101214355W)
0-0:96.1.1(xxxxxxxxxxserienummerweggehaaldxxxxxxxxxxx)
1-0:1.8.1(001599.784*kWh)
1-0:1.8.2(001505.096*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.566*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00002)
0-0:96.7.9(00002)
1-0:99.97.0(2)(0-0:96.7.19)(170604203937S)(0000006234*s)(000101000001W)(2147483647*s)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(002*A)
1-0:21.7.0(00.563*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(xxxxxxxxxxserienummerweggehaaldxxxxxxxxxxx)
0-1:24.2.1(190101210000W)(01269.375*m3)
!3759

Het lijkt dus fout te gaan met de timestamp in het telegram. Probleem bij de netbeheerder? Of zou ik het ergens anders moeten zoeken?

dennissiemensma, 2018-01-01:
Ik zal morgen nog eens beter kijken, maar het lijkt erop dat je gasmeter een jaar vooruit loopt: 190101210000W

Ik heb er niet direct een oplossing voor. Je zou de toekomst-check in de code kunnen uitschakelen. Morgen kijk ik wel even wat te doen.

MWolthuis, 2018-01-02:
In de loop van de ochtend is de tijd door de netbeheerder weer correct gesynchroniseerd. Alles werkt weer.

Wel heeft dit tot gevolg dat ik voor vandaag een hoog gasverbruik heb in de grafiek. Al het verbruik van de dagen dat DSMR telegrammen niet verwerkte is weggezet in de grafiek op het moment dat het weer functioneerde.

closed 2018-01-02 by MWolthuis

dennissiemensma, 2018-01-02:
Ok top en goed om te horen! Dat is mooi want, achteraf, had ik hier ook geen goede oplossing voor kunnen bedenken. 

Je kunt eventueel de dagtotalen qua gas handmatig bijwerken als de gas-piek je stoort.