dsmrreader/dsmr-reader #413: Hoge CPU belasting op rpi 2 icm DSMR 5.0 meter

meijjaa, 2018-01-04:
Hallo Dennis,

Mijn 5.0 meter verstuurt elke seconde een telegram waar mijn rpi 2 het behoorlijk zwaar mee heeft. (datalogger, backend, postgres).

Is het mogelijk om de telegram input the throttlen? 

Even een snelle hack (waar ik de gevolgen niet van in kan schatten) lijkt goed te helpen:
```

diff --git a/dsmr_backend/management/commands/dsmr_backend.py b/dsmr_backend/management/commands/dsmr_backend.py
index 6405b7b..dad010b 100644
--- a/dsmr_backend/management/commands/dsmr_backend.py
+++ b/dsmr_backend/management/commands/dsmr_backend.py
@@ -12,7 +12,7 @@ import dsmr_backend.signals
 class Command(InfiniteManagementCommandMixin, BaseCommand):
     help = _('Generates a generic event triggering apps for backend operations, cron-like.')
     name = __name__  # Required for PID file.
-    sleep_time = 1
+    sleep_time = 60
 
     def run(self, **options):
         """ InfiniteManagementCommandMixin listens to handle() and calls run() in a loop. """
diff --git a/dsmr_datalogger/management/commands/dsmr_datalogger.py b/dsmr_datalogger/management/commands/dsmr_datalogger.py
index dda1375..98b7380 100644
--- a/dsmr_datalogger/management/commands/dsmr_datalogger.py
+++ b/dsmr_datalogger/management/commands/dsmr_datalogger.py
@@ -10,7 +10,7 @@ import dsmr_datalogger.services
 class Command(InfiniteManagementCommandMixin, BaseCommand):
     help = _('Performs an DSMR P1 telegram reading on the COM port.')
     name = __name__  # Required for PID file.
-    sleep_time = 0.25
+    sleep_time = 60
```

dennissiemensma, 2018-01-04:
Bedankt voor je melding en suggestie. Dat lijkt me een prima workaround voor nu. Ik zal er een instelling van maken, zodat je dit netjes in je config kunt opnemen.

dennissiemensma, 2018-01-04:
Je kunt het vanaf de volgende release zelf overschrijven in `/home/dsmr/dsmr-reader/dsmrreader/settings.py` door deze regels toe te voegen en de processen te herstarten:

```
DSMRREADER_BACKEND_SLEEP = 60
DSMRREADER_DATALOGGER_SLEEP = 60
```

closed 2018-01-04 by dennissiemensma

meijjaa, 2018-01-05:
Dank je! Lijkt goed te werken.

dennissiemensma, 2018-01-14:
Zojuist uitgebracht in [release v1.12](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.12.0).