dsmrreader/dsmr-reader #418: dsmr_webinterface fails to start because of py3/gunicorn issue

swerveshot, 2018-01-08:
After doing a fresh install of dsmr-reader the dsmr_webinterface process fails to start.

When reviewing the logfile in /var/log/supervisor/dsmr_webinterface.log the following stacktrace can be seen.
```
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/sock.py", line 44, in set_options
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEPORT, 1)
OSError: [Errno 92] Protocol not available

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn", line 11, in <module>
    sys.exit(run())
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/app/wsgiapp.py", line 74, in run
    WSGIApplication("%(prog)s [OPTIONS] [APP_MODULE]").run()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/app/base.py", line 203, in run
    super(Application, self).run()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/app/base.py", line 72, in run
    Arbiter(self).run()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/arbiter.py", line 198, in run
    self.start()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/arbiter.py", line 157, in start
    self.LISTENERS = sock.create_sockets(self.cfg, self.log, fds)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/sock.py", line 180, in create_sockets
    sock = sock_type(addr, conf, log)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/sock.py", line 117, in __init__
    super(UnixSocket, self).__init__(addr, conf, log, fd=fd)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/sock.py", line 32, in __init__
    self.sock = self.set_options(sock, bound=bound)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/sock.py", line 46, in set_options
    if err[0] not in (errno.ENOPROTOOPT, errno.EINVAL):
TypeError: 'OSError' object is not subscriptable
[2018-01-08 12:38:46 +0100] [25720] [INFO] Starting gunicorn 19.7.1
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/gunicorn/sock.py", line 44, in set_options
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEPORT, 1)
OSError: [Errno 92] Protocol not available
```

After doing research it turns out that there is an exception handling issue with python3 and gunicorn 19.7.1: benoitc/gunicorn#1491. This is one of the requirements that is installed via PIP by the dsmrreader provisioning script.

The fix is already in master but has not yet been included in any of the releases.

So untill a release has been made available you can apply the fix yourselves by editing sock.py. benoitc/gunicorn@a3dbc3051b2da5694c797d333c27072065ab7ac2

swerveshot, 2018-01-08:
For completeness sake, I use the following software versions:
- ubuntu 16.04 LTS
- python 3.5.2
- gunicorn 19.7.1

dennissiemensma, 2018-01-08:
Bedankt voor je ticket en tevens het uitgebreide onderzoek. Het is jammer dat Gunicorn dan al bijna 10 maanden geen nieuwe release heeft gehad.

Ik denk dat je het beste voor nu handmatig de patch lokaal kan toepassen, want als er een nieuwe versie uitkomt dan mag die standaard geinstalleerd worden. Kun je proberen te kijken of dat het voor je oplost? Momenteel ben je de enige met dit issue, ik heb zelf geen fouten met python3 + deze Gunicorn-versie. :]

swerveshot, 2018-01-08:
Het was ook meer een kwestie van verslaglegging voor mensen die net als ik een nieuwe installatie doen en tegen hetzelfde probleem aanlopen. Je kunt het issue wat mij betreft direct sluiten hoor. Maar dan is het in ieder geval een known issue.

swerveshot, 2018-01-08:
Oh en het lokaal toepassen van de patch lost het probleem inderdaad op. Mijn web frontend doet het nu als een zonnetje. Off-topic;  is er ook een mogelijkheid om de applicatie onder een andere base url te draaien (bijvoorbeeld `http://server/dsmrreader`)? Dan kan ik NGNIX voor nog meer applicaties gebruiken.

dennissiemensma, 2018-01-08:
Oke top, bedankt voor je terugkoppeling.

dennissiemensma, 2018-01-08:
Je kunt in principe elke webserver ervoor zetten die proxy ondersteunt, zoals Apache. Je zou inderdaad ook `/dsmrreader/` kunnen gebruiken, al heb ik niet zo 1-2-3 een voorbeeld qua configuratie daarvan paraat. Als derde optie zou je Nginx (of de vhost van dsmr-reader) op een andere poort kunnen draaien, zoals in #169.

dennissiemensma, 2018-01-08:
Mocht je er zelf niet uitkomen, dan kan ik eventueel wel wat proberen te spelen met een Nginx-configuratie.

closed 2018-01-08 by dennissiemensma