dsmrreader/dsmr-reader #420: PVO upload stops during backups

mkruiver, 2018-01-10:
I noticed a gap in my PVO uploads and noticed these are overlapping with the database backup around 3AM. It seems to be the case that during the backup of my (3GB+) database, uploads to PVO are stopped. When they resume, it results in a gap in the consumption data on PVO.

[image]

https://www.pvoutput.org/intraday.jsp?id=54637&sid=49673&dt=20180110&gs=0&m=0

https://www.pvoutput.org/intraday.jsp?id=54637&sid=49673&dt=20180109&gs=0&m=1

dennissiemensma, 2018-01-10:
Bedankt voor het melden van dit issue. Zou je eens kunnen kijken in de log `/var/log/supervisor/dsmr_backend.log`, rond dat tijdstip, of er foutmeldingen terug te vinden zijn?

mkruiver, 2018-01-10:
Geen error, wel een rare delay?

2018-01-09 03:42:23.912664+01:00: Calling backend services                         Command completed. Sleeping for 1 second(s)...                                     2018-01-09 03:42:25.299302+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-01-09 03:42:26.722776+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-01-09 03:42:28.086009+01:00: Calling backend services
 - Creating new backup.
 - PVOutput | Uploading data @ 03:42
 - PVOutput | Delaying the next export until: 2018-01-09 03:11:36.911601+00:00
Command completed. Sleeping for 1 second(s)...

2018-01-09 04:06:38.434557+01:00: Calling backend services
 - Processed reading: 23217577 @ 2018-01-09 03:42:00+01:00 (0.152 kW)
 - Processed reading: 23217578 @ 2018-01-09 03:42:01+01:00 (0.152 kW)
 - Processed reading: 23217579 @ 2018-01-09 03:42:02+01:00 (0.151 kW)
 - Processed reading: 23217580 @ 2018-01-09 03:42:03+01:00 (0.151 kW)

mkruiver, 2018-01-10:
2018-01-09 04:11:35.552866+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-01-09 04:11:36.952439+01:00: Calling backend services
 - PVOutput | Uploading data @ 04:00
 - PVOutput | Delaying the next export until: 2018-01-09 03:16:38.051912+00:00
Command completed. Sleeping for 1 second(s)...

mkruiver, 2018-01-10:
Ohnee dat verschil is gewoon gmt+0 en gmt+1

dennissiemensma, 2018-01-10:
Hmmm, ik denk dat het proces mogelijk simpel wacht op de database. En omdat het geen webrequest is (by design) komt er ook geen timeout en gaat die later vrolijk verder.

Ik vraag me af of ik hier iets concreet aan kan doen. Het vreemde is wel dat (bij mij) de datalogger geen data lijkt te missen tijdens de backups. 

Een mogelijke workaround is simpelweg het verkorten van de backupduur/datagrootte. Ik ben actief bezig om #72 af te ronden, als laatste ticket van de volgende release. Ik denk dat dat sowieso aardig wat lucht gaat geven qua datagrootte en performance.

dennissiemensma, 2018-01-14:
Zojuist is #72 uitgebracht in [release v1.12](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.12.0).

Je zou retentie kunnen inschakelen (zal wel even wat tijd nodig hebben om oude data op te ruimen). Mogelijk lost dat indirect je probleem op, door de kleinere backup's op den duur.

dennissiemensma, 2018-01-16:
In hoeverre zijn je backup's al kleiner geworden door de retentie? Zit daar enige voortgang in? 

Voor het opschonen van schijfruimte in de database is vacuum nodig, maar je zou direct al (wat) kleinere backup's moeten zien in de nachten na retentie.

mkruiver, 2018-01-19:
Ik verwacht dat dit vannacht goed zal gaan.

```
pi@raspberrypimeterkast:~ $ df -h                                                  Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  5.6G  8.4G  41% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M  4.0K  464M   1% /dev/shm
tmpfs           464M   36M  429M   8% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   41M   21M   21M  51% /boot
tmpfs            93M   16K   93M   1% /run/user/1000
pi@raspberrypimeterkast:~ $
```