dsmrreader/dsmr-reader #421: HCC Apeldoorn presentatie over DSMR reader

hansb314, 2018-01-11:
Ter informatie achter +++ een invitatie voor een presentatie over DSMR reader .
 Ik moet de presentatie nog maken, maar het zal in ieder geval een positief verhaal worden. Ik ben onder de indruk van de kwaliteit van het geheel.
De presentatie is ook bedoeld om belangstelling te wekken voor de Raspberry Pi, bij voorkeur ook bij mensen die daar nog niet mee bekend zijn. Als er belangstelling is, gaan we een workshop houden hoe de Raspberry Pi in te richten en hoe DSMR reader te installeren.

++++++++++++++++++++++++++++++++++++++++++++++++++
donderdag 25 januari 2018
IOT, de slimme meter
donderdag 25 januari 2018
20:00 tot 23:00
Wijkcentrum DokZuid

1e Wormenseweg 460, 7333 GZ · Apeldoorn

Presentatie DSMR door Hans Boer.

Hans gaat de slimme energiemeter uitlezen met het DSMR (Dutch Smart Meter Reader) pakket op een Raspberry Pi.
Het software pakket is open source en dus door ieder vrij te gebruiken, is een Nederlands product en gemaakt door Dennis Siemensma

Steeds meer huizen hebben een slimme meter en er zijn nu 40 mogelijkheden met hardware en software om zelf je energiegebruik bij te houden. Ook als je een gratis schermpje krijgt als je naar een andere leverancier overstapt, zijn ze niet gratis. Niet alleen de apparatuur moet je betalen maar vaak zijn er ook abonnementskosten. Bovenal worden de gegevens van je energieverbruik “ergens anders” opgeslagen.

Je kunt het ook zelf doen: met het open source pakket DSMR (Dutch Smart Meter Reader) kun je je eigen gegevens uitlezen, bewaren en bekijken in grafieken en overzichten. Omdat alles open source is, kun je eventueel zelf uitbreidingen maken om nog meer met de gegevens te doen.

Ingebouwd is al: bekijken van diverse grafieken via een (eigen!) website; exporteren van uur- of dagverbruik om spreadsheets te maken; exporteren van database back-ups; doorgeven dagelijks gebruik aan je telefoon.

DSMR werkt op een Raspberry Pi en is een pakket met Linux applicaties. Dankzij de uitstekende documentatie en meegeleverde procedures is installatie en gebruik ook goed mogelijk voor mensen zonder kennis van Raspberry Pi, databases, webservers en Linux.
Voor wie dat allemaal toch wat veel vindt, we gaan binnenkort verder in workshops samen de hardware en software installeren en voor gebruik klaar maken.

dennissiemensma, 2018-01-11:
Hallo Hans, bedankt voor je uitnodiging en dat je een presentatie over dit project houdt. Ik voel mij daarmee vereerd!

Mocht je nog meer informatie of feedback nodig hebben voor je presentatie, dan kun je mij per mail bereiken op github apenstaartje dennissiemensma .nl. 

Wat betreft mijn aanwezigheid die avond, momenteel heb ik helaas al iets anders gepland staan. Echter, waarschijnlijk kan ik alsnog aanwezig zijn in Apeldoorn. Ik kom daar op later moment nog op terug.

*P.S.: Het is wellicht goed om duidelijk te maken dat de term `DSMR` (Dutch Smart Meter Requirements) op zichzelf een initiatief is van de Nederlandse netbeheerders. En dat dit project `DMSR-reader` heet. Om eventuele verwarring te voorkomen. ;-)*

dennissiemensma, 2018-01-16:
Ik ben die avond aanwezig, al ben ik mogelijk iets later dan 20:00.

closed 2018-01-16 by dennissiemensma