dsmrreader/dsmr-reader #422: Recent gas consumed (m3) 1 hour behind

Wolverine65, 2018-01-12:
Hi, 

Why is the Recent gas consumed (m3) graph 1 hour behind? I thought that every hour the gas meter is read. Im using influxdb as a backend ( dsmrreader :mqtt  => domoticz in => domoticz out => influxdb => grafana.

In grafana the timestamp seems to be correct. However the graphs in dsmrreader is 1 hour behind for the gas meter.

Wolverine65, 2018-01-12:
Uit een andere # 218 post haal ik het volgende

[quote]
@balk77 gasmeters geven hun meterstanden alleen elk uur door. Daarom zie je in de gasgrafieken puur het verschil in meterstanden tussen de meetpunten. De gasmeetpunten zijn in dit geval op hun kleinst een uur.
Dit is wel veranderd bij nieuwe gasmeters en er komt nog ondersteuning voor die kleinere intervallen.

Wanneer er bijvoorbeeld op de gas X-as 18:00 staat dan is dat verbruik in het uur van 18:00 tot 19:00 gemeten. Of simpel gezegd, het verschil tussen de meterstanden op 18:00 en 19:00.
[/quote]

dennissiemensma, 2018-01-13:
Bedankt voor je vraag. Ik kan me voorstellen dat het wat verwarrend overkomt. Je hebt zelf al een antwoord gevonden op hoe je de grafieken moet lezen. Specifiek zul je echter je in #40 moeten zoeken. Ik zal nog even toelichten waarom dat is. 

Van origine was het gasverbruik eerst andersom geplot. Dus het verbruik van `18:00` tot `19:00` stond genoteerd als `19:00`. 
Echter, het elektriciteitsverbruik had ik destijds (eerder) ingebouwd zoals het momenteel ook al is. Dus verbruik van bijvoorbeeld minuut `18:30` is eigenlijk het verbruik van `18:30` tot `18:31`.

Ik vond dat wat verwarrend overkomen omdat gas/elektra daardoor beide een andere manier van aflezen hadden. Daarom heb ik ze gelijkgetrokken in #40, en is de start van het uur het uur waar het verbruik over gaat.

dennissiemensma, 2018-01-13:
Mocht je nog meer vragen hebben dan verneem ik ze uiteraard graag.

closed 2018-01-13 by dennissiemensma