dsmrreader/dsmr-reader #423: Na update version 1.12 , na reboot 1.11

BBasss, 2018-01-14:
Hoi Dennis,

Nadat ik de software update naar 1.12 werkt dit correct zover ik kan beoordelen. Echter als ik mijn raspbery reboot, komt hij weer terug op 1.11 ? Wat doe ik fout ?

Klein ander vraagje: 
Moet ik voor "het DSMR5.0 probleem" nu /home/dsmr/dsmr-reader/dsmrreader/settings.py   aanpassen met: 
DSMRREADER_BACKEND_SLEEP = 60
DSMRREADER_DATALOGGER_SLEEP = 60

dennissiemensma, 2018-01-14:
**v1.11**
Wat zijn de laatste regels van de output als je dit doet:
```
sudo su - dsmr
cat dsmrreader/__init__.py
```
Er zou iets moeten staan als:
```
VERSION = (1, 12, 0, 'final', 0)

__version__ = get_version(VERSION)
```

**Settings**
Ze zijn juist zoals je ze daar hebt neergezet. Vergeet niet de applicatie/processen te reloaden. Dat kan simpelweg met `./reload.sh` als `dsmr` user.

dennissiemensma, 2018-01-14:
Ik kan je overigens aanraden eerst een lagere interval nemen en dat op te hogen totdat je een goede balans hebt bereikt qua data/performance. Bijvoorbeeld eerst:

```
DSMRREADER_BACKEND_SLEEP = 30
DSMRREADER_DATALOGGER_SLEEP = 15
```

BBasss, 2018-01-14:
Na de update idd:
VERSION = (1, 12, 0, 'final', 0)
__version__ = get_version(VERSION)

als ik daarna de raspbery reboot (gewoon sudo reboot) en daarna nogmals de version opvraag:
VERSION = (1, 11, 0, 'final', 0)
__version__ = get_version(VERSION)

dennissiemensma, 2018-01-14:
Ik heb het idee dat je SD-kaart de wijzigingen niet persistent opslaat. Ik zou verder geen andere oorzaak weten waarom fysieke bestanden na een reboot anders zouden zijn.

Ik heb voor de zekerheid hier ook een reboot gedaan en daarmee kon ik het (helaas) niet reproduceren.

dennissiemensma, 2018-01-14:
Worden de metingen en alles verder wel goed opgeslagen en bewaard?

BBasss, 2018-01-14:
Da's idd wat ik ook al dacht, maar heb er geen verklaring voor.
De metingen worden gewoon goed opgeslagen en bewaard.

Ik denk dat het dan maar eens tijd word om een nieuw sd kaartje te kopen en vanaf zero alles opnieuw op te zetten.

dennissiemensma, 2018-01-14:
Het lijkt me vreemd dat het zich beperkt tot één deel van de kaart. Je zou nog kunnen proberen om alles onder `/home/dsmr/dsmr-reader` weg te gooien, te rebooten (kijken of het terugkomt) en zo niet, het project opnieuw te klonen/installeren. De database kun je dan laten staan.

Al je instellingen staan verder in de database, met uitzondering van de twee regels eerder in dit ticket in de `settings.py`.

BBasss, 2018-01-15:
Hi Dennis,

Ik heb hetzelfde probleem met mn tweede applicatie die ik op de raspberry draai. (Pimatic) Als ik hier instellingen wijzig, blijven deze niet na een reboot.
Mijn kennis van Linux is te beperkt, waardoor het opnieuw beginnen met een nieuwe sd kaart de snelste oplossing is.

dennissiemensma, 2018-01-15:
Oke dan denk ik dat het inderdaad aan je kaartje ligt, helaas! 

Voor nu sluit ik dan dit ticket, maar als je nog ergens hulp bij nodig hebt, kun je hem uiteraard heropenen of een vraag stellen.

closed 2018-01-15 by dennissiemensma