dsmrreader/dsmr-reader #424: update naar v 1.12

dvd77, 2018-01-17:
Zojuist 2x getracht te updaten van v1.11 naar v1.12 zonder succes.

Ik heb het normale update process gevolgd maar na afloop staat er doodleuk:
 --- Deployed version: 
1.11

Iemand een idee ?

dennissiemensma, 2018-01-17:
Bedankt voor je vraag. 

* Is v1.11 de eerste versie die je hebt geinstalleerd?
* Wat krijg je te zien als je het volgende doet:
```
sudo su - dsmr
git fetch
git status
git tag
```
Zoals ik het hier heb:
* `git fetch` geeft geen output wanneer je helemaal bij bent (qua repo sync)
* `git status` geeft:
```
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
```
* `git tag` geeft een hele lijst, waaronder `v1.12.0`

Krijg je soortgelijke output?

dvd77, 2018-01-18:
1.11 is inderdaad de eerste versie en dit dus mijn eerste update

git status geeft: 

``` branch master
Your branch is behind 'origin/master' by 2 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   dsmrreader/config/base.py

no changes added to commit (use "git add" and/or "git commit -a") 
```

git tag geeft 1.12.0 wel weer in de lijst

dennissiemensma, 2018-01-18:
Het lijkt erop dat je lokaal wijzigingen hebt aangebracht in `dsmrreader/config/base.py`. Mogelijk weigert GIT daardoor automatisch te updaten.

Vermoedelijk wil je die wijzigingen overzetten naar `dsmrreader/settings.py`. Kun je eens kijken wat `git diff` als output geeft?

dvd77, 2018-01-18:
staat mij even niet bij dat ik wijzigingen heb aangebracht in base.py... maar 100% zeker ben ik ook niet.

git diff: 

```
diff --git a/dsmrreader/config/base.py b/dsmrreader/config/base.py
index 28c3b27..46f40d6 100644
--- a/dsmrreader/config/base.py
+++ b/dsmrreader/config/base.py
@@ -74,7 +74,7 @@ MIDDLEWARE_CLASSES = (
     'django.contrib.auth.middleware.AuthenticationMiddleware',
     'django.contrib.auth.middleware.SessionAuthenticationMiddleware',
     'django.contrib.messages.middleware.MessageMiddleware',
-    'django.middleware.clickjacking.XFrameOptionsMiddleware',
+#    'django.middleware.clickjacking.XFrameOptionsMiddleware',
     'django.middleware.security.SecurityMiddleware',
 
     # Local.
```

closed 2018-01-18 by dvd77

dennissiemensma, 2018-01-19:
Ok top, bedankt voor je terugkoppeling.