dsmrreader/dsmr-reader #426: Twee vragen

alwisiebers, 2018-01-20:
Mijn zoon heeft DSMR reader voor me geinstalleerd. Ik ben hier heel enthousiast over over. 
Twee vraagjes / verzoeken:
1. Ik heb het idee dat het tijdstip van de gasmeting / resultaten 1 uur achterloopt. Weeregegeven meting van bv 11 uur vindt volgens mij om 12 uur plaats. Klopt dit of is er een instelling verkeerd;
2. is het mogelijk om bij de overzichtsgrafieken (dag - maand - jaar) ook de temperatuurgegevens weer te geven?
Met een vriendelijke groet,
Alwi Siebers

dennissiemensma, 2018-01-20:
Hallo Alwi, bedankt voor je vragen.

* Alle grafieken m.b.t. verbruik hebben betrekking op de minuut of uur waarvoor ze geplot zijn. Dus als er staat `18:00`, dan betekent dat (voor gas) `van 18:00 tot 19:00`.

* Ik heb de temperatuurgegevens bewust alleen in de tabellen gezet, zodat het overzichtelijk blijft. De redenatie die ik er achter heb, is dat bij terugkijken in het verleden het vaak niet zo belangrijk is wat de exacte temperatuur in een bepaald uur was. Maar wel wat de min/max/gemiddelde temperatuur geweest is gedurende die dag (om bijvoorbeeld gasverbruik te verklaren). Daarom zijn alleen die gegevens zichtbaar.

Is dat afdoende voor je?

alwisiebers, 2018-01-20:
Dag Dennis,

Dank voor je snelle reactie. 
- Gastijden in het dashboard zijn duidelijk zo. Dank. Geldt dit ook voor de weergegeven gastijden in het archief? Lijkt me niet want dan zou er een uur tijdverschil ontstaan.
- Ik heb pas een warmtepomp en ben het gebruik van die hybride pomp i.r.t. onze verwarmingsketel en de thermostaat aan het inregelen. 
Een overzicht van elektriciteitsverbruik (warmtepomp - schakelt uit bij een temperatuur onder de 4 graden) en gasverbruik (CV ketel) per uur i.r.t. de temperatuur in dat uur zou mij daarbij helpen. 
Ik kan me jouw keuze i.v.m. de overzichtelijkheid echter ook voorstellen.

Met een vriendelijke groet,
Alwi Siebers

dennissiemensma, 2018-01-21:
* Dit geldt voor alle tijden, want voor grafieken waarbij ik 1 tijdstip moet kiezen zal het ofwel het begin van het uur zijn ofwel het eind. Ik heb in het verleden voor het eerste gekozen. Mogelijk dat ik ooit nog onderzoek of ik simpelweg iets kan tonen als `18:00-19:00`, maar volgens mij werd de grafiek daar niet veel beter van.

* Ik zal het overwegen om de temperatuur per uur inzichtelijk te krijgen.

alwisiebers, 2018-01-21:
Temperatuur per uur inzichtelijk zou heel fijn zijn.
Groet, Alwi

Verstuurd vanaf mijn iPad

alwisiebers, 2018-01-22:
> Op 21 jan. 2018 om 15:55 heeft Alwi Gmail <alwi.siebers@gmail.com> het volgende geschreven:

Dag Dennis,

Nog een verzoek.
Bij het exporteren van de gegevens naar excel zou het ook heel fijn zijn als de temperatuur gegevens per uur / het daggemiddelde meegegeven worden.
Ik hoop dat mijn praktische gebruikersvraagjes niet al te hinderlijk zijn voor je project.

Met een vriendelijke groet,
Alwi Siebers

Verstuurd vanaf mijn iPad

dennissiemensma, 2018-01-22:
Bedankt voor je feedback en geen zorgen hoor. Alle voorstellen waar ik wat in zie laat ik sowieso hier open staan. Op een later moment pak ik die dan op, of ik zie er van af, net hoe de situatie op dat moment is.

alwisiebers, 2018-03-16:
Dag Dennis,
Naast de temperatuurmetingen per uur en de gemiddelde dagelijkse temperaturen in ‘het archief’, is het misschien ook nog een idee om de ‘zonuren’ (zie:http://www.zonurencalculator.nl/sun_hours_calculation) in het archief op te nemen.
De temparaturen geven namelijk een mooi vergelijk met het gasverbruik en de zonuren met de productie / teruglevering van de zonnepanelen.
Tot voorlopig slot, door de koppeling met ‘minder gas’ krijg ik daar mooie grafieken i.r.t. de graaddagen. Is het nog een idee om ook de graaddagen in de DSMR reader op te nemen. 
Ook zonder al deze extra’s vind ik het nog steeds om je ‘reader’ te gebruiken.
Groet, Alwi Siebers

alwisiebers, 2018-03-16:
Foutje, ik wilde hem helemaal niet sluiten.

dennissiemensma, 2018-03-16:
Bedankt voor je toevoeging! Ik kan helaas geen externe data opnemen in dit project. 

En wat betreft graaddagen, je zou kunnen kijken naar `mindergas.nl`. Daar kun je dagelijks je gasmeterstanden naar laten oploaden door DSMR-reader en zij berekenen ook iets met graaddagen zag ik. Moeite waard om te proberen.

alwisiebers, 2018-03-16:
Dank voor je reactie.
Ik maak al gebruik van de upload naar ‘mindergas’, vandaar mijn vraagje over de graaddagen in jouw reader. Werkt ook prima, maar wel twee applicaties. Ik snap echter ook dat je geen externe data opneemt.
Ik zou het nog steeds wel mooi vinden als de temeratuurdata van jouw reader ook in het archief zichtbaar worden.

dennissiemensma, 2018-06-08:
Ik heb nog even gekeken of ik dit makkelijk kan toevoegen, maar dat valt helaas wat tegen, dus ik laat het hierbij.

alwisiebers, 2018-06-09:
Ok, dank voor de moeite.

dennissiemensma, 2018-12-11:
Voor de volgende release heb ik in ieder geval alvast het exporteren van de temperatuur in CSV-formaat toegevoegd. Voor het tonen in grafieken ben ik er nog niet helemaal uit.

alwisiebers, 2018-12-11:
Dag Dennis

Dat is al heel fijn. Dankjewel.
Als het ook nog in de grafieken komt, vind ik het helemaal super.

Groet, Alwi

dennissiemensma, 2019-09-10:
Ik heb besloten om dit niet verder uit te breiden. Ondanks dat ik goed snap dat de feature fijn voor je is, laat ik het bij alleen kunnen exporteren via de CSV.

closed 2019-09-10 by dennissiemensma