dsmrreader/dsmr-reader #427: Reconnect to postgresql

helmo, 2018-01-22:
After installing regular updates to postgress yesterday the dsmr backend stopped working.  I guess because postgresql was restarted.
Unfortunately I didn't notice it until this morning.
Could you add an attempt to reconnect in the exception handling?

Here'e the log:
```
 >>> Uncaught exception :: connection already closed
 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_consumption>> ::   File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 30, in run
    raise current_response

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/dispatch/dispatcher.py", line 228, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_consumption/apps.py", line 20, in _on_backend_called_signal
    dsmr_consumption.services.compact_all()

  File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 20, in compact_all
    for current_reading in DsmrReading.objects.unprocessed()[0:1024]:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 250, in __iter__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 1118, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 887, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 254, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 231, in _cursor
    return self._prepare_cursor(self.create_cursor(name))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 231, in _cursor
    return self._prepare_cursor(self.create_cursor(name))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/postgresql/base.py", line 220, in create_cursor
    cursor = self.connection.cursor()
  File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 30, in run
    raise current_response

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/dispatch/dispatcher.py", line 228, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_consumption/apps.py", line 20, in _on_backend_called_signal
    dsmr_consumption.services.compact_all()

  File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 20, in compact_all
    for current_reading in DsmrReading.objects.unprocessed()[0:1024]:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 250, in __iter__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 1118, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 887, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 254, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 231, in _cursor
    return self._prepare_cursor(self.create_cursor(name))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 231, in _cursor
    return self._prepare_cursor(self.create_cursor(name))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/postgresql/base.py", line 220, in create_cursor
    cursor = self.connection.cursor()

```

helmo, 2018-01-23:
De fout herhaald zich elke seconde.

Voor supervisord blijft de status op RUNNING staan.

`service postgresql restart` is genoeg voor mij om het te reproduceren.

-- 

Met vriendelijke groet / Regards,

Herman van Rink
Initfour websolutions

dennissiemensma, 2018-01-23:
Bedankt, ik heb het kunnen reproduceren en een mogelijke fix kunnen maken voor de volgende release. 

Langlopende commands (zoals de backend en datalogger) breken straks na elke run de DB-verbinding af, zodat Django geforceerd een nieuwe verbinding moet maken. Dat lost deze issues wel op.

closed 2018-01-23 by dennissiemensma

dennissiemensma, 2018-01-23:
De betreffende release is zojuist uitgebracht.