dsmrreader/dsmr-reader #428: ESP8266 werkt niet meer sinds update naar 1.13

Calimerorulez, 2018-01-25:
Na de update naar 1.13 werkt de esp8266 niet meer. Ik heb hem destijds aan de praat gekregen volgens #187

De error die ik krijg is:

Sending telegram to: http://<ip>:8081/api/v1/datalogger/dsmrreading
[!] API error: Invalid data

Ik zie het telegram wel binnenkomen in het script...

dennissiemensma, 2018-01-25:
Bedankt voor je melding. Wat geven de logs aan, staat daar nog een gedetaileerde foutmelding in?

http://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

Rayvw, 2018-01-25:
Ik liep tegen hetzelfde probleem aan en ben naar aanleiding daarvan nog even naar de foutmelding gaan kijken. Volgens mij heeft het te maken met het volgende: https://code.djangoproject.com/ticket/28201 (een null character in het telegram form veld). Sinds Django 2.0 resulteert dat is een validation error, zie ook: https://docs.djangoproject.com/en/2.0/ref/validators/#prohibitnullcharactersvalidator

Als ik de validation error log krijg ik deze foutmelding:
`{'telegram': [ValidationError(['Null characters are not allowed.'])]}`

dennissiemensma, 2018-01-25:
Bedankt voor je hint en uitzoekwerk @Rayvw! 

Ik zie inderdaad dat null bytes de validatie stukmaken. Helaas had ik daar nog geen benul van dat die data meekwam en daar dus ook geen test voor gemaakt. Ik zal kijken of ik een oplossing kan bedenken.

In de tussentijd kunnen jullie het beste teruggaan naar versie 1.12:
```
sudo su - dsmr
git fetch
git checkout tags/v1.12.0
./post-deploy.sh
```

Calimerorulez, 2018-01-26:
Dank voor de hulp alvast!

dennissiemensma, 2018-01-28:
Ik heb een mogelijke fix gemaakt in een aparte branch. Zouden jullie willen proberen of dat het oplost?

Je kunt wisselen met:

```
sudo su - dsmr
git fetch
git checkout origin/428-django2-null
./post-deploy.sh
```

Als het goed is komt de validatiefout dan niet meer voor. Mocht het toch niet werken, of loop je tegen iets anders aan, dan gelden de bovenstaande instructies nog steeds voor terugwisselen naar v1.12:

```
sudo su - dsmr
git fetch
git checkout tags/v1.12.0
./post-deploy.sh
```

Calimerorulez, 2018-01-28:
Getest en ziet er goed uit!

dennissiemensma, 2018-01-28:
Bedankt voor het testen. Ik zal dit dan releasen als opvolger en dan kunnen jullie ook weer gewoon de laatste versie draaien.

dennissiemensma, 2018-01-28:
De release is uit: https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.13.1

Als het goed is kunnen jullie weer switchen naar de `master` branch met:

```
sudo su - dsmr
git fetch
git checkout master
./deploy.sh
```

Versie is dan: `1.13.1`

closed 2018-01-28 by dennissiemensma

Calimerorulez, 2018-01-29:
Getest en werkt, dank je wel voor je mooie werk en hulp.