dsmrreader/dsmr-reader #429: 2de slimme meter uitlezen op dezelfde raspberry/server

RAdV1975, 2018-01-28:
Hallo,
Allereerst: top project!
Is het mogelijk om een 2de slimme meter uit te lezen met dezelfde raspberry/server en hierop ook een 2de installatie van dsmr te draaien?
Zo ja: wat moet daarvoor gedaan worden?

Met vriendelijke groet,
René de Vries

dennissiemensma, 2018-01-28:
Hallo René, bedankt voor je vraag.

In principe moet dat mogelijk zijn, al heb ik dat nog nooit in de praktijk gedaan. In principe zou je de complete applicatie kunnen dupliceren en dubbel draaien op dezelfde machine, inclusief Supervisor-processen.

Je zit dan wel met de volgende aanpassingen (uit mijn hoofd):
* Andere `tty` voor uitlezen meterkabel (is instelbaar in admin-interface)
* Andere databasenaam (is in te stellen in `settings.py`)
* Supervisor processen kopieren (is te dupliceren en te hernoemen in de configs in `/etc/supervisor/conf.d`, inclusief de paden)
* Upstream socket-pad van Nginx naar Supervisor/Gunicorn (is in te stellen in een kopie van de vhost in Nginx)
* Verder zou het kunnen zijn dat er nog bijwerkingen zijn, zoals automatische backup's (al pakken die volgens mij, bij toeval, de instellingen uit de `settings.py`).

Als je zelf erg handig bent in Linux dan zou je een poging kunnen wagen en dan ik kan mogelijk op afstand wat hints geven. Wie weet werkt het wel in één keer.

RAdV1975, 2018-01-28:
Hoi Dennis, dank voor je snelle reactie.

Ik zal het deze week eens uitproberen. Geheid dat er nog vragen uit komen door mijn beperkte Linux kennis. Zijn die er, dan hoor je van mij. Thnx!

dennissiemensma, 2018-03-05:
@RAdV1975 ben je hier inmiddels uitgekomen?

dennissiemensma, 2018-03-13:
Ik hoor het graag alsnog wanneer je er niet uitkomt of nog hulp nodig hebt.

closed 2018-03-13 by dennissiemensma

ttencate, 2018-05-19:
Is het mogelijk om meerdere meters aan dezelfde instance van DSMR Reader te hangen? Elk telegram bevat het serienummer van de meter, dus verschillende API keys zijn hiervoor niet eens noodzakelijk. Of is de database hier niet op ingericht?

Context: ik gebruik momenteel mijn eigen [bultje hacks](https://github.com/ttencate/prikmeter), maar nu zijn er meerdere (minder technische) mensen die mij gevraagd hebben of zij ook zo'n bultje hacks kunnen krijgen. Mijn server-code is daar weliswaar wel op berekend, maar heeft lang niet zo veel features als DSMR Reader, dus ik zit even te kijken of ik kan switchen.

dennissiemensma, 2018-05-19:
@ttencate helaas is DSMR-reader beperkt tot een enkele slimme meter, specifiek voor het uitlezen van de thuismeter met een Pi of een ander servertje. 

Ik heb geen rekening gehouden met meerdere meters, sterker nog, wanneer de meter bij iemand wordt vervangen, zijn de verbruiksgegevens van die dag incorrect. En ik vermoed dat het inbouwen van ondersteuning voor meerdere meters ook aardig wat uitdagingen oplevert.

Wellicht kun je wat met [Peter-Slump/energy-dashboard](https://github.com/Peter-Slump/energy-dashboard), ik dacht dat die wat meer scheiding had van meters, maar ik heb zelf geen ervaring met zijn software. In ieder geval moeite waard voor je om te kijken hoe dat werkt.

ttencate, 2018-05-21:
Okee, bedankt!

Hans-grietje1976, 2024-05-12:
Hé kan je zo slimme meter correctie doen ?