dsmrreader/dsmr-reader #430: Admin pagina onbereikbaar

EppoNL, 2018-01-31:
Goedendag Dennis,

Vanuit Tweakers werdt ik hierheen gestuurd met mijn probleem, mogelijk kan (en wil) je mij verder helpen.

Op mijn RPI3 staat nu het volgende:
- DSMR reader ([Deze installatie gebruikt](http://dsmr-reader.readthedocs.io/nl/latest/installation.html#method-a-quick-install)) (aangepast naar port 5000)
- Pi-Hole ([deze installatie gebruikt](https://pi-hole.net/))
- Home Assistant ([deze installatie gebruikt](https://home-assistant.io/docs/installation/raspberry-pi/))

Ik heb het ook in die volgorde geïnstalleerd. Nu leek het allemaal te werken, alleen werkt de admin pagina van DSMR reader niet meer. Ik krijg daar de volgende foutmelding:

```
Server Error
Exception:
TemplateDoesNotExist: admin/login.html
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 156, in _get_response
    response = response.render()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/template/response.py", line 106, in render
    self.content = self.rendered_content

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/template/response.py", line 81, in rendered_content
    template = self.resolve_template(self.template_name)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/template/response.py", line 63, in resolve_template
    return select_template(template, using=self.using)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/template/loader.py", line 47, in select_template
    raise TemplateDoesNotExist(', '.join(template_name_list), chain=chain)
```

Gezien mijn beperkte kennis zou ik niet weten wat er precies fout gaat. Het enige wat mijn opvalt is dat ik een andere versie van python heb geinstalleerd met Hass. In die handeleiding maak ik gebruik van python 3 ipv 3.5. Kan dat een probleem zijn? En zo ja, hoe los ik dat dan weer op?
Of kan het komen omdat ik DSMR reader heb verplaatst naar port 5000?

dennissiemensma, 2018-01-31:
Bedankt voor je melding. Als de rest van de applicatie het wél doet, vermoed ik niet dat het te maken heeft met ofwel poort 5000 ofwel de Python-versie. 

De applicatie wordt (voor een groot deel) [automatisch getest tegen meerdere Python-versies](https://travis-ci.org/dennissiemensma/dsmr-reader/builds/334466658) en 3.4 t/m 3.6 zouden het gewoon moeten doen. Standaard is `python3` niets meer dan een alias voor de geinstalleerde versie. Je kunt het zien door simpelweg `python3` te typen op je systeem.

**PC:**
```
(dsmrreader) dennis@pc:~/workspace/dsmr-reader$ python3
Python 3.5.2 (default, Nov 23 2017, 16:37:01) 
```

**Pi:**
```
(dsmrreader)dsmr@rpi3:~/dsmr-reader$ python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
```

----

Echter, dat gaat allemaal op voor de applicatie zelf en niet voor het onderliggende framework, Django, dat ik gebruik. De specifieke foutmelding `TemplateDoesNotExist: admin/login.html` geeft aan dat deze Django-template niet gevonden kan worden.

Je zou kunnen proberen om te dubbelchecken of Django goed geinstalleerd is door dit uit te voeren:
```
sudo su - dsmr

# Mits je DSMR-reader v1.12+ gebruikt
pip3 install --upgrade --force-reinstall django==2.0.1

# Het kan zijn dat je draaiende applicatie crasht tijdens de herinstallatie, dit forceert een reload.
./reload.sh
```

Ik zal ook nog even speuren naar andere oorzaken, ik kan niet direct een reden bedenken dat dit gebeurt.

dennissiemensma, 2018-01-31:
Ik heb het hier ook nog even geprobeerd qua poort, maar ik kan het issue helaas niet reproduceren.

In `/etc/nginx/sites-enabled/dsmr-webinterface` heb ik `80` veranderd naar `5000` en Nginx herstart. Dat ging zonder problemen en daarna draait de applicatie, inclusief admin-pagina, gewoon op poort 5000. Oftewel, ik denk dat dat sowieso geen oorzaak is.

EppoNL, 2018-01-31:
Herinstalleren lost het probleem inderdaad op. Wat ik nu fout heb gedaan weet ik niet, maar zou wel per ongeluk iets hebben verwijderd. Bedankt voor je snelle response.

dennissiemensma, 2018-01-31:
Fijn om te horen! Bedankt voor je terugkoppeling

closed 2018-01-31 by dennissiemensma