dsmrreader/dsmr-reader #432: [API] Gas cost missing at start of day

jwveldhuis, 2018-02-04:
It seems `gas` and `gas_cost` are missing from the JSON output until the next reading of the gas meter (which can take up to 1 hour). 

API: `/api/v2/consumption/today`
```json
{"electricity1":0.029,"total_cost":0.01,"electricity2":0.0,"day":"2018-02-04","electricity1_returned":0.0,"electricity2_cost":0.0,"electricity1_cost":0.01,"electricity2_returned":0.0}
```

Could these be initialized to 0.0?

dennissiemensma, 2018-02-04:
Bedankt voor je melding, het lijkt er inderdaad op dat [die gegevens verborgen worden tot de eerste meting](https://github.com/dennissiemensma/dsmr-reader/blob/v1.13.2/dsmr_consumption/services.py#L225). 

Ik weet niet meer precies of ik dat bewust zo heb gemaakt met een reden. Waarschijnlijk kan ik dat wel aanpassen.

dennissiemensma, 2018-03-05:
@jwveldhuis ik heb een mogelijke fix hiervoor gemaakt. Zou je die eens willen uitproberen?

```
sudo su - dsmr
git fetch
git checkout -b 432-api-missing-gas-start-of-day origin/432-api-missing-gas-start-of-day
./deploy.sh
```

Als het goed is krijg je dan altijd een nul-waarde wanneer gas mist.

jwveldhuis, 2018-03-05:
@dennissiemensma heb de versie zojuist deployed. Zal morgen kijken of ik nog errors in de log van de client zie!

jwveldhuis, 2018-03-05:
Meteen even na 00.00 gecheckt en het werkt naar behoren! Bedankt voor de fix!

dennissiemensma, 2018-03-06:
Bedankt voor het uitproberen, ik neem hem dan mee voor de volgende release. Tot die tijd kun je deze branch blijven draaien.

closed 2018-03-06 by dennissiemensma

dennissiemensma, 2018-03-11:
Uitgebracht in de [nieuwe release](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.14.0).