dsmrreader/dsmr-reader #434: Omit gas readings all together

henkjanwithaar, 2018-02-06:
Hi Dennis, first off: thanks for a great project!

Instead of natural gas I use an air-to-water heatpump for heating my house so I have no use for gas readings in the dashboard, statistics and trends. Would it be possible to omit these reading all together via a configuration option?

dennissiemensma, 2018-02-07:
Bedankt voor je verzoek! Ik dacht dat er al e.e.a. verborgen werd als er geen gasmetingen zijn, maar wellicht moet ik daar dan nog een keer opnieuw naar kijken. :]

henkjanwithaar, 2018-02-07:
Ik ben geswitched van gas naar een warmtepomp: wellicht dat het feit dat mijn meter geen 0-waarde doorgeeft ervoor zorgt dat dit gedrag optreedt?

dennissiemensma, 2018-02-07:
Hmm oke, dan gaat het niet werken nee. De applicatie kijkt of je uberhaupt ooit gas (of zonnepanelen) hebt gehad in het verleden, maar als je geswitched bent werkt dat niet. Dan mag het inderdaad een losse optie worden.

dennissiemensma, 2018-06-28:
@henkjanwithaar ik zat hier nog over te denken. Als er een optie komt om het gasverbruik te verbergen, dan worden ook alle gasmetingen en statistieken van je gas (periode vóórdat je warmtepomp gebruikte) verborgen in de applicatie. 

Heb je die gas-gegevens nog nodig of wil je die alsnog inzien later?

henkjanwithaar, 2018-06-29:
Goeie vraag. In mijn specifieke geval is het geen probleem wanneer historische data niet langer in te zien valt; ik ben DSMR pas gaan gebruiken nadat ik van gas naar luchtwarmtepomp ben geswitched (DSMR heeft dus slechts de eindwaarde uitgelezen).

dennissiemensma, 2018-06-29:
Oke bedankt voor de informatie, ik zal dan nog kijken wat handig is.

dennissiemensma, 2018-07-05:
Ik heb een opzet hiervoor gemaakt voor de volgende release. Met een instelling is dan een bepaald soort gebruik geforceerd uit te zetten.

dennissiemensma, 2018-07-11:
Het is nu beschikbaar met de nieuwe release.

henkjanwithaar, 2018-07-11:
Werkt feilloos. Dank Dennis!