dsmrreader/dsmr-reader #435: dsmr_datalogger keeps restarting, no logging anymore

hunter-nl, 2018-02-10:
dsmr_datalogger keeps restarting and does not log any telegram since update 1.13.2.

logging of dsmr_datalogger (every new line appears after every 5 seconds):
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...

dennissiemensma, 2018-02-10:
Bedankt voor je melding, welke versie draaide je vóór de upgrade?

hunter-nl, 2018-02-10:
1.13.1 Daarmee had ik ook al last hiervan, maar dat kwam heel af en toe voor en na reboot van systeem werkte het wel weer een tijdje. En daarvoor had ik 1.12.x en totaal geen problemen.

dennissiemensma, 2018-02-10:
Dat is vervelend om te horen, je bent helaas de eerste waarvan ik nu weet dat er iets mis lijkt te gaan. 

* Ik neem aan dat een reboot (of `sudo supervisorctl restart all`) het nu niet oplost?
* Weet je toevallig welke versie van Python je draait (`sudo su - dsmr` en dan `python --version`)?
* Is de datalogger de enige applicatie die die P1-poort uitleest via een kabel?
* Als je zelf deze kabel test via `cu`, dan neem ik aan dat je wel gewoon telegrammen ziet? (http://dsmr-reader.readthedocs.io/nl/latest/installation.html#your-first-reading-optional)

dennissiemensma, 2018-02-10:
Je kunt trouwens in de tussentijd terug naar 1.12 met [deze instructies](https://github.com/dennissiemensma/dsmr-reader/issues/428#issuecomment-360618251).

hunter-nl, 2018-02-10:
- Klopt, reboot / supervisor restart helpt niet.
- Python 3.5.3
- Ja
- Ja

dennissiemensma, 2018-02-11:
Bedankt voor de informatie. Ik heb nu een aftakking voor je gemaakt van de laatste versie, maar met wat extra debugging. 

Je kunt die draaien met:
```
sudo su - dsmr
git fetch
git checkout -b 435-datalogger origin/435-datalogger
./deploy.sh
```

Vervolgens zou je in de log van de datalogger een hoop extra informatie moeten zien over de flow van het proces.

Overigens kwam ik tijdens het testen net dezelfde situatie tegen als bij jou, qua logging, en toen zat er op een onbekende plek een fout. Ik kwam er dus achter dat er geen logging van die fouten aanstaat, dus dat heb ik nu ook aangepast. Mogelijk zie je dan direct al een fout, dat moet dan te zien zijn in een zin de begint met: ` [!] Exception raised in run(): `.

Ik hoor graag wat je in je logfiles ziet.

----

Je kunt altijd weer terug naar v1.12 met de instructies uit mijn eerdere comment hierboven.

hunter-nl, 2018-02-11:
Bedankt voor je support!

Ik heb zojuist de aftakking getest.

Ik krijg bijvoorbeeld het volgende in de log (stukjes even eruit geknipt):

- Data: b'1-0:1.8.2(0011\xb85.800*kWh)\r\n'
 [!] Exception raised in run(): 'utf-8' codec can't decode byte 0xb8 in position 14: invalid start byte
Command completed. Sleeping for 0.25 second(s)...
 - Data: b'1-0:1.8.2(001185.800*\xebWh)\r\n'
 [!] Exception raised in run(): 'utf-8' codec can't decode byte 0xeb in position 21: invalid continuation byte
Command completed. Sleeping for 0.25 second(s)...
- Data: b'/XMX5LGBBLA44\xb15306897\r\n'
 [!] Exception raised in run(): 'utf-8' codec can't decode byte 0xb1 in position 13: invalid start byte
Command completed. Sleeping for 0.25 second(s)...
- Data: b'0-0:1.0.0(180211165828\xd7)\r\n'
 [!] Exception raised in run(): 'utf-8' codec can't decode byte 0xd7 in position 22: invalid continuation byte
Command completed. Sleeping for 0.25 second(s)...

dennissiemensma, 2018-02-11:
Bedankt voor je input, dat is erg waardevol. Het is vreemd dat die rare characters in jouw datastroom voorbij komen. Het is mij onduidelijk waarom dat gebeurt en waar ze vandaan komen.

Ik heb echter een mogelijke fix gemaakt in de eerdere branch, zou je nogmaals willen wisselen en kijken of het nu beter gaat?

```
sudo su - dsmr
git checkout 435-datalogger
./deploy.sh
```

hunter-nl, 2018-02-12:
Het geeft geen error meer in de log. Alleen de telegrammen zijn niet ok. Ik zie ook op de Statistieken pagina dat de CRC niet meer kloppen van de telegrammen.

Voorbeeld telegram (ik heb serienummers verwijderd):
```
/XMX5LGBBLA4415XXXXXX

1-3:02.8(50)
0-0:1.0.0(180212122118W)
0-0:96.1.1(XXXXXXXXXXXXXXXX)
1-0:1.8.1(001331.930*kWh)
1-0:1.8.2001190955*kW)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.939*kW)
1-0:2.7.0(00.00*W)
0-0:96.7.21(00002)
0-0:96.7.9(00000)
**1-0:99.97.0(0)(0-0:96.7.19)**
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(236.0*V)
1-0:31.7.0(004*A)
1-0:21..0(00.939*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(XXXXXXXXXXXXXXXXXXXXXXXX)
0-1:24.2.1(180212122000W)(00483.333*m3)
!0975
Command completed. Sleeping for 0.25 second(s)...
```

Opvallend is dat in elke telegram nu de vette regel niet meer op twee regels staat. Nu worden elke telegram als foute CRC aangemerkt.

En af en toe valt een karakter weg.

dennissiemensma, 2018-02-12:
Bedankt voor je terugkoppeling, het is natuurlijk niet de bedoeling dat andere tekens ook gestript worden. Ik zal kijken voor een andere oplossing.

dennissiemensma, 2018-02-13:
Ik ben hier gisteravond nog even mee bezig geweest, maar tot nu toe nog geen oplossing kunnen vinden. Het is ook erg jammer dat ik het niet kan reproduceren met mijn eigen meter.

dennissiemensma, 2018-02-13:
Ik heb nog een wijziging voor je om uit te proberen, als het goed is `v1.13.3b3`. [Instructies hier](https://github.com/dennissiemensma/dsmr-reader/issues/435#issuecomment-364774410). 

Ik hoor graag van je of dit het voor je oplost.

hunter-nl, 2018-02-14:
b3 versie maakt het erger. regels met \r\n komen dan veelvuldig voor. de newline characters worden dan niet eruit gehaald waardoor de telegram ook niet correct is. Ik ga weer terug naar versie 1.12 want ik mis nu te veel informatie.
Mijn slimme meter stuurt elke seconde de telegrammen door. Misschien verwerkt de dsm_datalogger op mijn raspberry pi3 het niet snel genoeg waardoor telegrammen corrupt raken. Met CU commando zie ik namelijk alles gewoon goed voorbij komen.

dennissiemensma, 2018-02-14:
Bedankt voor het uitproberen. Voor nu heb ik dan helaas geen oplossing meer voor je. Het is mij een raadsel waarom je last hebt van die vreemde tekens, je bent ook de enige die er tot nu toe melding van heeft gemaakt.
Een raspberrypi 3 zou overigens snel zat moeten zijn hoor, het uitlezen van de poort kost relatief weinig kracht.

JSellis, 2018-03-04:
Het lijkt er op dat ik exact hetzelfde probleem heb als hunter-nl. Heb pas sinds eergisteren een slimme meter (Sagemcom XS210 ESMR5) en heb daarom geen ervaring met eerdere versies, maar zowel bij 1.13.2 als 1.12 speelt dit. De datalogger blijft restarten en in de log alleen maar 'Starting infinite command loop...'

Het enkele telegram dat af en toe nog verwerkt wordt, wordt genegeerd vanwege een CRC-mismatch. Ik draai Python 3.5.3. Een testmeting doen via `cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q` werkt gewoon prima. Als ik nog meer kan doen om te helpen testen/debuggen, laat maar weten.

dennissiemensma, 2018-03-04:
Bedankt voor je melding. Ik heb hier helaas nog geen oplossing voor. Je zou kunnen kijken of je een duidelijke foutmelding ziet wanneer je de `development` branch draait:

```
sudo su - dsmr
git fetch
git checkout -b development origin/development
./deploy.sh
```

JSellis, 2018-03-04:
Het draaien van de development-branch gaf inderdaad een duidelijkere foutmelding in de datalogger tail: bleek dat ik nog een ander script had draaien dat een conflict gaf. Die uitgezet en alles draait weer perfect, zowel op development als master (1.13.2).

dennissiemensma, 2018-03-04:
Bedankt voor het terugkoppelen, en daardoor tevens het uitsluiten dat het dezelfde oorzaak heeft.

dennissiemensma, 2018-06-08:
Ik heb hier helaas geen oplossingen of nieuwe inzichten meer voor, dus ik laat dit voor wat het is.

Het is jammer dat ik je verder niet kan helpen en dat je hierdoor aan v1.12 vast zit. Hopelijk geeft die versie je alsnog voldoende meter-informatie voor je eigen behoeftes. :]

closed 2018-06-08 by dennissiemensma