dsmrreader/dsmr-reader #437: After Reboot dsmr-reader is gone?

Kareltje1980, 2018-02-13:
Installed everything, made a small change to the sites-enabled of nginx, to run everything on port 81, as 80 was taken by pihole.  I was sending the data using a POST command from a different server, and that worked fine for 2 days.
WebGui was working with updates, backups to dropbox were working. 

Then I did a sudo reboot now, and all traces of DSMR are gone, nginx is gone, as is postgreSQL. I have no idea why this happened. 

Any advice would be welcome.

dennissiemensma, 2018-02-13:
Bedankt voor je ticket. Gezien je complete filesystem geraakt lijkt, vermoed ik dat je SD-kaart defect of corrupt is. Je kunt overigens de database-backup die je nog wél hebt, gebruiken bij de herinstallatie.

Het lijken namelijk de symptonen van dit ticket: https://github.com/dennissiemensma/dsmr-reader/issues/400#issuecomment-350582596

Kareltje1980, 2018-02-13:
Ja, na wat verder zoeken denk inderdaad dat dat het is. Bedankt voor het antwoord en de software!

closed 2018-02-13 by Kareltje1980