dsmrreader/dsmr-reader #438: Logs lopen snel vol en hoge load

Kareltje1980, 2018-02-16:
Ik heb een script dat via een http post de data naar de RPI met PiHole en DSRM stuurt. Dit draait nu een week en resultaten en webui werken prima. 
Alleen valt het me op dat de load erg hoog (1.2) is en in de logs zie ik een paar gekke dingen. Misschien hoor het, maar het viel me op.

supervisor.log (15MB) staat vol met
```
2018-02-16 21:05:30,282 INFO spawned: 'dsmr_datalogger' with pid 8810
2018-02-16 21:05:31,414 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2018-02-16 21:05:34,623 INFO exited: dsmr_datalogger (exit status 1; not expected)
2018-02-16 21:05:34,818 INFO spawned: 'dsmr_datalogger' with pid 8819
2018-02-16 21:05:35,939 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2018-02-16 21:05:39,224 INFO exited: dsmr_datalogger (exit status 1; not expected)
2018-02-16 21:05:39,330 INFO spawned: 'dsmr_datalogger' with pid 8828
2018-02-16 21:05:40,456 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2018-02-16 21:05:43,734 INFO exited: dsmr_datalogger (exit status 1; not expected)
```

dsmr_backend.log staat vol met (en er zijn al 5 bestanden):
```
 - Processed reading: 36924 @ 2018-02-16 20:40:59+01:00 (0.456 kW)
Command completed. Sleeping for 1 second(s)...

2018-02-16 20:40:55.096577+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-16 20:40:56.228303+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-16 20:40:57.355848+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-16 20:40:58.486376+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-16 20:40:59.616255+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-16 20:41:00.749435+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-16 20:41:01.877287+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-16 20:41:03.007334+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-16 20:41:04.137456+01:00: Calling backend services
 - Processed reading: 36925 @ 2018-02-16 20:41:09+01:00 (0.460 kW)
Command completed. Sleeping for 1 second(s)...
```

en de dsmr_datalogger.log staat vol (500k x) met:
```
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
Starting infinite command loop...
```

dennissiemensma, 2018-02-16:
Bedankt voor je ticket. Ik vermoed dat de datalogger telkens crasht, omdat die probeert te concurreren met PiHole. 

Als je de telegrammen via de API instuurt, is de datalogger niet nodig en kun je hem gerust uitzetten. Vermoedelijk komt de hoge load door het constant proberen om I/O uit te voeren. 

Uitzetten kan door:
*  `/etc/supervisor/conf.d/dsmr-reader.conf` te bewerken en daarin het volgende blok uit te commenten met `#` of het weg te halen:
```
[program:dsmr_datalogger]
command=/usr/bin/nice -n 5 /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger
directory=/home/dsmr/dsmr-reader/
pidfile=/var/tmp/dsmrreader--%(program_name)s.pid
user=dsmr
group=dsmr
autostart=true
autorestart=true
startsecs=1
startretries=100
stopwaitsecs=20
stdout_logfile=/var/log/supervisor/%(program_name)s.log
stdout_logfile_maxbytes=5MB
stdout_logfile_backups=5
```

* Vervolgens: `sudo supervisorctl reread` en daarna `sudo supervisorctl update`
* Als het goed is 'verdwijnt' het `dsmr_datalogger` proces en vermindert dat aanzienlijk de load. De telegramverwerking via de API bij ongewijzigd.

dennissiemensma, 2018-02-16:
Je hoeft je overigens niet druk te maken om de logfiles en dat het er 5 zijn. Zoals je kan zien in de Supervisor configuratie is dat de bedoeling en roteert (= verwijdert de oudste) hij de logbestanden wanneer de groter zijn dan 5 MB, max 5 bestanden:
```
stdout_logfile=/var/log/supervisor/%(program_name)s.log
stdout_logfile_maxbytes=5MB
stdout_logfile_backups=5
```

Kareltje1980, 2018-02-17:
Super, dat loste het probleem op! Thanks!

closed 2018-02-17 by Kareltje1980