dsmrreader/dsmr-reader #439: 502 Bad Gateway

mkruiver, 2018-02-20:
Helaas lijkt mijn dsmr installatie weer corrupt te zijn geworden. Sinds 09:05AM vanmorgen kreeg ik geen power usage data meer binnen op pvoutput. Gezien de production upload via Youless ook weg was gevallen ging ik er vanuit dat internet er even uit lag. Vanmiddag om 13:10 is de productie upload weet hervat, maar de consumptiedata via dsmr reader niet. 

Ik heb zojuist even een deploy.sh gedraaid op de pi, maar werd helaas met een "bad gateway" geconfronteerd. Dit doet me wel een beetje twijfelen aan de robuustheid van deze setup van dsmr reader op een pi.

dennissiemensma, 2018-02-20:
Vervelend om te horen. De "bad gateway" is puur een standaard foutmelding van Nginx, wanneer de achterliggende applicatieserver niet reageert, dus ik ben daar benieuwd naar de foutmelding van de applicatieserver (dsmr_webinterface) zelf.

Wat vertellen de verschillende logfiles van de dsmr-processen qua foutmeldingen, rond tijdstip van ontstaan (en nu als het nog steeds speelt)?

mkruiver, 2018-02-21:
Hij ligt er nog altijd uit en ik heb nog geen tijd om te kijken wat er stuk
is. Zal van het weekend even kijken als het lukt. Moet ik nog actie
ondernemen om iets veilig te stellen tegen overschrijven?

dennissiemensma, 2018-02-21:
Je zou de supervisor-logfiles kunnen bewaren, afhankelijk van hoe snel ze 5x roteren (en daarn verwijderd worden)

mkruiver, 2018-02-22:
Got them. Als ik tijd heb ga ik ze doorspitten.

markruys, 2018-02-23:
Ik had hetzelfde symptoom probleem na een crash van de server. Dit was voldoende om hem weer werkend te krijgen:

```
root@energie:/var/log/supervisor# cat '/var/tmp/gunicorn--dsmr_webinterface.pid'
1658
root@energie:/var/log/supervisor# ps waux | grep 1658
root      1658  0.0  0.3   5068  1596 ttyS0    Ss+  10:34   0:00 /sbin/agetty --keep-baud 115200 38400 9600 ttyS0 vt220
root      2347  0.0  0.3   3684  1560 pts/0    S+   11:58   0:00 grep 1658
root@energie:/var/log/supervisor# rm /var/tmp/gunicorn--dsmr_webinterface.pid
```

mkruiver, 2018-02-23:
Hmmmm, dat werkt bij mij helaas niet...

Server Error
Exception:
OperationalError: could not connect to server: Connection refused
	Is the server running on host "localhost" (::1) and accepting
	TCP/IP connections on port 5432?
could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 126, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 69, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 89, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 150, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/dashboard.py", line 30, in get_context_data
    frontend_settings = FrontendSettings.get_solo()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 487, in get_or_create
    return self.get(**lookup), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 397, in get
    num = len(clone)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 254, in __len__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 1179, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1062, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 255, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 232, in _cursor
    self.ensure_connection()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 216, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 216, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 194, in connect
    self.connection = self.get_new_connection(conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/postgresql/base.py", line 168, in get_new_connection
    connection = Database.connect(**conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/psycopg2/__init__.py", line 130, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)

dennissiemensma, 2018-02-23:
Ow, als poort `5432` niet bereikbaar is, dan is je database down en dan werkt niet van de applicatie meer. Zou je eens kunnen kijken of PostgreSQL gecrasht is of iets? En of die weer wil starten.

mkruiver, 2018-02-25:
Volgens mij gaat er inderdaad iets fout met de database.

```
pi@raspberrypimeterkast:~ $ sudo sudo -u postgres createuser -DSR dsmrreader
createuser: could not connect to database postgres: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?
pi@raspberrypimeterkast:~ $ sudo su postgres
postgres@raspberrypimeterkast:/home/pi$ test
postgres@raspberrypimeterkast:/home/pi$ exit
pi@raspberrypimeterkast:~ $ sudo service postgresql restart
pi@raspberrypimeterkast:~ $ sudo sudo -u postgres createuser -DSR dsmrreader
createuser: could not connect to database postgres: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?
pi@raspberrypimeterkast:~ $
```

dennissiemensma, 2018-02-25:
Kun je wat terugvinden in de logfiles van PostgreSQL over de reden dat die niet meer wil starten?

mkruiver, 2018-02-25:
GNU nano 2.7.4                     File: postgresql-9.6-main.log

2018-02-25 10:27:34.923 UTC [30910] FATAL:  data directory
"/var/lib/postgresql/9.6/main" has gro$

2018-02-25 10:27:34.923 UTC [30910] DETAIL:  Permissions should be u=rwx
(0700).

GNU nano 2.7.4                    File: postgresql-9.6-main.log.1

2018-02-23 12:23:52.437 UTC [419] FATAL:  data directory
"/var/lib/postgresql/9.6/main" has group$

2018-02-23 12:23:52.437 UTC [419] DETAIL:  Permissions should be u=rwx
(0700).

dennissiemensma, 2018-02-25:
Het lijkt erop dat de data-directory van PostgreSQL niet of onjuiste rechten heeft gekregen. Ik heb dat nog nooit in de praktijk gezien.

Ik kan de rechten uiteraard niet zien. Zou je eens willen proberen of dit het oplost:

`sudo chmod 700 /var/lib/postgresql/9.6/main`
`sudo service postgresql restart`

mkruiver, 2018-02-25:
Ja, gefixt!

:O

Maar hoe dit ooit door een power failure van de pi gekomen kan zijn? SD
kaartje semi-corrupted?

dennissiemensma, 2018-02-25:
Fijn! Ik durf er geen uitspraken over te doen eerlijk gezegd. Zorg er in ieder geval voor dat je ook af en toe een (Dropbox)backup van je database ergens apart opslaat. Dan weet je zeker dat je nooit te veel data kwijtraakt wanneer hij echt een keer corrupt is. :]

(of je moet de DB draaien op een HDD of externe server, maar kost wel wat meer tijd/complexiteit helaas)

closed 2018-02-25 by dennissiemensma

mkruiver, 2018-02-25:
Dropbox backup staat aan, ja.

Bedankt voor je support Dennis!