dsmrreader/dsmr-reader #444: Na vervanging meter DSMR Reader instabiel

markruys, 2018-02-24:
Mijn meter is vervangen, heb er nu eentje met DSMR 5.0 die iedere seconde een meting uitstuurt. Dus in de settings gezet:

```
DSMRREADER_BACKEND_SLEEP = 60
DSMRREADER_DATALOGGER_SLEEP = 60
```

Als ik zelf naar de /dev/ttyO1 kijk, ziet alles er goed uit. Toch krijg ik excepties, dus vang ik die af me deze code:

```
        try:
            # Make sure weird characters are converted properly.
            data = str(data, 'utf-8')
        except TypeError:
            pass
        except UnicodeDecodeError:
            dsmrreader_logger.info('UTF decode error: 0x{}'.format(''.join(format(x, '02x') for x in data)))
            data = ''
            pass
```

Dit wordt dan gelogd:

```
[2018-02-24 00:01:20,426] INFO @ services | UTF decode error: 0x9272ba72824282929ab2728252b2a50d0a
[2018-02-24 00:01:40,426] INFO @ services | UTF decode error: 0x72824282929ab2728252b2a50d0a
[2018-02-24 00:41:27,311] INFO @ services | UTF decode error: 0xaab2aa525a5da50d0a
```

In het statusoverzicht zie ik nu dat het gas niet meer uitgelezen kan worden, dat moet ik nog uitzoeken. 

P1 DSMR TELEGRAMMEN  worden wel goed bijgewerkt. Dat is gek, want de datalogger lijkt gestopt te zijn wegens een signal:

```
!1B7E
Command completed. Sleeping for 60 second(s)...

/XMX5LGBBLB2410064571

1-3:0.2.8(50)
0-0:1.0.0(180224003639W)
0-0:96.1.1(4530303336303033373839373338373136)
1-0:1.8.1(000001.787*kWh)
1-0:1.8.2(000007.809*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000005.273*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.294*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00014)
0-0:96.7.9(00000)
1-0:99.97.0(0)(0-0:96.7.19)
1-0:32.32.0(00000)
1-0:52.32.0(00000)
1-0:72.32.0(00000)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(0237.0*V)
1-0:52.7.0(0238.0*V)
1-0:72.7.0(0238.0*V)
1-0:31.7.0(0.53*A)
1-0:51.7.0(0.55*A)
1-0:71.7.0(0.21*A)
1-0:21.7.0(00.124*kW)
1-0:41.7.0(00.118*kW)
1-0:61.7.0(00.050*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!EEA3
Command completed. Sleeping for 60 second(s)...

Detected signal #15, exiting on next run...
Exited due to signal detection
```

Ook `/var/log/supervisor/dsmr_datalogger.log` wordt niet meer bijgeschreven. In `/home/dsmr/dsmr-reader/dsmrreader/provisioning/supervisor/dsmr-reader.conf` staat onder meer:

```
[program:dsmr_datalogger]
command=/usr/bin/nice -n 5 /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_datalogger
directory=/home/dsmr/dsmr-reader/
pidfile=/var/tmp/dsmrreader--%(program_name)s.pid
user=dsmr
group=dsmr
autostart=true
autorestart=true
startsecs=1
startretries=100
stopwaitsecs=120
stdout_logfile=/var/log/supervisor/%(program_name)s.log
stdout_logfile_maxbytes=5MB
stdout_logfile_backups=5
```

dsmreader v1.13.2

dennissiemensma, 2018-02-24:
Bedankt voor je melding. Dit lijkt enigszins op #435, waar ik helaas nog in het duister tast, dus ik ben heel blij dat je dit meldt.

**Excepties**
Zou je eens de data-regels willen printen, net onder `except UnicodeDecodeError:`? Ik ben benieuwd of ze er hetzelfde uitzien als in #435.

**Gas**
Ik mis in bovenstand telegram inderdaad de regel met de gasmeter, kan het zijn dat ze die niet goed hebben aangesloten? Wil je anders eens proberen met `du` of je daar dezelfde output krijgt, waarbij de gasmeter ook ontbreekt?
Zo zou die missende regel er uit moeten zien: `0-1:24.2.1(151110190000W)(00845.206*m3)`

**Signal**
Signal #15 is een extern signaal van buiten de applicatie.
`SIGTERM      15       Term    Termination signal`

Ik gebruik deze bijvoorbeeld in het deploy-script en Supervisor om [de applicatie graceful zichzelf te laten herstarten](https://github.com/dennissiemensma/dsmr-reader/blob/v1.13.2/dsmr_backend/mixins.py#L55) ([exiten](https://github.com/dennissiemensma/dsmr-reader/blob/v1.13.2/dsmr_backend/mixins.py#L79)). Dat zou dus niet zomaar moeten gebeuren.

**Versie**
In #435 lijkt dit issue (los van je missende gas) niet voor te komen in `v1.12`. Zou je eens willen proberen te downgraden, om te kijken of dat datalogger weer werkend maakt? [Instructies staan hier](https://github.com/dennissiemensma/dsmr-reader/issues/428#issuecomment-360618251) en je kunt [zo altijd weer maar master](https://github.com/dennissiemensma/dsmr-reader/issues/428#issuecomment-361095700).

Mocht blijken dat die versie ook bij jou goed loopt, dan zal ik een POC maken om weer te downgraden naar Django 1.11, om te kijken of dat een oorzaak is. Het lijkt me overigens vreemd, want in de datalogger komt Django pas laat in het proces in actie, maar wie weet.

markruys, 2018-02-24:
Ik ben teruggegaan naar 1.12, die draait stabiel. Ik krijg nog steeds geen gasmetingen door. Kan dat zijn dat dat is omdat ik wegens een DSMR 5 meter de leesinterval op 60 secs heb gezet en dan zo de uurmetingen mis? Ik zal eens teruggaan naar 0.25 sec delay, kijken wat er dan gebeurt.

dennissiemensma, 2018-02-24:
Je kunt even je meter handmatig uitlezen met DU om te kijken of daar wel gas in staat. Ik heb hier geen link, maar het staat in docs op installatiepagina.

markruys, 2018-02-24:
Je bedoelt CU ipv DU? Als ik kijk naar de gelogde DSMR berichtjes, dan zie ik daar geen m3 waardes tussen. Ik vermoed dat de gasmeter nog niet is geactiveerd op de vervangen slimme meter, misschien een weekje wachten.

dsmr-reader 1.12 met een interval van 60 of 10 secs lijkt stabiel te werken. Bij de default waarde van 0,25 hangt op bepaalt moment de datalogger. Het proces draait wel, maar de status pagina geeft aan dat er geen metingen meer binnenkomen.

dennissiemensma, 2018-02-25:
Ja sorry, ik was niet helemaal wakker, ik bedoelde inderdaad `cu`. Ik vermoed dat je gasmeter dan niet (goed?) is aangesloten. Je zou ze een berichtje kunnen sturen of dat normaal is.

Bedankt voor je terugkoppeling over de oudere versie. Ik ga dan wat verder zoeken in de verschillen tussen die versies. Ik neem aan dat je gewoon een RaspberryPi 3 draait?

markruys, 2018-02-25:
Nee, een BeagleBone, die is redelijk vergelijkbaar met een rPi3. Een BeagleBone heeft echter 512MB DDR3 800MHz (Pi 3 heeft 1GB LPDDR2). Ik heb echter geen OOM killers in de logs gezien.

markruys, 2018-02-25:
Oh ja, ik had je nog een mailtje gestuurd naar je github email-adres. Lees je dat ook? Een PM in Github is volgens mij niet mogelijk.

dennissiemensma, 2018-02-25:
Bedankt, ik heb je als het goed is toevallig net een reply gestuurd. Zou je, als je weer eens tijd hebt, op de v1.13 eens willen proberen om de `data` regels te printen? Ik zoek een compleet telegram, inclusief foutieve regels/tekens.

markruys, 2018-02-25:
De update gaf me 1.14b1.

Met als settings in `./dsmrreader/settings.py` alleen `DSMRREADER_LOG_TELEGRAMS = True`, kreeg ik:

```
[2018-02-25 22:22:34,148] INFO @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCTEIyNDEwMDY0NTcxDQoNCjEtMzowLjIuOCg1MCkNCjAtMDoxLjAuMCgxODAyMjUyMjIyMzFXKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMzM2MzAzMDMzMzczODM5MzczMzM4MzczMTM2KQ0KMS0wOjEuOC4xKDAwMDAyNy40ODgqa1doKQ0KMS0wOjEuOC4yKDAwMDAwNy44MDkqa1doKQ0KMS0wOjIuOC4xKDAwMDAxMS41ODgqa1doKQ0KMS0wOjIuOC4yKDAwMDAwNS4yNzMqa1doKQ0KMC0wOjk2LjE0LjAoMDAwMSkNCjEtMDoxLjcuMCgwMS4wNDkqa1cpDQoxLTA6Mi43LjAoMDAuMDAwKmtXKQ0KMC0wOjk2LjcuMjEoMDAwMTQpDQowLTA6OTYuNy45KDAwMDAwKQ0KMS0wOjk5Ljk3LjAoMCkoMC0wOjk2LjcuMTkpDQoxLTA6MzIuMzIuMCgwMDAwMCkNCjEtMDo1Mi4zMi4wKDAwMDAwKQ0KMS0wOjcyLjMyLjAoMDAwMDApDQoxLTA6MzIuMzYuMCgwMDAwMCkNCjEtMDo1Mi4zNi4wKDAwMDAwKQ0KMS0wOjcyLjM2LjAoMDAwMDApDQowLTA6OTYuMTMuMCgpDQoxLTA6MzIuNy4wKDIzNi4wKlYpDQoxLTA6NTIuNy4wKDIzNi4wKlYpDQoxLTA6NzIuNy4wKDIzNi4wKlYpDQoxLTA6MzEuNy4wKDAwMSpBKQ0KMS0wOjUxLjcuMCgwMDIqQSkNCjEtMDo3MS43LjAoMDAwKkEpDQoxLTA6MjEuNy4wKDAwLjM5MSprVykNCjEtMDo0MS43LjAoMDAuNDcyKmtXKQ0KMS0wOjYxLjcuMCgwMC4xODUqa1cpDQoxLTA6MjIuNy4wKDAwLjAwMCprVykNCjEtMDo0Mi43LjAoMDAuMDAwKmtXKQ0KMS0wOjYyLjcuMCgwMC4wMDAqa1cpDQohMURBOQ0K'
[2018-02-25 22:22:36,184] INFO @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCTEIyNDEwMDY0NTcxDQoNCjEtMzowLjIuOCg1MCkNCjAtMDoxLjAuMCgxODAyMjUyMjIyMzNXKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMzM2MzAzMDMzMzczODM5MzczMzM4MzczMTM2KQ0KMS0wOjEuOC4xKDAwMDAyNy40OTEqa1doKQ0KMS0wOjEuOC4yKDAwMDAwNy44MDkqa1doKQ0KMS0wOjIuOC4xKDAwMDAxMS41ODgqa1doKQ0KMS0wOjIuOC4yKDAwMDAwNS4yNzMqa1doKQ0KMC0wOjk2LjE0LjAoMDAwMSkNCjEtMDoxLjcuMCgwMS4wNDUqa1cpDQoxLTA6Mi43LjAoMDAuMDAwKmtXKQ0KMC0wOjk2LjcuMjEoMDAwMTQpDQowLTA6OTYuNy45KDAwMDAwKQ0KMS0wOjk5Ljk3LjAoMCkoMC0wOjk2LjcuMTkpDQoxLTA6MzIuMzIuMCgwMDAwMCkNCjEtMDo1Mi4zMi4wKDAwMDAwKQ0KMS0wOjcyLjMyLjAoMDAwMDApDQoxLTA6MzIuMzYuMCgwMDAwMCkNCjEtMDo1Mi4zNi4wKDAwMDAwKQ0KMS0wOjcyLjM2LjAoMDAwMDApDQowLTA6OTYuMTMuMCgpDQoxLTA6MzIuNy4wKDIzNi4wKlYpDQoxLTA6NTIuNy4wKDIzNi4wKlYpDQoxLTA6NzIuNy4wKDIzNi4wKlYpDQoxLTA6MzEuNy4wKDAwMSpBKQ0KMS0wOjUxLjcuMCgwMDIqQSkNCjEtMDo3MS43LjAoMDAwKkEpDQoxLTA6MjEuNy4wKDAwLjM5NSprVykNCjEtMDo0MS43LjAoMDAuNDYyKmtXKQ0KMS0wOjYxLjcuMCgwMC4xODcqa1cpDQoxLTA6MjIuNy4wKDAwLjAwMCprVykNCjEtMDo0Mi43LjAoMDAuMDAwKmtXKQ0KMS0wOjYyLjcuMCgwMC4wMDAqa1cpDQohMTExOQ0K'
[2018-02-25 22:22:38,170] INFO @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCTEIyNDEwMDY0NTcxDQoNCjEtMzowLjIuOCg1MCkNCjAtMDoxLjAuMCgxODAyMjUyMjIyMzVXKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMzM2MzAzMDMzMzczODM5MzczMzM4MzczMTM2KQ0KMS0wOjEuOC4xKDAwMDAyNy40OTEqa1doKQ0KMS0wOjEuOC4yKDAwMDAwNy44MDkqa1doKQ0KMS0wOjIuOC4xKDAwMDAxMS41ODgqa1doKQ0KMS0wOjIuOC4yKDAwMDAwNS4yNzMqa1doKQ0KMC0wOjk2LjE0LjAoMDAwMSkNCjEtMDoxLjcuMCgwMS4wNDUqa1cpDQoxLTA6Mi43LjAoMDAuMDAwKmtXKQ0KMC0wOjk2LjcuMjEoMDAwMTQpDQowLTA6OTYuNy45KDAwMDAwKQ0KMS0wOjk5Ljk3LjAoMCkoMC0wOjk2LjcuMTkpDQoxLTA6MzIuMzIuMCgwMDAwMCkNCjEtMDo1Mi4zMi4wKDAwMDAwKQ0KMS0wOjcyLjMyLjAoMDAwMDApDQoxLTA6MzIuMzYuMCgwMDAwMCkNCjEtMDo1Mi4zNi4wKDAwMDAwKQ0KMS0wOjcyLjM2LjAoMDAwMDApDQowLTA6OTYuMTMuMCgpDQoxLTA6MzIuNy4wKDIzNi4wKlYpDQoxLTA6NTIuNy4wKDIzNi4wKlYpDQoxLTA6NzIuNy4wKDIzNi4wKlYpDQoxLTA6MzEuNy4wKDAwMSpBKQ0KMS0wOjUxLjcuMCgwMDIqQSkNCjEtMDo3MS43LjAoMDAwKkEpDQoxLTA6MjEuNy4wKDAwLjM5NSprVykNCjEtMDo0MS43LjAoMDAuNDYyKmtXKQ0KMS0wOjYxLjcuMCgwMC4xODcqa1cpDQoxLTA6MjIuNy4wKDAwLjAwMCprVykNCjEtMDo0Mi43LjAoMDAuMDAwKmtXKQ0KMS0wOjYyLjcuMCgwMC4wMDAqa1cpDQohMkM2NQ0K'
[2018-02-25 22:22:40,150] INFO @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCTEIyNDEwMDY0NTcxDQoNCjEtMzowLjIuOCg1MCkNCjAtMDoxLjAuMCgxODAyMjUyMjIyMzdXKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMzM2MzAzMDMzMzczODM5MzczMzM4MzczMTM2KQ0KMS0wOjEuOC4xKDAwMDAyNy40OTEqa1doKQ0KMS0wOjEuOC4yKDAwMDAwNy44MDkqa1doKQ0KMS0wOjIuOC4xKDAwMDAxMS41ODgqa1doKQ0KMS0wOjIuOC4yKDAwMDAwNS4yNzMqa1doKQ0KMC0wOjk2LjE0LjAoMDAwMSkNCjEtMDoxLjcuMCgwMS4wNDUqa1cpDQoxLTA6Mi43LjAoMDAuMDAwKmtXKQ0KMC0wOjk2LjcuMjEoMDAwMTQpDQowLTA6OTYuNy45KDAwMDAwKQ0KMS0wOjk5Ljk3LjAoMCkoMC0wOjk2LjcuMTkpDQoxLTA6MzIuMzIuMCgwMDAwMCkNCjEtMDo1Mi4zMi4wKDAwMDAwKQ0KMS0wOjcyLjMyLjAoMDAwMDApDQoxLTA6MzIuMzYuMCgwMDAwMCkNCjEtMDo1Mi4zNi4wKDAwMDAwKQ0KMS0wOjcyLjM2LjAoMDAwMDApDQowLTA6OTYuMTMuMCgpDQoxLTA6MzIuNy4wKDIzNi4wKlYpDQoxLTA6NTIuNy4wKDIzNi4wKlYpDQoxLTA6NzIuNy4wKDIzNi4wKlYpDQoxLTA6MzEuNy4wKDAwMSpBKQ0KMS0wOjUxLjcuMCgwMDIqQSkNCjEtMDo3MS43LjAoMDAwKkEpDQoxLTA6MjEuNy4wKDAwLjM5NSprVykNCjEtMDo0MS43LjAoMDAuNDYyKmtXKQ0KMS0wOjYxLjcuMCgwMC4xODcqa1cpDQoxLTA6MjIuNy4wKDAwLjAwMCprVykNCjEtMDo0Mi43LjAoMDAuMDAwKmtXKQ0KMS0wOjYyLjcuMCgwMC4wMDAqa1cpDQohRjhCMA0K'
[2018-02-25 22:22:42,148] INFO @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCTEIyNDEwMDY0NTcxDQoNCjEtMzowLjIuOCg1MCkNCjAtMDoxLjAuMCgxODAyMjUyMjIyMzlXKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMzM2MzAzMDMzMzczODM5MzczMzM4MzczMTM2KQ0KMS0wOjEuOC4xKDAwMDAyNy40OTEqa1doKQ0KMS0wOjEuOC4yKDAwMDAwNy44MDkqa1doKQ0KMS0wOjIuOC4xKDAwMDAxMS41ODgqa1doKQ0KMS0wOjIuOC4yKDAwMDAwNS4yNzMqa1doKQ0KMC0wOjk2LjE0LjAoMDAwMSkNCjEtMDoxLjcuMCgwMS4wNDUqa1cpDQoxLTA6Mi43LjAoMDAuMDAwKmtXKQ0KMC0wOjk2LjcuMjEoMDAwMTQpDQowLTA6OTYuNy45KDAwMDAwKQ0KMS0wOjk5Ljk3LjAoMCkoMC0wOjk2LjcuMTkpDQoxLTA6MzIuMzIuMCgwMDAwMCkNCjEtMDo1Mi4zMi4wKDAwMDAwKQ0KMS0wOjcyLjMyLjAoMDAwMDApDQoxLTA6MzIuMzYuMCgwMDAwMCkNCjEtMDo1Mi4zNi4wKDAwMDAwKQ0KMS0wOjcyLjM2LjAoMDAwMDApDQowLTA6OTYuMTMuMCgpDQoxLTA6MzIuNy4wKDIzNi4wKlYpDQoxLTA6NTIuNy4wKDIzNi4wKlYpDQoxLTA6NzIuNy4wKDIzNi4wKlYpDQoxLTA6MzEuNy4wKDAwMSpBKQ0KMS0wOjUxLjcuMCgwMDIqQSkNCjEtMDo3MS43LjAoMDAwKkEpDQoxLTA6MjEuNy4wKDAwLjM5NSprVykNCjEtMDo0MS43LjAoMDAuNDYyKmtXKQ0KMS0wOjYxLjcuMCgwMC4xODcqa1cpDQoxLTA6MjIuNy4wKDAwLjAwMCprVykNCjEt [... cut ...]
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 62, in handle
    self.run(**options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 25, in run
    telegram = dsmr_datalogger.services.read_telegram()
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services.py", line 87, in read_telegram
    data = str(data, 'utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xaa in position 1: invalid start byte
```

Zoals je kan zien ging na een paar seconde de datalogger weer verder.

dennissiemensma, 2018-02-26:
@ualex73 totdat #225 gebouwd is, zul je bij wisseling van meters voor één dag een groot min-verschil zien in de dag/uurstatistieken. Je kunt dat zelf handmatig herstellen in de admin-interface. 
Ná die eerste dag lopen de metingen gewoon door alsof er niets aan de hand is. Hij kijkt altijd naar de vorige dag/metingen.

ualex73, 2018-02-26:
@dennissiemensma - Thanks, dan laat ik de statistieken gewoon staan en dan accepteer ik die ene mismatch dag ... Dus valt gelukkig mee ;-)

dennissiemensma, 2018-02-26:
@markruys bedankt voor het proberen! Dus het lijkt erop dat wanneer je dsmrreader v1.13 of 1.14b draait, **sommige** metingen mislukken door UTF-8 fouten, maar niet allemaal? Of begrijp ik het verkeerd?

Ik heb één van die base64 metingen hier bekeken en daar lijken geen gekke tekens in te staan.

dennissiemensma, 2018-02-26:
@ualex73 zie [hier voor instructies overigens](https://github.com/dennissiemensma/dsmr-reader/issues/242#issuecomment-273861795).

dennissiemensma, 2018-02-26:
Fijn dat ze je de gasmeter willen fixen voor je. Dan heb je straks weer volledige gegevens.

Wat betreft je theorie over dat de data naar verloop van tijd corrupt raakt, het is iets om uit te proberen. Ik heb in een branch een aanpassing gemaakt in het maximaal aantal keer dat de datalogger een telegram uitleest, voordat die zichzelf exit (Supervisor zou hem vervolgens weer moeten starten).

Wil je eens kijken of dat de situatie verandert? Voor nu heb ik hem op max `10` runs gezet, maar je kunt dat [zelf eventueel aanpassen](https://github.com/dennissiemensma/dsmr-reader/compare/development...444-datalogger-chars#diff-ed17b1d6c709e0b06ed9fdde8a69b936R15) in de datalogger om mee te spelen.

```
sudo su - dsmr
git fetch
git checkout -b 444-datalogger-chars origin/444-datalogger-chars
./deploy.sh
```

markruys, 2018-02-27:
Met de default van 10 runs, deed hij het een aantal uur goed:

```
# grep "Reached max number of runs (10)" /var/log/supervisor/dsmr_datalogger.log | wc
     50     350    2250

# tail /var/log/supervisor/supervisord.log
2018-02-27 01:12:33,158 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2018-02-27 01:14:26,870 INFO exited: dsmr_datalogger (exit status 0; expected)
2018-02-27 01:14:27,882 INFO spawned: 'dsmr_datalogger' with pid 6547
2018-02-27 01:14:28,886 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2018-02-27 01:16:17,132 INFO exited: dsmr_datalogger (exit status 0; expected)
2018-02-27 01:16:18,153 INFO spawned: 'dsmr_datalogger' with pid 6574
2018-02-27 01:16:19,169 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2018-02-27 01:18:12,721 INFO exited: dsmr_datalogger (exit status 0; expected)
2018-02-27 01:18:13,730 INFO spawned: 'dsmr_datalogger' with pid 6607
2018-02-27 01:18:14,734 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)

# ls -ltr /var/log/supervisor/
total 84680
-rw-r--r-- 1 root root 10490044 Feb 21 02:35 dsmr_backend.log.3
-rw-r--r-- 1 root root 10490844 Feb 22 06:09 dsmr_datalogger.log.3
-rw-r--r-- 1 root root 10489811 Feb 22 12:34 dsmr_backend.log.2
-rw-r--r-- 1 root root 10486916 Feb 23 11:41 dsmr_datalogger.log.2
-rw-r--r-- 1 root root 10488926 Feb 25 14:35 dsmr_datalogger.log.1
-rw-r--r-- 1 root root 10490982 Feb 25 19:00 dsmr_backend.log.1
-rw------- 1 root root        0 Feb 26 00:23 dsmr_backend-stderr---supervisor-RhJev7.log
-rw------- 1 root root        0 Feb 26 00:23 dsmr_webinterface-stderr---supervisor-FQs_xU.log
-rw------- 1 root root        0 Feb 26 00:23 dsmr_datalogger-stderr---supervisor-vsrNW_.log
-rw-r--r-- 1 root root   630845 Feb 26 23:42 dsmr_webinterface.log
-rw-r--r-- 1 root root  8798552 Feb 27 01:18 dsmr_datalogger.log
-rw-r--r-- 1 root root 11678135 Feb 27 01:18 supervisord.log
-rw-r--r-- 1 root root  2596951 Feb 27 09:31 dsmr_backend.log
```

Ik zal de max op 1 zetten en kijken wat er dan gebeurt.

dennissiemensma, 2018-02-27:
Bedankt voor het proberen! Dus dit probleem komt altijd pas na een tijdje voor? En het helpt niet dat die na 10x herstart?

dennissiemensma, 2018-05-02:
@markruys is je situatie nog gewijzigd? Ik neem aan dat je inmiddels ook gas gelogd ziet?

PieterRuys, 2018-05-02:
Dat het gas niet werkte werd veroorzaakt door een verkeerde gasmeter.

De versie die ik nu draai (1.12) doet het goed, ik heb niet meer met nieuwere versies getest.

dennissiemensma, 2018-05-02:
Bedankt voor de info. In dat geval kun je denk ik het beste op 1.12 blijven. 

Ik heb tot nu toe slechts één andere gebruiker gehoord met hetzelfde probleem, maar helaas nog geen oorzaak of oplossing kunnen vinden. Ik verwacht niet dat nieuwere versies het oplossen.

ualex73, 2018-05-04:
@markruys : Als je een Pi of een normale pc/laptop beschikbaar heb, zou je kunnen proberen of deze hetzelfde gedrag geven als met je Beagleboard. Indien niet, dan zou het die combinatie kunnen zijn (en dan wordt fixen lastig).

dennissiemensma, 2018-06-08:
Ik heb hier helaas geen oplossingen of nieuwe inzichten meer voor, dus ik laat dit voor wat het is.

Het is jammer dat ik je verder niet kan helpen en dat je hierdoor aan v1.12 vast zit. Hopelijk geeft die versie je alsnog voldoende meter-informatie voor je eigen behoeftes. :]

closed 2018-06-08 by dennissiemensma