dsmrreader/dsmr-reader #445: Upload/export to PVoutput doesn't work?

raoulteeuwen, 2018-02-24:
I think i've set up everything as required for an automatic upload to PVoutput, but even after more than 8 hours i don't see any data on PVoutput.org.

In dsmr_backend.log i find:

 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_pvoutput>> ::   File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 30, in run
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/apps.py", line 20, in _on_backend_called_signal
    dsmr_pvoutput.services.export()
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 34, in export
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 11, in should_export
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/apps.py", line 20, in _on_backend_called_signal
    dsmr_pvoutput.services.export()
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 34, in export
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 11, in should_export
 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_pvoutput>> ::   File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 30, in run
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/apps.py", line 20, in _on_backend_called_signal
    dsmr_pvoutput.services.export()
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 34, in export
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 11, in should_export
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/apps.py", line 20, in _on_backend_called_signal
    dsmr_pvoutput.services.export()
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 34, in export
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 11, in should_export
 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_pvoutput>> ::   File "/home/dsmr/dsmr-reader/dsmr_backend/management/commands/dsmr_backend.py", line 30, in run
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/apps.py", line 20, in _on_backend_called_signal
    dsmr_pvoutput.services.export()
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 34, in export
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 11, in should_export
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/apps.py", line 20, in _on_backend_called_signal
    dsmr_pvoutput.services.export()
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 34, in export
  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 11, in should_export

Not a big problem, but wondering how i can solve it.

dennissiemensma, 2018-02-25:
Bedankt voor je melding. Zou je eens willen wisselen naar de volgende branch, om te kijken of er een andere foutmelding getoond wordt in die zelfde logfile?

```
sudo su - dsmr
git fetch
git checkout -b 445-pvoutput-should-export-error origin/445-pvoutput-should-export-error
./deploy.sh
```

En zou dan een foutmelding gelogd moeten worden met iets als: `[!] Exception raised in run():......`

raoulteeuwen, 2018-02-25:
Dank. Na update/fetch krijg ik keurig

`--- Deployed version: 
1.14b2`

Output in /var/log/supervisor/dsmr_backend.log (reeks zoals):

`2018-02-25 17:51:19.058018+01:00: Calling backend services
 - Processed reading: 15271 @ 2018-02-25 17:50:48+01:00 (0.623 kW)
Command completed. Sleeping for 1 second(s)...

2018-02-25 17:51:20.270444+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-25 17:51:21.482692+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-25 17:51:22.699876+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-25 17:51:23.852581+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-25 17:51:25.008773+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-25 17:51:26.196807+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-25 17:51:27.405748+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-25 17:51:28.558723+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...

2018-02-25 17:51:29.779316+01:00: Calling backend services
 - Processed reading: 15272 @ 2018-02-25 17:50:58+01:00 (2.669 kW)
Command completed. Sleeping for 1 second(s)...

2018-02-25 17:51:31.034048+01:00: Calling backend services
Command completed. Sleeping for 1 second(s)...`

raoulteeuwen, 2018-02-25:
Ik krijg het gevoel dat hij zo mijn SD lekker gaat vullen ... moet ik dat stoppen en zo ja: hoe :-)?

raoulteeuwen, 2018-02-25:
Ik zie dat tussen alle 'tijdregels' soms 1 of meer regels staan als

`2018-02-25 17:53:59.949930+01:00: Calling backend services
 - Processed reading: 15285 @ 2018-02-25 17:53:08+01:00 (0.614 kW)
 - Processed reading: 15286 @ 2018-02-25 17:53:18+01:00 (0.611 kW)
 - Processed reading: 15287 @ 2018-02-25 17:53:28+01:00 (0.608 kW)
Command completed. Sleeping for 1 second(s)...
`

raoulteeuwen, 2018-02-25:
Sorry, ben nog niet zo handig met unix en zoeken van teksten in veel andere tekst. Als scrollend zie ik soms ook

2018-02-25 04:13:57.666227+01:00: Calling backend services
 - PVOutput | Uploading data @ 04:13
 [!] PVOutput upload failed (HTTP 401): Unauthorized 401: Invalid System ID
 - PVOutput | Delaying the next export until: 2018-02-25 03:18:58.525005+00:00
Command completed. Sleeping for 1 second(s)...

dennissiemensma, 2018-02-25:
Bedankt voor het uitproberen.

* In principe is die logging normaal en roteert hij de logfiles zelf (max 5 MB per stuk, max 5 x roteren voor verwijderen)
* De tijdregels zijn die van de metingen.

* Dit is echter de boosdoener voor jouw issue: `[!] PVOutput upload failed (HTTP 401): Unauthorized 401: Invalid System ID`

Ik denk dat je het SystemID niet of een ongeldige hebt ingevuld. Volgens mij is het een getal die je uit PVOutput kopieert. Zou je je instellingen willen controleren?

raoulteeuwen, 2018-02-25:
Excuses: ik had ipv systemID de systemName ingevuld :-|. Nu zie ik keurig data op PVoutput verschijnen. My bad.

closed 2018-02-25 by raoulteeuwen

dennissiemensma, 2018-02-25:
Top. Vergeet niet om terug te wisselen naar de laatste stabiele versie (v1.13.x):

```
su - dsmr
git checkout master
./deploy.sh
```

dennissiemensma, 2018-03-11:
Uitgebracht in de [nieuwe release](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.14.0).