dsmrreader/dsmr-reader #446: Teruglevering los van datalogger

trizz, 2018-02-27:
Mijn situatie is als volgt:

- Ik heb DSMR Reader op een server op het internet draaien
- Een datalogger op een Pi in de meterkast
- De datalogger stuurt netjes alle telegrammen door en worden verwerkt door DSMR Reader

Tot zo ver niets aan de hand.

Echter ben ik in het bezit van [Winddelen](https://www.windcentrale.nl/hoe-werkt-het/) en heb ik [een tooltje](https://github.com/trizz/windcentrale-cli) gemaakt die de huidige stand van zaken (+/- 5 a 10s vertraging) uitleest. Dat wil zeggen: de huidige opbrengst en de dag/maand/jaar opbrengst.  Wat mijn winddelen aan kWh opwekken, wordt in mindering gebracht op mijn energierekening. (belasting, transport, etc. moeten nog wel betaald worden). Dit kan dus gezien worden als zijnde 'teruglevering'.

Nu heb ik mijn tool uitgebreid dat bij elke nieuwe meting (dus om de +/- 6s) een [reading wordt gepost](http://dsmr-reader.readthedocs.io/en/latest/api.html#id2) naar m'n DSMR Reader. Alleen omdat dit natuurlijk helemaal los staat van mijn slimme meter in huis, post ik alleen `electricity_returned_2`, `electricity_currently_returned` en `timestamp` (de overige velden met een waarde `0`). En hier gaat een paar dingen niet helemaal lekker:

Omdat ik geen verbruik post, maar alleen teruglevering, krijg je effectief gezien ook een "0-meting" voor verbruik etc. In de grafieken per minuut zie je dit niet zo snel terug, maar als je ze op 'per meting' zet, zie je het heel duidelijk. Ook de tellers boven in beeld 'kloppen niet' (immers: er is _of_ 0 verbuik, _of_ 0 teruglevering).

Het zou mooi zijn als dit op een of andere manier "los te koppelen" is van elkaar, zodat ik via 2 "kanelen" data in kan schieten.

Daarnaast klopt de grafiek van teruglevering niet als het gegroepeerd op minuut is. Ik 'lees' de data in Watt, dus deel de waarde door 1000 om kW te krijgen. Echter in de grafiek zie ik dan bijv. 290 terugkomen als 29. Deel ik door 100, klopt de grafiek wel, maar ik weet niet precies waar het dan verder in het systeem eventueel verkeerd kan gaan.

trizz, 2018-02-28:
Na een kleine dagje draaien, blijkt ook ook het totaal "teruggeleverde" aantal niet te kloppen. De data die ik aanlever klopt wel, (zie screenshot: 11.696), maar het dashboard geeft 7.85 aan.

[image]

dennissiemensma, 2018-02-28:
Bedankt voor je uitgebreide ticket en leuk dat je zo enthousiast bent met koppelen.

Helaas is de applicatie puur geschreven voor het uitlezen van de meter(standen) of telegrammen. Er is geen ondersteuning voor de aanvoer van losse compensatie/teruglevering. 
Verder is het belangrijk om te weten dat:
* De API-velden zijn **meterstanden** en geen kWh som (behalve op de eerste installatiedag van je meter).
* Al het verbruik en teruglevering wordt gebaseerd op de **voorgaande** meting. Dus als je opeens nulstanden meegeeft, weet ik niet of de applicatie daar mee kan omgaan.

Wat betreft de tellers/grafieken, kloppen die wel wanneer je géén extra data inschiet via de API? De tellers bovenin zijn vrij simpel, ze tonen of er in de meest recente meting (terug)levering gerapporteerd is. 
Mogelijk kan die niet mee overweg wanneer je wisselend nulstanden aanlevert: 
* Je meter geeft vermoedelijk nulstanden voor teruglevering en plusstanden voor verbruik. 
* En je script geeft plusstanden voor teruglevering en nulstanden voor verbruik.

Vermoedelijk zijn de foute gegevens een combinatie van een of meerdere oorzaken hierboven. Gezien het project gemaakt is om alleen je meter uit te lezen (en die gegevens weer op zoveel mogelijk manier aan jou beschikbaar te stellen), kan ik hier helaas geen ondersteuning voor bouwen.

Ik hoor wel graag van je of de gegevens kloppen wanneer je niet nulmetingen opstuurt met je script. De gegevens uit de meter zouden dan moeten kloppen. Als dat niet zo is dan kijk ik graag voor je verder.

trizz, 2018-03-01:
Bedankt voor je uitgebreide antwoord! Ik heb inmiddels m'n eigen script er weer tussenuit gehaald, helaas gaat het op deze manier dan niet werken wat ik in gedachten had. Ik ga eens onderzoeken of ik de gewenste data voor het versturen van de telegram erbij in kan 'mergen', zodat DSMR Reader denkt dat alles van mijn meter af komt.

Wat betreft die nulmetingen; ik had niet het idee dat de gegevens vanaf m'n meter hierdoor beinvloed werden. De data/grafieken zagen er uit zoals ik verwachtte, zonder verder opvallende dingen. Wanneer de grafieken per meting getekend worden, zie je uiteraard wel een nulmeting er tussen, maar dat is te verwachten. Daarna 'springt' de grafiek weer naar de waarde van de meting erna.

dennissiemensma, 2018-03-01:
Dat moet opzich wel mogelijk zijn zoals je hierboven beschrijft. Je zou kunnen kijken of je je meter kan uitlezen met bijvoorbeeld https://github.com/ndokter/dsmr_parser en dan die resultaten opslaan in een (SQLite?) database of iets. 
En dan met je andere script die opgeslagen metingen verrijken, door de teruglevering van je windenergie toe te voegen aan die metingen. 
Om tot slot natuurlijk de metingen door te sturen naar DSMR-reader API, waardoor het net lijkt alsof je 'live' teruglevering hebt.

Het is wel complexer dan je initiele opzet, maar als je dit heel graag wilt, moet het mogelijk zijn. Mocht je nog ergens niet uitkomen dan hoor ik dat uiteraard graag van je.

closed 2018-03-01 by dennissiemensma