dsmrreader/dsmr-reader #448: COMPoort By-ID

Baseje, 2018-03-02:
Het zou mooi zijn als je ipv;
/DEV/TTYUSB1 / 2 / 0

De bij ID kan gebruiken;
DEV/SERIAL/BY-ID/P1 kabel...

Als dan namelijk de host herstart weet je zeker dat altijd het juiste apparaat word geselecteerd.

dennissiemensma, 2018-03-02:
Ik begrijp hier niet helemaal wat je bedoelt. Heb je last van wisselende toewijzigingen voor je kabel? En hoe kan ik dit nabootsen?

Baseje, 2018-03-02:
Ik heb in mijn Ubuntu machine meerdere USB apparaten zitten,
Daardoor komt het wel is voor dat TTYUSB1 opeens mijn RFXCom is ipv mijn P1-kabel.

In Home Assistant kan ik mijn apparaten aanduiden middels /dev/serial/by-id/"APPARAAT ID"
Hierdoor heb ik geen last van als er een reboot plaatst vind ik het het ID moet aanpassen.

Ik heb het ook in DSMR reader geprobeerd, maar dit lijkt niet te werken.

dennissiemensma, 2018-03-02:
Ik wist niet van het bestaan daarvan. Echter, de applicatie gebruikt `pyserial` om simpelweg te binden op het pad dat je opgeeft. 

In mijn geval werken beide varianten (net even geprobeerd + reboot):
* `/dev/ttyUSB0`
* `/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A5Z5VGV5-if00-port0`

Kan het zijn dat je de slash vooraan bent vergeten, zoals in je eerste comment?

Baseje, 2018-03-03:
Ik heb het samen met een collega geprobeerd,
En ook hebben we dit beide met een /dev/ ervoor.

Ik denk dat ik het al gevonden heb;
we draaien dit beide in een docker en als ik
docker exec dsmr ls /dev/ zie, zie ik serial niet staat.

Ik denk dus dat het niet zozeer in je applicatie ligt maar in de docker configuratie.

dennissiemensma, 2018-03-04:
Dat kan e.e.a. verklaren, bedankt voor je terugkoppeling!

closed 2018-03-04 by dennissiemensma