dsmrreader/dsmr-reader #449: Meterstatistieken via MQTT beschikbaar

f0nt4, 2018-03-05:
Zoals besproken in https://github.com/dennissiemensma/dsmr-reader/issues/367 zou ik graag de meterstatistieken via MQTT kunnen verzenden waarbij het mij in eerste instantie om de "Electricity tariff" gaat zodat ik in Home Assistant kan weergeven wat het huidige tarief is.

dennissiemensma, 2018-03-13:
@f0nt4 ik heb een opzet voor je gemaakt. Wil je eens proberen of dat afdoende voor je werkt? Het is toegevoegd als metere categorie 'meter statistieken per split topic' onder alle MQTT instellingen.

Testen kan met:
```
sudo su - dsmr
git fetch
git checkout -b 449-datalogger-statistics-via-mqtt origin/449-datalogger-statistics-via-mqtt
./deploy.sh
```

f0nt4, 2018-03-13:
@dennissiemensma Getest en lijkt allemaal te werken! :)
https://i.imgur.com/Jj3vral.png

dennissiemensma, 2018-03-13:
Oke top, bedankt! Ik neem aan dat dat niet de waarden zijn die ik zie, maar iets anders? Tarief hoort 1 of 2 te zijn.

f0nt4, 2018-03-13:
Klopt, dat zijn het aantal ontvangen mqtt berichten.

dennissiemensma, 2018-03-14:
Top, dan zal ik dit straks terugmergen naar de algemene development branch.

dennissiemensma, 2018-03-14:
Gemerged

closed 2018-03-14 by dennissiemensma

f0nt4, 2018-03-14:
Cool, thanks!

dennissiemensma, 2018-03-15:
Ik probeer elke X aantal weken wat te releasen, maar je kunt rustig op deze branch blijven, of wisselen naar `development`. Daar zit die al in.

f0nt4, 2018-03-15:
I see. Ik draai de applicatie in Docker, dus dat iets minder simpel dan gezegd :P

dennissiemensma, 2018-03-21:
Uitgebracht in [v1.15](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.15.0).

f0nt4, 2018-03-21:
Cool en thanks! Ik heb de docker-compose Dockerfile geüpdated.