dsmrreader/dsmr-reader #450: Meter vervangen... en nu?

jicho, 2018-03-08:
Hi,

Vandaag is de slimme gasmeter vervangen en begint de telling dus weer bij nul. DSMR geeft nu een negatief verbruik weer ;)

Hoe kan ik instellen dat de logica weer bij nul moet beginnen? Kan zo 1-2-3 niets vinden hierover...

Alvast bedankt voor de reactie.

dennissiemensma, 2018-03-08:
Bedankt voor je vraag! Zie #242. Ik heb hier momenteel helaas nog geen oplossing voor.

Als het goed is krijg je de dag erna wél de goede standen (hij kijkt telkens naar de voorgaande dag). 

Vervolgens zou je in de admin-interface eenmalig handmatig de foutieve dag kunnen corrigeren (zul je zelf even moeten rekenen). Meer informatie staat hier: https://github.com/dennissiemensma/dsmr-reader/issues/242#issuecomment-273861795

closed 2018-03-08 by dennissiemensma

jicho, 2018-03-09:
Duidelijk :) De volgende dag ging de telling inderdaad goed!

Iets minder duidelijk is wanneer ik op een gasmeting klik die ik wil aanpassen... 'k Heb in 1.13.2 wel een knop opslaan, maar geen veld om wijzigingen aan brengen (Voorpagina > Verbruik > Gasverbruik > datum/tijdstip).

Toen bleek dat ik niets aan kon passen heb ik gekozen voor verwijderen, het gevolg een mooie foutmelding:

```
Server Error

Exception:

PermissionDenied: 
Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 126, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/options.py", line 574, in wrapper
    return self.admin_site.admin_view(view)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/utils/decorators.py", line 142, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/sites.py", line 223, in inner
    return view(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/utils/decorators.py", line 62, in _wrapper
    return bound_func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/utils/decorators.py", line 142, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/utils/decorators.py", line 58, in bound_func
    return func.__get__(self, type(self))(*args2, **kwargs2)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1596, in changelist_view
    response = self.response_action(request, queryset=cl.get_queryset(request))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1330, in response_action
    response = func(self, request, queryset)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/actions.py", line 29, in delete_selected
    raise PermissionDenied
```

Enig idee hoe ik hieromheen kan?

dennissiemensma, 2018-03-10:
Je hoeft geen metingen aan te passen, alleen de dag- en uurstatistieken als het goed is. Die worden gebruikt in het historisch overzicht/archief.