dsmrreader/dsmr-reader #454: mqtt broker returns values with dots while dsmr has comma's

EvtK, 2018-03-17:
Hi,

At first: awesome program have you made. Many thanks.

I have activated the mqtt broker to get reading in Home Assistant. Got this pretty fast up and running, but notice that in home assistant I have dots, where as in dsmr reader I have comma's :).

So :

9,29 kWH -> 9.290 kWH

Now this might be a problem of the broker, but I don't know how to determine where this (hopefully) can be fixed.

Any guidance would really be appreciated.

Keep up the good work!

dennissiemensma, 2018-03-18:
Bedankt voor je vragen. 

Wat betreft de komma's en punten:
* De opslag, verwerking en uitwisseling (zoals naar HA) van alle gegevens is in Engels formaat. Dus met punten als decimaal-scheiding. Wellicht dat je het in HA kan omzetten naar Nederlands formaat.
* De weergave van gegevens in de applicatie is op basis van de taal in je browser. Als je browser Nederlands als voorkeur heeft, zal de dsmr-reader alles tonen in het Nederlands.

Wat betreft `electricity_currently_delivered`. Dat veld (en eigenlijk alles) [is opgeslagen in kW](https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_datalogger/models/reading.py#L47), net zoals de slimme meter dat communiceert. 
Het omzetten naar Watt is aan de applicatie die het toont. In het dashboard van dsmr-reader zet ik het handmatig om. Wellicht kun je dat ook doen in HA (ik heb zelf geen ervaring met HA).

EvtK, 2018-03-20:
Hi Dennis, bedankt voor je uitgebreide reactie. Ik ga aan de slag met een stukje conversie binnen HA. Mocht ik daar een goede oplossing voor vinden, dan zal ik dat hier terugkoppelen!

dennissiemensma, 2018-03-20:
Top, succes!

closed 2018-03-20 by dennissiemensma