dsmrreader/dsmr-reader #457: can't access Port

trekker25, 2018-03-21:
I  just removed the DSMR from Home Assistant and installed DSMR Reader and pointed it to the same USB port but all i get is this messages in the log:

 [!] Exception raised in run(): read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Command completed. Sleeping for 0.25 second(s)...

Landis G350 meter on port /dev/ttyUSB-P1 with protocol 4+.

As far as i know no other process is using the port?

What to do?

trekker25, 2018-03-21:
i changed the port to:

and now i also see this message:

/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A403ON88-if00-port0

Detected signal #15, exiting on next run...
Starting infinite command loop...
Detected signal #15, exiting on next run...
Starting infinite command loop...
Detected signal #15, exiting on next run...
Starting infinite command loop...
Detected signal #15, exiting on next run...
Starting infinite command loop...

what could be wrong?

dennissiemensma, 2018-03-21:
Bedankt voor je vragen. Heb je het toevallig al geprobeerd met `cu`?

http://dsmr-reader.readthedocs.io/nl/latest/installation.html#your-first-reading-optional

Hiervoor zul je de `dsmr_datalogger` wel even moeten stoppen.

trekker25, 2018-03-21:
-su: cu: command not found

lijkt me installatie toch niet goed gegaan?

dennissiemensma, 2018-03-21:
Je hebt hem niet perse nodig voor de applicatie, maar het is wel makkelijk om te testen of de rechten/verbinding voor de dsmr-gebruiker goed staan.

Je kunt hem installeren met: `sudo apt-get install -y cu`

trekker25, 2018-03-21:
Setting up cu (1.07-23) ...
Processing triggers for man-db (2.7.6.1-2) ...
pi@raspberrypi:~ $ cu -l /dev/ttyUSB-P1 -s 115200 --parity=none -E q
Connected.

en daar blijft het bij...?

dennissiemensma, 2018-03-21:
Je zult het wel moeten uitvoeren als `dsmr` gebruiker. Tijdens de installatie heb je als het goed is `dialout` rechten gegeven via groepen aan die gebruiker. Ik weet niet of het helpt, maar je kunt het proberen.

Hier werkt het in ieder geval met `cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q` (DSMR v4)

dennissiemensma, 2018-03-21:
Binnen 10 seconden zou je dan in ieder geval een telegram moeten zien. Na `Connected.`.

trekker25, 2018-03-21:
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Mar 21 16:57:14 2018 from 10.8.0.3
pi@raspberrypi:~ $ sudo su - dsmr
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ cu -l /dev/ttyUSB-P1 -s 115200 --parity=none -E q
Connected.

en gebeurt verder niks.

Datalogger heb ik gestopt via de website, of moet dat via de supervisor??

dennissiemensma, 2018-03-21:
Je kunt hem stoppen met `sudo supervisorctl stop dsmr_datalogger`.

Het blijft vreemd dat je geen data binnenkrijgt. Gebruik je toevallig een zelfgemaakte kabel of iets? En ik neem aan dat je met dezelfde (cu) instellingen het werkend had in HA?

trekker25, 2018-03-21:
heb ik nu gestopt en nog maals aangezet en zie alleen Connected.

onder HA nooit getest met commanda CU, deed alleen:

sensor:
  - platform: dsmr
     port: /dev/ttyUSB-P1

onder domoticz werkte dit ook prima. Is een gekochte kabel.

ik heb wel persistent USB devices aangezet:

SUBSYSTEM=="tty", ATTRS{idVendor}=="0658", ATTRS{idProduct}=="0200", SYMLINK+="ttyUSB-ZStick-5G"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="A403ON88", SYMLINK+="ttyUSB-P1"

in dit bestand:
/etc/udev/rules.d/99-usb-serial.rules

dus is een soort van SymLink die usb poort? Kan dit er mee te maken hebben?

bedankt alvast voor het meedenken!

Kan ik checken of die USB poort door iets anders wordt uitgelezen? (al heb ik die sensor in HA weg gehaald!)

dennissiemensma, 2018-03-21:
Ik moet bekennen dat ik op dit punt geen ideeen meer heb om te debuggen waar dit in zit. Tot nu toe lag het meestal ofwel aan de kabel ofwel aan de instellingen (DSMR protocol versie) en was het vrij snel te vinden.
Je zou nog een keer kunnen proberen te rebooten of de kabel eruit en erin doen. Mocht je dat nog niet geprobeerd hebben.

Verder kun je nog voor de zekerheid kijken welke seriele poort je moet hebben, maar ook dat heb je volgens mij al gedaan: `ls -l /dev/serial/by-id/`

trekker25, 2018-03-21:
kabel is nog goed.

sudo supervisorctl stop all

en HA opnieuw opgestart met dit in de config:
- platform: dsmr
  port: /dev/ttyUSB-P1 

en ik heb al mijn meter waardes weer in HA.
LET wel: als ik supervisor niet stop en start HA dan bleven de meter waardes leeg! (dus lijkt dat DSMR-reader wel iets op die poort aan het doen is.)

 Kan het toch door die mapping komen:

pi@raspberrypi:~ $ ls -l /dev/serial/by-id/
lrwxrwxrwx 1 root root 13 Mar 21 18:29 usb-0658_0200-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Mar 21 18:29 usb-FTDI_FT232R_USB_UART_A403ON88-if00-port0 -> ../../ttyUSB0

Hier staat die toch op ttyUSB0 ??? Terwijl ik in HA toch echt /dev/ttyUSB-P1  gebruik!?!

dennissiemensma, 2018-03-21:
En als je zowel DSMR-reader als HA helemaal stopt? Kun je hem dan wel uitlezen via `cu`? Je kunt zowel `/dev/ttyUSB-P1` als `/dev/ttyUSB0` een keer proberen.

trekker25, 2018-03-21:
sudo nano /etc/udev/rules.d/99-usb-serial.rules 
bestand leeg gemaakt en reboot van Rpi.

Home assistant en DSMR reader stopgezet.

pi@raspberrypi:~ $ sudo supervisorctl stop all
dsmr_webinterface: stopped
dsmr_backend: stopped
dsmr_datalogger: stopped
pi@raspberrypi:~ $ ls -l /dev/serial/by-id/
total 0
lrwxrwxrwx 1 root root 13 Mar 21 19:31 usb-0658_0200-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Mar 21 19:31 usb-FTDI_FT232R_USB_UART_A403ON88-if00-port0 -> ../../ttyUSB0
pi@raspberrypi:~ $ sudo su - dsmr
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q
Connected.

En niks aan Data? Terwijl het onder HA meteen werkt....?  

Nog opties om te debuggen?

trekker25, 2018-03-21:
heb um aan de gang. iets met rechten icm CU:

pi@raspberrypi:~ $ sudo cu -l /dev/ttyUSB0 -s 9600 --parity=none
cu: open (/dev/ttyUSB0): Permission denied
cu: /dev/ttyUSB0: Line in use
pi@raspberrypi:~ $ ls -la /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Mar 21 19:31 /dev/ttyUSB0
pi@raspberrypi:~ $ sudo chmod 666 /dev/ttyUSB0
pi@raspberrypi:~ $ sudo cu -l /dev/ttyUSB0 -s 9600 --parity=none
Connected.
/KMP5 ZABF001586898411

https://superuser.com/questions/794309/linux-tells-me-a-serial-port-is-in-use-but-it-isnt

Echter werkt ook alleen met -s 9600 als baudrate. is dat een probleem?

dennissiemensma, 2018-03-21:
Mooi! Echter je zult het moeten testen als `dsmr` gebruiker. Eerst `sudo su - dsmr` doen en dan `cu`. Als dat niet werkt, werkt de applicatie ook niet.

Een baudrate van `9600` lijkt me op een oudere versie van het DSMR-protocol. Wat geeft het eerste veld van het telegram aan als je die ziet met `cu`? Als het goed is zou je iets als `1-3:0.2.8(42)` moeten zien voor DSMR v4+.

trekker25, 2018-03-21:
de G350 van Landys blijkt dus wel het oude 2 protocol te hebben!

heb hem op /dev/ttyUSB0 met protocol 2 gezet en Rpi reboot gegeven en leest die dus wel in.

Misschien was het dus niet eens het rechten probleem, maar gewoon dat ik foute baudrate had... Dacht ook dat het een E350 was, maar dus eens ff in de meterkast gaan kijken!

Maar hartelijk voor zoveel hulp!

closed 2018-03-21 by trekker25

dennissiemensma, 2018-03-21:
Bedankt voor je terugkoppeling, helpt wellicht iemand in de toekomst met hetzelfde probleem!