dsmrreader/dsmr-reader #458: Vreemde PVOutput upload verschijnselen

mrvanes, 2018-03-28:
### Jouw omgeving
* DSMR-reader versie v1.15
* Hardware Intel(R) Atom(TM) CPU  230   @ 1.60GHz (cpuinfo)
* DSMR-protocol versie van slimme meter V2

### Bug, wens of iets anders
DSMR Reader slaat consequent rondom het uur twee PVOutpout upload tijden over in de data van de upload (de upload vindt wel plaats, alleen de tijden lopen niet op).
Hieronder is te zien dat rond 05:00 uur dsmr reader drie keer data uploadt voor het 05:00 uur tijdstip, terwijl het de tweede en derde keer al vijf of tien over is. Dit resulteert in hiaten in m'n pvoutput data.

Ik heb de output van de PVOutput backend service iets verboser gemaakt (alle data ipv alleen tijdstip) omdat ik wilde zien of dsmrreader misschien verkeerde getallen doorstuurt, wat niet het geval is, zo te zien.
```
Mar 28 04:50:37 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '04:50'), ('v3', 765), ('v4', 160)]
Mar 28 04:50:37 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 02:55:00.751691+00:00
Mar 28 04:55:39 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '04:54'), ('v3', 777), ('v4', 160)]
Mar 28 04:55:39 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 03:00:00.894626+00:00
Mar 28 05:00:42 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '05:00'), ('v3', 793), ('v4', 160)]
Mar 28 05:00:43 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 03:05:00.157231+00:00
Mar 28 05:05:44 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '05:00'), ('v3', 793), ('v4', 160)]
Mar 28 05:05:45 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 03:10:00.220794+00:00
Mar 28 05:10:46 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '05:00'), ('v3', 793), ('v4', 160)]
Mar 28 05:10:47 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 03:15:00.200521+00:00
Mar 28 05:15:48 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '05:14'), ('v3', 837), ('v4', 230)]
Mar 28 05:15:49 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 03:20:00.397554+00:00
Mar 28 05:20:50 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '05:20'), ('v3', 860), ('v4', 230)]
```
Wat wel opvalt is, is dat dsmrreader hier drie keer dezelfde gegevens uploadt, alsof er niets nieuws gezien is. Wellicht heeft m'n servertje het rond die tijd te druk met het verwerken van de uurlijkse statistieken? Ben benieuwd wat ik daar dan aan kan doen? Ik heb de backend en datalogger al wat minder actief gemaakt in settings.py:
```
DSMRREADER_BACKEND_SLEEP = 60
DSMRREADER_DATALOGGER_SLEEP = 60
```

mrvanes, 2018-03-28:
Hmm... ik zie wel drie keer een exception optreden in de metingen precies rond die tijd:
```
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: /XMX5XMXABCE000063492
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: #015
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 0-0:96.1.1(31333634303036392020202020202020)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 1-0:1.8.1(04428.237*kWh)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 1-0:1.8.2(04288.022*kWh)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 1-0:2.8.1(01878.522*kWh)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 1-0:2.8.2(04541.816*kWh)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 0-0:96.14.0(0001)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 1-0:1.7.0(0000.16*kW)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 1-0:2.7.0(0000.00*kW)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 0-0:96.13.1()
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 0-0:96.13.0()
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 0-1:96.1.0(3238303131303038333133333635333133)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 0-1:24.1.0(03)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: 0-1:24.3.0(180328040000)(2C)(60)(1)(0-1:24.2.0)(m3)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: (06718.892)
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: !
Mar 28 05:00:59 kronos dsmr_datalogger[10215]:  [!] Exception raised in run(): 'NoneType' object has no attribute 'group'
Mar 28 05:00:59 kronos dsmr_datalogger[10215]: Command completed. Sleeping for 60 second(s)...
```
Terwijl de meting er niet heel anders uitziet dan andere?

dennissiemensma, 2018-03-28:
Bedankt voor je bericht.

Wegens omstandigheden kan ik tijdelijk een aantal dagen niet achter de computer zitten (en makkelijker debuggen), dus zal ik je via mail proberen te helpen. Echter zie ik momenteel niet heel scherp, dus kom ik hier over een aantal dagen op terug.

Er is in versie 1.14 (dacht ik) iets veranderd dat de uploadtijden van pvoutput recht trekt. Je kunt even in de changelog kijken.

Mocht het echt je data verpesten, dan zou je tot die tijd een eerdere versie kunnen draaien.

Met vriendelijke groet,

Dennis

mrvanes, 2018-03-28:
M'n leven hangt er niet van af, ik kan wachten. Beterschap! Ben zelf ook python developer, dus kom wel een eind, moet alleen even in de backend duiken om te kijken welke variabele opeens onverwacht een "None" waarde krijgt in run() en waarom?

mrvanes, 2018-03-28:
Hmm... ik zal de CRC check eens uitzetten:
```
    try:
        content, crc = matches.groups()
    except AttributeError:
        # AttributeError: 'NoneType' object has no attribute 'groups'. This happens where there is not support for CRC.
        content = crc = None
```

mrvanes, 2018-03-28:
Ik heb 'm al gevonden:
In de regels waar het goed staat ziet m'n gas reading er zo uit:
```
0-1:24.3.0(180327090000)(08)(60)(1)(0-1:24.2.0)(m3)
```
En als het fout gaat zo:
```
0-1:24.3.0(180327090000)(2C)(60)(1)(0-1:24.2.0)(m3)
```
Het gaat mis op deze regular expression:
```
    legacy_gas_result = re.search(
        r'[^(]+\((\d+)\)\(\d+\)\(\d+\)\(\d+\)\([0-9-.:]+\)\(m3\)\(([0-9.]+)\)',
        legacy_gas_line
    )
```
De tweede component matcht dan niet op de \d+ omdat er een letter in staat.
Die \d moet dus een \w worden:
```
    legacy_gas_result = re.search(
        r'[^(]+\((\d+)\)\(\w+\)\(\d+\)\(\d+\)\([0-9-.:]+\)\(m3\)\(([0-9.]+)\)',
        legacy_gas_line
    )
```
Zal eens testen of 't helpt.

mrvanes, 2018-03-28:
Opgelost:
```
Mar 28 10:09:46 kronos dsmr_datalogger[23473]: 0-1:24.3.0(180328090000)(2C)(60)(1)(0-1:24.2.0)(m3)
Mar 28 10:09:46 kronos dsmr_datalogger[23473]: (06720.775)
Mar 28 10:09:46 kronos dsmr_datalogger[23473]: !
Mar 28 10:09:46 kronos dsmr_datalogger[23473]: Command completed. Sleeping for 60 second(s)...
```
```
Mar 28 09:55:06 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '09:55'), ('v3', 1965), ('v4', 60)]
Mar 28 09:55:07 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 08:00:00.296476+00:00
Mar 28 10:00:09 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '10:00'), ('v3', 1969), ('v4', 70)]
Mar 28 10:00:09 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 08:05:00.775297+00:00
Mar 28 10:05:11 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '10:04'), ('v3', 1973), ('v4', 130)]
Mar 28 10:05:11 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 08:10:00.815738+00:00
Mar 28 10:10:13 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '10:10'), ('v3', 1980), ('v4', 10)]
Mar 28 10:10:14 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 08:15:00.183541+00:00
Mar 28 10:15:15 kronos dsmr_backend[10681]:  - PVOutput | Uploading data [('d', '20180328'), ('delay', 2), ('n', 1), ('t', '10:15'), ('v3', 1983), ('v4', 400)]
Mar 28 10:15:16 kronos dsmr_backend[10681]:  - PVOutput | Delaying the next export until: 2018-03-28 08:20:00.270994+00:00
```

Ik neem aan dat ik hier geen PR voor hoef te maken, toch?

dennissiemensma, 2018-03-28:
Bedankt voor je terugkoppeling! Ik zal dit op een later moment teruglezen en verwerken.

Met vriendelijke groet,

Dennis

dennissiemensma, 2018-04-03:
Mooi dat je zelf al de oorzaak en oplossing hebt gevonden! Mijn testdata voor DSMR v2.x is beperkt, dus vandaar dat ik dit helaas gemist heb.

Ik zal het direct voor je verwerken.

dennissiemensma, 2018-04-03:
Ik heb de fix doorgevoerd in de `development` branch. Je kunt die eventueel tot de volgende release draaien.

```
sudo su - dsmr
git fetch
git checkout -b development origin/development
./deploy.sh
```

closed 2018-04-03 by dennissiemensma

dennissiemensma, 2018-04-04:
Uitgebracht in [v1.16](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.16.0).