dsmrreader/dsmr-reader #459: PVOutput laat de backend hangen

trekker25, 2018-03-29:
### Jouw omgeving
DSMR 1.14
Raspberry 3

### Bug, wens of iets anders
Om de zoveel tijd hangt bij mij het backend process. Supervisor laat hem wel zien als running maar worden geen telegrammen meer verwerkt.

Laatste log uit backend:
2018-03-28 14:30:00.357092+02:00: Calling backend services
 - Processed reading: 54370 @ 2018-03-28 14:29:09.438525+02:00 (0.640 kW)
 - Processed reading: 54371 @ 2018-03-28 14:29:19.939664+02:00 (0.640 kW)
 - Processed reading: 54372 @ 2018-03-28 14:29:30.451726+02:00 (0.610 kW)
 - Processed reading: 54373 @ 2018-03-28 14:29:40.973108+02:00 (0.610 kW)
 - Processed reading: 54374 @ 2018-03-28 14:29:51.476183+02:00 (0.610 kW)
 - PVOutput | Uploading data @ 14:30

Heb soms wel het idee dat dit gebeurd als de RPi ff zwaar belast wordt door bv mysql, echter daarna pakt die het niet meer op. Of het is een fout richting PVoutput. Herstart via Supervisor en gaat weer goed. Is dit op te lossen? En herstart supervisor via de webinterface zou ook al helpen, dan kun je vanaf de telefoon ook ff snel restart geven.

dennissiemensma, 2018-04-03:
Bedankt voor je melding. Ik zou helaas niet zo 1-2-3 een oorzaak weten als de logs ook verder geen fouten tonen.

* Staan er ook geen fouten in de andere logs? In `/home/dsmr/dsmr-reader/logs/django.log` bijvoorbeeld?

* Je geeft aan 'om de zoveel tijd', kun je er zelf een patroon of interval aan koppelen?

* Het is (toevallig) mogelijk om Supervisor te ontsluiten via een webinterface. Daar heeft die tool zelf ingebouwde ondersteuning voor. Het is zo makkelijk als dit: 
-- ZOrg dat je sudo/root gebruiker bent
-- Open `/etc/supervisor/supervisord.conf` met een teksteditor (zoals `vi`)
--- Voeg een nieuw blok toe (je kunt de regels met user/password weglaten als je netwerk gesloten/beveiligd is:
```
[inet_http_server]
port=0.0.0.0:9001
username=test
password=testtest
```
-- Herstart het hoofdproces: `sudo supervisorctl reload`
-- Als het goed is kun je er nu bij via de URL naar DSMR-reader, op de poort die je opgegeven hebt (`9001`?): http://ip-adres-van-je-pi:9001

trekker25, 2018-04-06:
ik heb nu de PV Output uitgeschakeled en draait nu al 3 dagen zonder
probleem. Dus lijkt toch echt daar ergens in te zitten.

Maar ga mijn opwek via ander proces uploaden naar PVoutput, data kabel
rechtstrreeks aan de omvormer.

evengoed bedankt!

dennissiemensma, 2018-04-06:
Staan er toevallig nog fouten in die andere logfiles? Tot nu toe ben je de enige die hier last van lijkt te hebben.

trekker25, 2018-04-09:
Heb hem voor nu uitgezet en sinds donderdag geen vastloper. Zal binnenkort nog eens testen met pv output@

dennissiemensma, 2018-06-08:
@trekker25 heb je nog een update over dit issue? Lijkt de upload naar PVOutput de oorzaak?

trekker25, 2018-06-09:
Ja zolang pvoutput niet draait draait dsmr dagen/weken zonder issues.

 

Met pvoutput aan ging die achter lopen.

dennissiemensma, 2018-06-09:
Bedankt voor je update. Je bent helaas momenteel (nog) de enige bij wie het structureel niet werkt. 

Wil je nog hier verder mee debuggen of vind je de huidige situatie prima?

 Als je verder wilt debuggen mag je PVOutput weer aanzetten en dan in de logfiles kijken of je een fout ziet hier:
* `/home/dsmr/dsmr-reader/logs/django.log`
* `/var/log/supervisor/dsmr_backend.log`

Daarnaast kun je nog updaten naar de laatste versie (v1.18) zodat het geen issue is dat ondertussen opgelost is. :]

dennissiemensma, 2018-07-23:
@trekker25 Wil je nog hier verder mee debuggen of vind je de huidige situatie prima? Gezien je eerder aangaf het anders te hebben ingericht.

trekker25, 2018-07-23:
Beetje warm opnzolder bij de pc. Maar ik zal er werk van maken!

Verstuurd vanaf mijn iPhone

dennissiemensma, 2018-07-23:
Je hoeft hiervoor niet perse in deze hitte dit uit te zoeken hoor, geniet lekker van het weer buiten. ;-)

Maar dan laat ik je ticket nog openstaan en weet ik dat ik zelf geen actie hoef te ondernemen.

dennissiemensma, 2018-07-23:
Overigens is het goed om te weten dat in de latere releases now ook de up te loaden data inzichtelijk is in de backend-logfile. Als je tzt kijkt, staat het als het goed is met de tekst als `PVOutput | Uploading`.

dennissiemensma, 2018-08-27:
@trekker25 er zijn inmiddels weer nieuwe releases beschikbaar. 

Er zitten geen specifieke fixes in voor dit probleem, maar je kunt het altijd nog even uitproberen wanneer je er weer aan toekomt.

dennissiemensma, 2018-09-18:
@trekker25 speelt dit ook nog bij de meest recente release?

trekker25, 2018-09-19:
Ik ben over gegaan naar een NUC maar heb nog probleem de oude database geimporteerd te krijgen, ik hou je op de hoogte!

trekker25, 2018-09-26:
Loopt nu al enkele uren, echter is er maar 1x geupload om 11:00 en daarna niet meer terwijl interval op 5 minuten staat.

dennissiemensma, 2018-09-26:
* Alle instellingen die je via de interface doet, staan ook in de database. 
* Kun je aangeven welke meldingen je exact kreeg bij de import van je backup? Eventueel zou je een tweede database ernaast kunnen maken (bijvoorbeeld `temp_dsmrreader`), en daar de import nogmaals kunnen doen.

trekker25, 2018-09-26:
Tweede import wil ik wel proberen. 

Maar waar in dsmr wordt de koppeling gemaakt tussen welke DB wordt gebruikt?

Verstuurd vanaf mijn iPhone

dennissiemensma, 2018-09-26:
Dat staat in de setting.py, maar ik doel meer op het reproduceren van de meldingen bij import, mocht je ze eerder niet opgeslagen hebben.

Ter voorbeeld een import die goed lukt:
```
postgres@pc:~$ dropdb dsmrreader ; createdb -O dsmrreader dsmrreader ; psql -f dsmrreader-postgresql-backup-*.sql dsmrreader
SET
SET
psql:dsmrreader-postgresql-backup-Monday.sql:10: ERROR:  unrecognized configuration parameter "idle_in_transaction_session_timeout"
SET
SET
 set_config 
------------
 
(1 row)

SET
SET
SET
CREATE EXTENSION
COMMENT
SET
SET
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE TABLE
[... 511 lines cut ...]
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
CREATE INDEX
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
```

trekker25, 2018-09-26:
Ok moet ik even uitzoeken hoe dit met docker zit om bij de bestanden te komen. 

Als ik een goede import doe, dan moet ik meteen alle archief data kunnen zien? En aantal telegrams bij statistieken?

Verstuurd vanaf mijn iPhone

dennissiemensma, 2018-09-26:
Ja. Wat nog wel zou kunnen, is dat je DB-dump van een oude DSMR-reader versie is. Dan kun je het beste na import even `./deploy.sh` doen zodat die ook alle migraties en alles uitvoert.

Het is trouwens belangrijk om te weten dat je de import alleen kunt doen op een **lege database** waar niets in zat. DSMR-reader moet dan ook helemaal gestopt zijn als je de import doet, om te voorkomen dat die nieuwe data wegschrijft tijdens de import.

trekker25, 2018-09-26:
Bij een drop verwijder je toch de hele database of maakt die hem alleen leeg?

 

Ik ga morgen even puzzelen, zeker omdat het icm docker is.

dennissiemensma, 2018-09-26:
Een drop verwijdert inderdaad de gehele database.

trekker25, 2018-09-26:
Maar je kunt dus wel een import doen na een drop database? Omdat ik dan geen lege DB om mee te starten. 

Verstuurd vanaf mijn iPhone

dennissiemensma, 2018-09-26:
Je kunt hem droppen en dan weer aanmaken en dan importeren. Zie het command dat ik gebruik in mijn voorbeeld hierboven.

trekker25, 2018-09-27:
Via docker is dit de te volgen procedure:

 

Or drop the database and restore a backup:

*	docker-compose stop dsmr
*	docker exec -t dsmrdb dropdb dsmrreader -U postgres
*	docker exec -t dsmrdb createdb -O dsmrreader dsmrreader -U postgres
*	cat /tmp/<your_dump>.sql | docker exec -i dsmrdb psql -U postgres
*	docker-compose start dsmr

 

En als .sql file gebruik ik een dropbox backup. De out van dit commando at /tmp/<your_dump>.sql | docker exec -i dsmrdb psql -U postgres naar een tekst bestand geeft dit:

SET

SET

SET

SET

SET

SET

SET

SET

CREATE EXTENSION

COMMENT

SET

SET

SET

ALTER TABLE

ALTER TABLE

ALTER SEQUENCE

ALTER TABLE

ALTER TABLE

ALTER SEQUENCE

ALTER TABLE

ALTER TABLE

ALTER SEQUENCE

ALTER TABLE

ALTER TABLE

ALTER TABLE

ALTER SEQUENCE

ALTER TABLE

[... 1146 lines cut ...]
ERROR:  constraint "auth_group_permissions_group_id_b120cbf9_fk_auth_group_id" for relation "auth_group_permissions" already exists

ERROR:  constraint "auth_permission_content_type_id_2f476e4b_fk_django_co" for relation "auth_permission" already exists

ERROR:  constraint "auth_user_groups_group_id_97559544_fk_auth_group_id" for relation "auth_user_groups" already exists

ERROR:  constraint "auth_user_groups_user_id_6a12ed8b_fk_auth_user_id" for relation "auth_user_groups" already exists

ERROR:  constraint "auth_user_user_permi_permission_id_1fbb5f2c_fk_auth_perm" for relation "auth_user_user_permissions" already exists

ERROR:  constraint "auth_user_user_permissions_user_id_a95ead1b_fk_auth_user_id" for relation "auth_user_user_permissions" already exists

ERROR:  constraint "django_admin_log_content_type_id_c4bce8eb_fk_django_co" for relation "django_admin_log" already exists

ERROR:  constraint "django_admin_log_user_id_c564eba6_fk" for relation "django_admin_log" already exists

joris@nuc:~$

 

En als ik dan dsmr en dsmrdb weer opstart dan is de database wel 180mb groot echter geen settings of oude data terug te halen?

 

Is:

trekker25, 2018-09-27:
Dennis,

 

Ik heb iets gevonden. Als ik een docker database maak (heeft gister een dag gedraaid) en terug zet gaat het wel goed. De sql file die in dropbox wordt gedaan pakt die niet.

 

https://github.com/xirixiz/dsmr-reader-docker

 

backup maken:

 

*	docker-compose stop dsmr
*	docker exec -t dsmrdb pg_dumpall -c -U postgres > /tmp/dump_date +%d-%m-%Y""%H%M%S.sql
*	docker-compose start dsmr

 

 

en terug zetten:

 

*	docker-compose stop dsmr
*	docker exec -t dsmrdb dropdb dsmrreader -U postgres
*	docker exec -t dsmrdb createdb -O dsmrreader dsmrreader -U postgres
*	cat /tmp/<your_dump>.sql | docker exec -i dsmrdb psql -U postgres
*	docker-compose start dsmr

hier wordt ook het commando psql gebruikt alleen net op iets andere manier (mijn linux kennis is ook vrij beperkt)

 

de layout is ook gewoon net anders.

De vraag is nu of ik die paar maanden data maar als verloren moet zien, of is die dropbox sql backup op een of andere manier toch te gebruiken?

trekker25, 2018-09-27:
Hmmm maybe toch de versie van de DB. Ik ga dit proberen!

trekker25, 2018-09-27:
./deploy.sh wil natuurlijk niet zo zonder kennis van docker en linux etc.

 

Denk dat met nieuwe db van start gaan maar beter is 😉

dennissiemensma, 2018-09-27:
Als je een reply stuurt via mail, dan mag je denk ik even het bericht waarop je replied weghalen, want anders is telkens je prive-emailadres hier zichtbaar. Ik haal het nu elke keer weg, maar het is makkelijker als je het voorkomt.. ;-)

dennissiemensma, 2018-09-27:
De meldingen zoals 
```
ERROR: constraint "django_admin_log_user_id_c564eba6_fk" for relation "django_admin_log" already exists
```

Geven aan dat de tabellen of data al bestaat in de database. Het kan twee oorzaken hebben:
* Of de database is niet leeg.
* Of de dump bevat dubbele data (zou vreemd zijn).

Wat je nog kunt doen is even in de database kijken of je gegevens ziet in de tabellen `dsmr_stats_daystatistics` en `dsmr_stats_hourstatistics`. Dat zijn de belangrijkste gegevens voor je historie en als je die kunt terughalen, heb je je gegevens voor het grootste deel terug (sowieso in Archief).

trekker25, 2018-09-28:
Ik zag dat er nog iemand uit nederland was met dit probleem:

 

 

https://github.com/xirixiz/dsmr-reader-docker/issues/31#issuecomment-425237617

 

Als ik mijn DB bekijk, zit er wel voor 180MB aan data in, alle lossen en ook de dag statistieken. Echter DSMR herkent ze niet?

 

Gr. Joris

dennissiemensma, 2018-09-28:
Hmmm, ik vind het lastig eerlijk gezegd. Het zou letterlijk gewoon zo simpel moeten zijn als de paar commando's uitvoeren zoals in de handleiding. Wanneer ik zelf ontwikkel dan restore ik zeer vaak op mijn ontwikkel-PC een back-up van mijn Pi-database, zodat ik historische data heb om mee te werken en dat gaat altijd goed.

Echter, omdat het hier om Docker gaat, gaan mijn bovenstaande ervaringen niet meer op, want ik heb zelf nog nooit docker hiervoor gebruikt. 

Wat je nog eventueel kan doen is:
* Export maken van alle gegevens die in het archief gebruikt worden: `pg_dump dsmrreader -t dsmr_stats_daystatistics -t dsmr_stats_hourstatistics --data-only`
* Database verwijderen en opnieuw aanmaken (leeg).
* Gewoon de handleiding volgen alsof het een schone installatie is, dus `./manage migrate` uitvoeren later in het traject. Nu wordt de database gevuld met de lege applicatie (de structuur is er, maar geen data).
* Vervolgens de backup uit stap 1 terugzetten.
* Nu zou je in het archief je oude gegevens moeten zien. Je mist dan wel alles tussen het moment van je initiele backup en je de applicatie weer zelf laat meten. Maar dat doe je nu vermoedelijk ook al.

Daarmee gooi je al je eerdere metingen overboord, maar heb je in ieder geval de dag- en uurgegevens nodig. En die zijn eerlijk gezegd toch het belangrijkste (historisch gezien).

dennissiemensma, 2018-09-28:
Daarnaast kun je natuurlijk ook gewoon de reactie van Xiriz afwachten in het door jou gelinkte ticket, gezien iemand anders dus exact hetzelfde ervaart. Mogelijk ziet hij waar het issue (met docker?) zit.

trekker25, 2018-10-01:
Ik ga kijken of dat nog lukt, anders beginnen we gewoon nu weer met nieuwe data 😊

trekker25, 2018-10-01:
Oeps email adres weer…

dennissiemensma, 2018-10-01:
Je kunt nog even kijken naar de reactie van `xirixiz` in het issue dat je hierboven hebt gelinked.

dennissiemensma, 2018-10-06:
Zie ook zijn update: https://github.com/xirixiz/dsmr-reader-docker/issues/31#issuecomment-426620113

trekker25, 2018-10-22:
> Mocht ik nog iets voor je kunnen doen dan hoor ik het graag en heropen ik het issue weer voor je.

Bedankt! Ik laat het dan weten!