dsmrreader/dsmr-reader #462: Get live usage trough API

Cyw00d, 2018-04-05:
Hi there, I really enjoy using the application, thanks for developing this and keeping it up to date!
I would like to get the live usage data trough the API. So not for today, but for this very moment I do the request, including (if entered) the costs.

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.14`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3B+`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v4`

dennissiemensma, 2018-04-05:
Bedankt voor je verzoek. Bedoel je met huidig verbruik de Wattage en kosten per uur, die rechtsbovenin op het dashboard getoond wordt?

Cyw00d, 2018-04-05:
Hoi Dennis, ja dat is precies wat ik bedoel! Dit zou ik graag willen zien:

dennissiemensma, 2018-04-05:
Ik zal er t.z.t. naar kijken. Voor gas is het denk ik niet mogelijk, omdat daar geen live verbruik (of afgeleide) van is.

swerveshot, 2018-04-30:
Is dit niet wat je bedoelt @Cyw00d?

[Example 2 - Fetch latest reading](http://dsmr-reader.readthedocs.io/en/latest/api.html#example-2-fetch-latest-reading)

Bij mij toont de API de volgende output, waarbij de 'electricity_currently_delivered' property overeenkomt met wat je normaal in de web interface ziet:
``` json5
{
    "count": 9423134,
    "next": "http://MYHOSTNAME/api/v2/datalogger/dsmrreading?limit=1&offset=1&ordering=-timestamp",
    "previous": null,
    "results": [
        {
            "id": 9459935,
            "timestamp": "2018-04-30T21:04:40+02:00",
            "electricity_delivered_1": "1911.333",
            "electricity_returned_1": "1.551",
            "electricity_delivered_2": "2135.992",
            "electricity_returned_2": "0.000",
            "electricity_currently_delivered": "0.442",
            "electricity_currently_returned": "0.000",
            "phase_currently_delivered_l1": null,
            "phase_currently_delivered_l2": null,
            "phase_currently_delivered_l3": null,
            "extra_device_timestamp": "2018-04-30T21:00:07+02:00",
            "extra_device_delivered": "2016.226"
        }
    ]
}
```

OK, de actuele kosten zitten hier dan nog niet bij en de performance kan wat tegenvallen als je geen retentie hebt ingesteld (zie issue #469)

Cyw00d, 2018-05-01:
Hoi Dennis,

Bedankt dit is wat ik tot nu toe gebruikte, alleen zit hier niet het Gas
verbruik bij helaas maar verder werkt dit prima!

mvg

Kevin

dennissiemensma, 2018-05-01:
Het was iemand anders die je hier @-tagt hoor, maar bedankt voor de info ;-)

swerveshot, 2018-05-02:
Dat was ik. Guilty as charged. :blush:
Gewoon een random eigenwijze Hollander die iets dacht te kunnen bijdragen in dit topic.

Cyw00d, 2018-05-02:
Bedankt Rudolf! Ik heb inmiddels live data op mijn touch screen staan, dus
ben al redelijk blij met wat ik nu heb :-)

swerveshot, 2018-05-02:
Oh, da's wel gaaf. Hoe heb je dat uiteindelijk gedaan nu dan? Via de API of een ander truukje? Ga zelf ook een beetje knutselen met een Raspberry Pi en een LCD scherm. Dus ik ben benieuwd wat je hebt verzonnen.

Cyw00d, 2018-05-02:
Ik heb een Angular 5 applicatie gemaakt - Gewoon via API. Screenshot kun je
hier bekijken:

https://gathering.tweakers.net/forum/list_message/55034935#55034935

swerveshot, 2018-05-03:
Oef, Angular klinkt als een te stijle learning curve voor mij. Ik ben nog wat aan het rommelen op een Raspberry Pi en een breadboardje om wat output op een LCD scherm te tonen. Verbruik van de slimme meter via dsmr-reader vond ik daar wel een leuke toepassing voor. Gebruik Python aangezien ik die taal 
 toch beter wil leren kennen.

Maar leuk project! :+1:

dennissiemensma, 2018-06-08:
@Cyw00d ik heb een kale opzet gemaakt, wil je die eens proberen?

Het makkelijkste is wanneer je al op de laatste versie draait, v1.18.

En dan vervolgens wisselt naar de branch:
```
sudo su - dsmr
git fetch
git checkout -b 462-live-electricity-api origin/462-live-electricity-api
./deploy.sh
```

Extra info:

[image]

swerveshot, 2018-06-11:
Met 2,4 miljoen readings in de database heeft mijn [Odroid](http://www.hardkernel.com/main/products/prdt_info.php?g_code=g138745696275) er ongeveer anderhalve seconde voor nodig om output te genereren. De nieuwe API call doet dit in 50-100 ms.

### Oude API call
[image]

### Nieuwe API call
[image]

Absoluut een verbetering en inderdaad goed geschikt om op een dashboard te gebruiken zoals @Cyw00d van plan is.

dennissiemensma, 2018-06-11:
Bedankt voor het uitproberen! In dat geval zal ik hem opnemen voor de volgende release.

Cyw00d, 2018-06-11:
Goed bezig!

dennissiemensma, 2018-06-11:
Gemerged naar `development`

closed 2018-06-11 by dennissiemensma

dennissiemensma, 2018-06-12:
Bij deze in [v1.19](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.19.0).

swerveshot, 2018-06-12:
Wow! Jij gaat als een raket met je releases.