dsmrreader/dsmr-reader #463: MQTT: Telegram als JSON, verkeerde tijd

sanderdw, 2018-04-06:
### Jouw omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.16
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3+ & Docker
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5'

### Bug, wens of iets anders
MQTT: Telegram (als JSON) uitvoer geeft een verkeerde tijd weer (het is nu 13:22): 
"timestamp": "2018-04-06T11:22:59Z"
RAW output geeft goede waarde:  0-0:1.0.0(180406132257S)

dennissiemensma, 2018-04-06:
Bedankt voor je verzoek. In principe klopt de tijd wel, want de tijdzone staat erachter (`Z` / Zulu / UTC). Dat komt doordat de database timezone-aware is.

Ik zal kijken of ik er de lokale tijd van kan maken, al breekt dat mogelijk wel bestaande implementaties van gebruikers.

dennissiemensma, 2018-04-09:
Ik begrijp het helemaal. Waarschijnlijk maak ik dan een optie waarbij je zelf kunt kiezen of UTC of de lokale tijdzone gebruikt wordt.

dennissiemensma, 2018-07-24:
@sanderdw ik heb bovenstaande oplossing gemaakt. Wil je deze eens proberen?

Normaal push ik het naar een aparte branch, maar nu zit het al per ongeluk in de algemene `development` branch. Maakt overigens verder niet uit, er zitten echter al meer features voor de volgende release in, dan alleen dit ticket.

Uitproberen kan met:
```
sudo su - dsmr
git fetch
git checkout -b development origin/development
./post-deploy.sh
```

Na de update staat er bij de MQTT-instellingen, voor JSON en split-topic (bij beiden), een nieuwe checkbox onderaan. 
Hiermee zou de applicatie voortaan de standaard tijdzone (Europe/Amsterdam) moeten toepassen op het tijdstip van de doorgestuurde metingen. Zowel het reguliere tijdstip als die van de gasmeter (als je die hebt).

dennissiemensma, 2018-07-31:
Dit gaat mee in de volgende release.

closed 2018-07-31 by dennissiemensma