dsmrreader/dsmr-reader #466: Error opening Settings page

Martinvdm, 2018-04-15:
Getting an error when opening the settings page. Did a full restore / rebuild of dsmr reader.

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: 1.16
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3b
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v2

### Bug, wens of iets anders
````

Server Error
Exception:

MultipleObjectsReturned: get() returned more than one User -- it returned 2!

Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/handlers/base.py", line 126, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/contrib/admin/sites.py", line 398, in login
    return LoginView.as_view(**defaults)(request)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 69, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/decorators.py", line 62, in _wrapper
    return bound_func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/decorators/debug.py", line 76, in sensitive_post_parameters_wrapper
    return view(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/decorators.py", line 58, in bound_func
    return func.__get__(self, type(self))(*args2, **kwargs2)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/decorators.py", line 62, in _wrapper
    return bound_func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/decorators.py", line 142, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/decorators.py", line 58, in bound_func
    return func.__get__(self, type(self))(*args2, **kwargs2)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/decorators.py", line 62, in _wrapper
    return bound_func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/decorators.py", line 58, in bound_func
    return func.__get__(self, type(self))(*args2, **kwargs2)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/contrib/auth/views.py", line 66, in dispatch
    return super().dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 89, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/edit.py", line 141, in post
    if form.is_valid():

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/forms/forms.py", line 179, in is_valid
    return self.is_bound and not self.errors

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/forms/forms.py", line 174, in errors
    self.full_clean()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/forms/forms.py", line 377, in full_clean
    self._clean_form()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/forms/forms.py", line 404, in _clean_form
    cleaned_data = self.clean()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/contrib/auth/forms.py", line 195, in clean
    self.user_cache = authenticate(self.request, username=username, password=password)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/contrib/auth/__init__.py", line 70, in authenticate
    user = _authenticate_with_backend(backend, backend_path, request, credentials)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/contrib/auth/__init__.py", line 116, in _authenticate_with_backend
    return backend.authenticate(*args, **credentials)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/contrib/auth/backends.py", line 16, in authenticate
    user = UserModel._default_manager.get_by_natural_key(username)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/contrib/auth/base_user.py", line 44, in get_by_natural_key
    return self.get(**{self.model.USERNAME_FIELD: username})

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 407, in get
    (self.model._meta.object_name, num)
````

dennissiemensma, 2018-04-16:
Bedankt voor je vraag. Er lijkt nu een dubbele gebruiker in de DB te zitten. Je kunt deze vrij makkelijk verwijderen via `psql`.

```
sudo su - postgres
psql dsmrreader

# Voer dit uit:

select id, username from auth_user;

# Je ziet nu als het goed is een lijst van gebruikers, waarbij ik verwacht dat er 
# één dubbel is. Verwijder er 1 van, wellicht de laatste, door het ID in deze 
# query te zetten op de plek van X:

delete from auth_user where id = X;
```
De foutmelding zou nu opgelost moeten zijn.

Martinvdm, 2018-04-16:
Thanks!
 idd stonden er 2 admins in. echter krijg ik nu de volgende error:
````
odePyCallable_Checkconn_closePyImport_FrozenModulesCONNECTION_AWAITING_RESPONSE__arr_PyAsyncGenASend_Typepgconn__ssize_tPyDictIterItem_TypePyODictItems_Typefsizeprintfuncdata_startPyODict_TypePyODictIter_Typepq_read_replication_message__pthread_slist_tCONNECTION_BADpq_raiserecursion_critical__timezonePQresultStatusPyDictProxy_Typebuiltins_copyexc_tracebackPyBufferProcsPyCmpWrapper_Typeml_flagstp_newarraysizeclosedPyClassMethod_Typepreparedfds_bitsnotice_listpq_clear_criticalPyErr_CheckSignalsnb_inplace_true_divideconsumingdestructorPyCFunctionmemset_IO_2_1_stderr_stderrresultPyObject_CallObjectPySet_Typename_Py_ascii_whitespace_sbufpq_resolve_criticalPyExc_UserWarningbuiltins_IO_save_endPQoidValue__nptrconn_text_from_chars__nexttp_delPyRange_TypePyInstanceMethod_Typekeepalive_intervalencodingPyEllipsis_Typestdouttp_nameconsume__time_tclosureinitfuncPGconnPQfreememerrorObjectPy_NoSiteFlagc_profilefunctp_as_sequencetp_as_bufferPQresStatuserr_forsizetypetstateitemsizePQclearnb_inplace_andfe_recvint64digitcurs_resetshort unsigned intPyGC_Headsigned charsysdictstring_typesasync_gen_firstitercriticalPyModule_TypePyMem_FreePyEnum_Typetp_allocpg_connml_docstatussuboffsetsnotuplesPyExc_TimeoutErrortstate_headtrash_delete_nesting__off64_tPyLong_FromSsize_t__lenpq_complete_error_IO_read_basePyZip_Type_offsetconnbinary_typesPyTupleIter_Typein_coroutine_wrapper_IO_buf_endtp_getattroallocfunc_PyNotImplemented_Typec_traceobjopterr_savePyType_IsSubtype_modePyReversed_Typetp_methods_IO_write_baseautocommitPQsendQueryPQnfieldsPGRES_COPY_BOTHtp_mroPyExc_MemoryErrorconnectionType_ISOC_tz_dsttimePyExc_BaseException_Py_SwappedOp_PyWeakref_ProxyTypePQputCopyEndPyExc_FloatingPointErrorclosePQgetCopyDataPyListIter_Type_PyTrash_delete_later_SVID__py_xdecref_tmplong intPGRES_EMPTY_QUERYPyErr_FetchPyExc_FileNotFoundErrornb_orpq_executedatabaseformatunaryfunc_IO_markerpsyco_default_binary_castPyByteArray_Type_Py_ctype_tabletimevalnb_floatdlopenflagsPyExc_SystemErrorsend_timePyDictValues_TypePyExc_ValueError_Py_PackageContexttraverseproc__pthread_internal_slistcontextPQsocketbufsizeinquirypq_get_guc_lockedPQfsizenb_invert_IO_2_1_stdout_psyco_set_error_pq_copy_out_v3ml_namePyExc_IndexErrorstrtolPySeqIter_Type__strduplong doublePyDictKeys_TypePyExc_Exceptionprotocoltp_as_numberfd_setPyStaticMethod_TypePy_BytesWarningFlagPyAsyncMethodstp_weaklistoffsetlong unsigned intml_methPyMem_Mallocreadonlytp_docpq_commitgetattrofuncpsyco_typesnewfuncWarning__errno_locationPyListRevIter_Typecharsrv_deferrablePyEval_SaveThreadnotifiesPySequenceMethodsstdin__spinson_delete_datatp_weaklist_IO_buf_basebufferinfo__nuserstpc_xidPGcancelnb_positivePyMap_Typehashfuncgetattrfunc_IO_read_endPyExc_ReferenceErrorcasts_IO_FILEqueryPyExc_NotADirectoryErrorPy
````

dennissiemensma, 2018-04-16:
Bedankt voor je terugkoppeling. Ik heb bovenstaande foutmelding nog nooit eerder gezien eerlijk gezegd. De admin-user heeft overigens geen rechten nodig, er zijn geen individuele rechten binnen de database zelf. Als je de installatie-guide hebt gevolgd staan ze prima.

Je geeft aan een restore/rebuild gedaan te hebben van het project. Kun je vertellen hoe je dat exact gedaan hebt? Is de installatie op een schoon OS geweest en heb je de vorige installatie van dsmr-reader helemaal verwijderd?

Martinvdm, 2018-04-16:
Hoi Dennis, fijn dat je reageerd. het is een schone install geweest, compleet volgens de manual. alleen restore van de db gedaan:
```sudo sudo -u postgres psql dsmrreader -f <PATH-TO-POSTGRESQL-BACKUP.sql>```

Wellicht een idee om de DB opnieuw aan te maken en een nieuwe restore te doen?

dennissiemensma, 2018-04-16:
Ik weet helaas niet of het aan de database ligt eerlijk gezegd. Ik neem aan dat je al geprobeerd hebt om de processen in Supervisor een keer te stoppen en starten?

Je kunt dat proberen en anders de database restoren van een back-up zoals je al aangeeft.

Martinvdm, 2018-04-16:
Herstart met supervisor heb ik al gedaan. Vreemde is wel dat het alleen het Control Panel betreft en niet de gewone pagina. De gegevens van de restore ook keurig in beeld staan en nieuwe gegevens ook inkomen. Ik ben alleen niet dusdanig postgresql technisch om een complete db restore te doen. 

Op een of andere manier denk ik ook niet dat het aan de db ligt omdat de normale frontend gewoon werkt maar logging van ngynx levert ook niks op. Wellicht iets met django?

dennissiemensma, 2018-04-16:
Ik had even gemist dat de nieuwe foutmelding alleen speelt bij de admin-interface en niet bij de rest van de applicatie.

Staan er toevallig nog foutmelding in de logfiles binnen dsmr-reader? In de map `logs`.

Martinvdm, 2018-04-16:
Ik krijg in IE (deed het eerst in Mozilla) een andere melding, waar je wellicht meer me kan:
````

Server Error

Exception:
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xfc in position 2: invalid start byte

Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/handlers/base.py", line 156, in _get_response
    response = response.render()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/response.py", line 106, in render
    self.content = self.rendered_content

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/response.py", line 81, in rendered_content
    template = self.resolve_template(self.template_name)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/response.py", line 63, in resolve_template
    return select_template(template, using=self.using)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/loader.py", line 42, in select_template
    return engine.get_template(template_name)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/backends/django.py", line 34, in get_template
    return Template(self.engine.get_template(template_name), self)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/engine.py", line 144, in get_template
    template, origin = self.find_template(template_name)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/engine.py", line 126, in find_template
    template = loader.get_template(name, skip=skip)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/loaders/cached.py", line 55, in get_template
    template = super().get_template(template_name, skip)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/loaders/base.py", line 24, in get_template
    contents = self.get_contents(origin)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/loaders/cached.py", line 24, in get_contents
    return origin.loader.get_contents(origin)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/template/loaders/filesystem.py", line 24, in get_contents
    return fp.read()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
````

dennissiemensma, 2018-04-16:
Het lijkt dan inderdaad iets anders te zijn, hoewel de foutmelding gaat over vreemde tekens. Je draait wel een vrij recente versie van Python zie ik, maar dat zou niet uit moeten maken. Tenminste, de tests voor 3.6 lijken goed te draaien (Ik heb zelf nog 3.4 en 3.5): https://travis-ci.org/dennissiemensma/dsmr-reader/builds/362162799

Ik zie ook geen vreemde fixes staan voor de aankomende release van django: https://docs.djangoproject.com/en/2.0/releases/2.0.5/

Staat er in de logfiles nog een andere foutmelding of is dat dezelfde? Vooral in `django.log` kan nog wel eens meer staan.

dennissiemensma, 2018-04-16:
Je zou nog eens kunnen proberen om django opnieuw te installeren:

```
sudo su - dsmr
pip install --upgrade --force-reinstall django
find ~/.virtualenvs/dsmrreader/ -iname *.pyc -type f -delete
```

* Daarna even de processen in supervisor herstarten. 

Het verwijderen van de *.pyc-files zorgt ervoor dat de cache van Python voor die files weg is. Daar komen heel af en toe ook vreemde dingen uit, dus goed om dat uit te sluiten.

Martinvdm, 2018-04-16:
supervisor seems to continuesly trying to start the services. STARTING / RUNNING switches

dennissiemensma, 2018-04-16:
Nginx is een externe tool van de applicatie. Ik moet echt weten wat de applicatie zelf doet.

* Werkt de rest van de processen in Supervisor wel of blijven ze allemaal herstarten?

* Kun je eens kijken in de logs van Supervisor wat de exacte foutmelding is? Bijvoorbeeld in `/var/log/supervisor/dsmr_webinterface.log` en `/home/dsmr/dsmr-reader/logs/django.log`

Martinvdm, 2018-04-16:
dsmr_webinterface.log
````
[2018-04-16 21:51:29 +0200] [18534] [ERROR] invalid syntax (__init__.py, line 1)
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/workers/base.py", line 135, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/app/wsgiapp.py", line 65, in load
    return self.load_wsgiapp()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/app/wsgiapp.py", line 52, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/gunicorn/util.py", line 352, in import_app
    __import__(module)
  File "/home/dsmr/dsmr-reader/dsmrreader/wsgi.py", line 16, in <module>
    application = get_wsgi_application()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/wsgi.py", line 12, in get_wsgi_application
    django.setup(set_prefix=False)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/apps/registry.py", line 89, in populate
    app_config = AppConfig.create(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/apps/config.py", line 90, in create
    module = import_module(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django_filters/__init__.py", line 1
    /* This file has been placed in the public domain. */
    ^
SyntaxError: invalid syntax
````

django.log
```
[2018-04-15 21:26:23,165] WARNING @ base | Not Found: /favicon.ico
[2018-04-16 18:08:33,676] WARNING @ base | Not Found: /favicon.ico
[2018-04-16 18:08:33,949] WARNING @ base | Not Found: /favicon.ico
[2018-04-16 20:39:41,372] WARNING @ base | Not Found: /favicon.ico
```

dennissiemensma, 2018-04-16:
Bedankt voor de info. Het lijkt toch telkens naar django te wijzen, gek genoeg. Ik zal kijken of ik lokaal python 3.6 kan installeren en het proberen te reproduceren. Dat gaat me helaas vandaag niet meer lukken.

Als je echt niet kan wachten kun je een schone reinstall proberen. In /home/dsmr/ gooi je dan het project en de .virtualenvs mappen helemaal weg. Database kun je eerst nog laten staan.

Martinvdm, 2018-04-16:
das vreemd. iets heel anders dan bij jou:
```
/* This file has been placed in the public domain. */
/* Following are the presentation styles -- edit away! */

html, body {margin: 0; padding: 0;}
body {background: white; color: black;}
:link, :visited {text-decoration: none; color: #00C;}
#controls :active {color: #888 !important;}
#controls :focus {outline: 1px dotted #222;}
h1, h2, h3, h4 {font-size: 100%; margin: 0; padding: 0; font-weight: inherit;}
```

dennissiemensma, 2018-04-16:
Ik denk dat `pip` een html-pagina of iets heeft gedownload, heel gek.

Je zou er dan ook nog voor kunnen kiezen om alleen de virtualenv (waar django enzo in staat) te wipen en herinstalleren, dat gaat wat vlotter dan alle stappen uitvoeren:

```
> Stop eerst alle processen in Supervisor.

# Delete
rm -rf /home/dsmr/.virtualenvs

# New
sudo sudo -u dsmr mkdir /home/dsmr/.virtualenvs
sudo sudo -u dsmr virtualenv /home/dsmr/.virtualenvs/dsmrreader --no-site-packages --python python3

# Reinstall
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt

```

dennissiemensma, 2018-04-16:
Als dat niet werkt kijk ik er morgen of overmorgen wel weer verder naar. Het moet bijna wel in de virtualenv zitten, want daar **lijken** al die fouten vandaan te komen...

Martinvdm, 2018-04-16:
you are the man! virtual env opnieuw geinstalleerd en werkt! zelfs de config pagina. very nice! thanks.
nu de configuratie weer instellen, deze zit niet in de dropbox dump denk ik....?
(overigens was het al een verse installatie, maar toch blijkbaar iets niet goed gegaan dan)

dennissiemensma, 2018-04-17:
Alle instellingen die je via de webinterface kan aanbrengen zitten in de database dump die in Dropbox staat. 

Het is echter wat lastig om alleen de instellingen uit de dump te halen, zoek je nog bepaalde instellingen? En werkt de applicatie nu helemaal, inclusief je bestaande verbruiksgegevens uit de database import eerder?

Martinvdm, 2018-04-17:
Ja het werkt weer top! ik mis geen instellingen meer. Alles was inderdaad terug gezet middels de restore, instellingen en data. Het enige wat wel ingevuld moest worden waren MQTT inschakelen en de PVOutput configuratie. Dit waren niet meegekomen met de restore. 
nogmaals dank voor de uitstekende support en het samen troubleshooten.

dennissiemensma, 2018-04-17:
Top, fijn dat je er toch uitgekomen bent zo, blijft soms lastig debuggen op afstand.

closed 2018-04-17 by dennissiemensma