dsmrreader/dsmr-reader #470: Geen gegevens gevonden

dwillemsen, 2018-05-01:
Ik heb DSMR reader geïnstalleerd en de datalogger draait, ik krijg via supervisor ook gewoon de drie processen te zien en de meter geeft ook data, alleen krijg ik in het dashboard geen gegevens te zien. Ik heb geprobeerd om het nogmaals te installeren maar met hetzelfde effect. Hebben jullie wellicht tips of suggesties wat ik zou kunnen doen want ik zie niet wat ik fout doe. Alvast bedankt

dennissiemensma, 2018-05-01:
Bedankt voor je ticket. Heb je toevallig ook al in de logs van Supervisor gekeken of daar iets in staat? 

In de log van de datalogger kun je zien of die data kan uitlezen en in de log van de backend kun je zien of de gegevens goed verwerkt worden.

Zie http://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

dwillemsen, 2018-05-02:
In de dsmr backend log zie ik dit soort berichten 
> _2018-05-02 00:36:53.578104+02:00: Calling backend services
> Command completed. Sleeping for 1 second(s)..._

In de log van de datalogger zie ik dit soort berichten, 
/ISK5\2M550E-1012

1-3:0.2.8(50)
0-0:1.0.0(180321120934W)
0-0:96.1.1(4530303433303036383931353037363137)
1-0:1.8.1(003368.531*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.219*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00009)
0-0:96.7.9(00006)
1-0:99.97.0(4)(0-0:96.7.19)(170508124328S)(0000000206*s)(170802124246S)(0000007674*s)(170802130512S)(0000001316*s)(171101234325W)(0000004990*s)
1-0:32.32.0(00008)
1-0:32.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(230.3*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.221*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0()
0-1:24.2.1(700101010000W)(00000000)
!492B
Command completed. Sleeping for 0.5 second(s)...

dus volgens mij gaat dat allemaal goed echter ik krijg ze niet te zien in het dashboard.

dennissiemensma, 2018-05-02:
In de log van de `backend` zou je als het goed is ook dit soort berichten moeten zien:

```
2018-05-02 18:36:00.219612+02:00: Calling backend services
 - Processed reading: 7328161 @ 2018-05-02 18:35:01+02:00 (0.230 kW)
 - Processed reading: 7328162 @ 2018-05-02 18:35:11+02:00 (0.235 kW)
 - Processed reading: 7328163 @ 2018-05-02 18:35:21+02:00 (0.229 kW)
 - Processed reading: 7328164 @ 2018-05-02 18:35:31+02:00 (0.228 kW)
Command completed. Sleeping for 1 second(s)...
```

* Zie je die wel?

* Staan er op de Statistieken-pagina een aantal metingen vermeld? (je moet dan op een knop klikken om de teller te tonen)

* En anders kun je het beste even in de database kijken wat die opgeslagen heeft:
```
sudo su - postgres

psql dsmrreader

select count(*) from dsmr_datalogger_dsmrreading;
```

Ik zie dan bijvoorbeeld:
```
pi@rpi3:~ $ sudo su - postgres 
postgres@rpi3:~$ psql dsmrreader
psql (9.4.15)
Type "help" for help.

dsmrreader=# select count(*) from dsmr_datalogger_dsmrreading;
 count  
--------
 196255
(1 row)
```

dwillemsen, 2018-05-02:
Dennis, ik heb het gecheckt en ik zie dan dat de count 0 is. Ik heb ook in de log van de backend niet de processed berichten staan

dennissiemensma, 2018-05-02:
Wanneer ik je meting lokaal probeer in te lezen krijg ik: 
```
CRC mismatch: 18731 (telegram) != 6971 (calculated)
```

Wat staat er in de andere logfiles?
* `/home/dsmr/dsmr-reader/logs/dsmrreader.log`
* `/home/dsmr/dsmr-reader/logs/django.log`

dennissiemensma, 2018-05-02:
Je zou dus kunnen proberen om CRC-verificatie uit te zetten in de instellingen. Al is het raar dat elk telegram afgewezen wordt (als dat daadwerkelijk zo is). 

Het zou wel verklaren waarom je nergens fouten ziet en de datalogger het wel lijkt te doen.

dwillemsen, 2018-05-04:
Ik denk dat ik het ga opgeven. Ik heb een volledig nieuwe install gedaan van mijn pi en van dsmr reader en nog steeds krijg ik geen data. Ik heb het zelfs twee keer gedaan om echt uit te sluiten dat er een fout in zat. Ik zou nu niet meer weten waarom het niet werkt.

dwillemsen, 2018-05-04:
Alles werkt verder naar behoren. Alle processen lopen de meter geeft data, alleen op 1 of andere manier komt het niet in de database

dennissiemensma, 2018-05-05:
Ik mis nog even wat er in bovenstaande logfiles staat en of CRC uitzetten helpt. 
Herinstalleren herstelt niets wanneer CRC de oorzaak is. Het staat onder Datalogger-instellingen dacht ik.

Momenteel kan ik trouwens alleen even per mail meelezen, dus wellicht dat ik iets gemist heb.

dwillemsen, 2018-05-05:
Ik heb CRC uitgezet (had ik ook al eerder gedaan) en ook dan geen gegevens. Ik heb ook geprobeerd om het via een andere poort te doen maar ook daar geen succes.

dwillemsen, 2018-05-08:
Sorry, ik was even een dagje weg. Ik heb in de dsmrreader.log gekeken en die is leeg. De django.log daarentegen is wel interessant want daarin staan dit soort berichten:

> [2018-05-08 12:10:27,702] ERROR @ services | Discarded telegram with future timestamp: /ISK5\2M550E-1012
> 
> 1-3:0.2.8(50)
> 0-0:1.0.0(180327153606S)
> 0-0:96.1.1(4530303433303036383931353037363137)
> 1-0:1.8.1(003428.812*kWh)
> 1-0:1.8.2(000000.000*kWh)
> 1-0:2.8.1(000000.000*kWh)
> 1-0:2.8.2(000000.000*kWh)
> 0-0:96.14.0(0001)
> 1-0:1.7.0(00.377*kW)
> 1-0:2.7.0(00.000*kW)
> 0-0:96.7.21(00009)
> 0-0:96.7.9(00006)
> 1-0:99.97.0(4)(0-0:96.7.19)(170508124328S)(0000000206*s)(170802124246S)(0000007674*s)(170802130512S)(0000001316*s)(171101234325W)(0000004990*s)
> 1-0:32.32.0(00008)
> 1-0:32.36.0(00001)
> 0-0:96.13.0()
> 1-0:32.7.0(229.5*V)
> 1-0:31.7.0(001*A)
> 1-0:21.7.0(00.395*kW)
> 1-0:22.7.0(00.000*kW)
> 0-1:24.1.0(003)
> 0-1:96.1.0()
> 0-1:24.2.1(700101010000W)(00000000)
> !95F0

Kennelijk accepteert hij geen telegrammen vanwege een future timestamp. Ik heb gecheckt maar voor zover ik kan zien staan de tijden correct.

dennissiemensma, 2018-05-08:
Aah juist, dat verklaart sowieso waarom je niets ziet, maar alles wel lijkt te werken.

Ik vraag me overigens af of je meter wel lekker loopt, want op `2018-05-08` zie ik `180327` staan, wat betekent dat de meter denkt dat het nog eind maart is? Zie je dat ook terug wanneer je de datalogger volgt? Het gaat om de regel die begint met `0-0:1.0.0`, waarbij het een datum + tijdstip is (YYMMDDhhmmss).

De oorzaak van het weigeren van de telegrammen zit overigens in de gasmeter (die je niet hebt?). Want daar staat `700101`, wat januari 2070 zou betekenen. Ik zal kijken of ik hier een uitzondering voor kan maken.

dennissiemensma, 2018-05-08:
@dwillemsen ik heb een mogelijke fix voor je. Wil je deze eens uitproberen?

```
sudo su - dsmr
git fetch
git checkout -b 470-future-timestamp origin/470-future-timestamp
./deploy.sh
```

Wanneer het gelukt is staat er rechtsbovenin de applicatie iets als `v1.16.1b1`.

Als je weer terug wilt wisselen:
```
sudo su - dsmr
git checkout master
./deploy.sh
```

Ik hoor graag of dit je probleem oplost.

dwillemsen, 2018-05-08:
Dennis, het lijkt nu beter te gaan. Ik krijg in ieder geval niet meer de melding dat er geen data is.
Ik zal het even een tijdje aankijken of hij ook echt alle data binnen krijgt. Maar in ieder geval super bedankt, het is al meer dan dat ik tot nu toe had. Ik hou je op de hoogte.

dwillemsen, 2018-05-10:
Dennis, ik krijg nu de waarden wel binnen alleen de datum is 1 maand en 1 week in het verleden, dus in de overzichten krijg ik geen grafieken te zien omdat die maar tot maximaal een maand terug gaan. Hoe zouden we dit kunnen oplossen?

dennissiemensma, 2018-05-10:
Ik denk dat de interne klok van je slimme meter verkeerd loopt. Als dat zo is, dan mag de netbeheerder dat oplossen vermoed ik, mogelijk kunnen ze dat op afstand.

Zou je hier voor de zekerheid een recent telegram willen posten + tijdstip van uitlezen? Dan kunnen we het dubbelchecken:
`sudo supervisorctl tail -f dsmr_datalogger`

dwillemsen, 2018-05-10:
Bijgaand een recent bericht van zojuist

> /ISK5\2M550E-1012
> 
> 1-3:0.2.8(50)
> 0-0:1.0.0(180329151847S)
> 0-0:96.1.1(4530303433303036383931353037363137)
> 1-0:1.8.1(003447.244*kWh)
> 1-0:1.8.2(000000.000*kWh)
> 1-0:2.8.1(000000.000*kWh)
> 1-0:2.8.2(000000.000*kWh)
> 0-0:96.14.0(0001)
> 1-0:1.7.0(00.384*kW)
> 1-0:2.7.0(00.000*kW)
> 0-0:96.7.21(00009)
> 0-0:96.7.9(00006)
> 1-0:99.97.0(4)(0-0:96.7.19)(170508124328S)(0000000206*s)(170802124246S)(0000007674*s)(170802130512S)(0000001316*s)(171101234325W)(0000004990*s)
> 1-0:32.32.0(00008)
> 1-0:32.36.0(00001)
> 0-0:96.13.0()
> 1-0:32.7.0(231.7*V)
> 1-0:31.7.0(001*A)
> 1-0:21.7.0(00.357*kW)
> 1-0:22.7.0(00.000*kW)
> 0-1:24.1.0(003)
> 0-1:96.1.0()
> 0-1:24.2.1(700101010000W)(00000000)
> !CE77
>

dwillemsen, 2018-05-10:
Gas doet het trouwens ook niet terwijl dat wel zou moeten

dennissiemensma, 2018-05-10:
Bedankt, kun je eens kijken op je meter zelf of je deze standen ziet? (om zeker te zijn dat alleen de tijd niet klopt, maar de standen wel):

* De meter geeft bij jouw telegram alleen deze stand door: `003447.244 kWh`

dennissiemensma, 2018-05-24:
@dwillemsen heb je bovenstaande al geprobeerd?

dwillemsen, 2018-05-26:
Dennis, de standen kloppen. Ook op de meter krijg ik alleen laag tarief en die komt overeen met wat er uitgelezen wordt. Ik heb inmiddels bij Stedin aangegeven dat mijn meter waarschijnlijk niet goed werkt en ze gaan er naar kijken. Zodra ze dat gedaan hebben, zal ik even terugkoppelen wat er uit is gekomen.

dwillemsen, 2018-05-28:
Heb vandaag terugkoppeling gehad van Stedin. De meter werkt niet, ze kunnen hem op afstand ook niet zien. Ik hoor in de loop van de week wat de vervolgstappen zijn.

dennissiemensma, 2018-05-28:
Bedankt voor je update! Dat verklaart inderdaad een hoop. Ik hoop voor je dat ze het fatsoenlijk voor je oplossen zodat je straks alsnog je meterstanden via de kabel kunt uitlezen. :]

dennissiemensma, 2018-06-11:
@dwillemsen hebben ze inmiddels je meter vervangen of gerepareerd?

dwillemsen, 2018-06-12:
Dennis, ik heb nog niets gehoord. Zal wel enige tijd duren.

dennissiemensma, 2018-07-22:
@dwillemsen zijn ze al bij je langsgeweest of blijft het stil van hun kant?

dwillemsen, 2018-07-22:
Ze komen op 1 augustus

Verstuurd vanaf mijn iPad

dennissiemensma, 2018-07-22:
Prima, dank voor je update.

dennissiemensma, 2018-08-05:
@dwillemsen doet je (nieuwe) meter het nu beter? Of nog niet kunnen uitproberen?

dwillemsen, 2018-08-05:
Ze stuurden een mail dat ze niet konden komen en ik moet gaan herplannen, maar nu eerst op vakantie

dennissiemensma, 2018-08-05:
Helder, veel plezier ;-)

dennissiemensma, 2018-09-18:
@dwillemsen heeft de netbeheerder je meter inmiddels vervangen?

dennissiemensma, 2018-09-27:
Ik sluit deze voor nu. Mocht je een nieuwe meter hebben en er dan nog steeds last van hebben dat de applicatie niet goed werkt, dan hoor ik het uiteraard graag!

closed 2018-09-27 by dennissiemensma